awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #731: curlCompleteSync(): curl perform failed for url https://kinesisvideo."ap-southeast-1".amazonaws.com/describeStream with result Couldn't resolve host name: Could not resolve host: kinesisvideo."ap-southeast-1".amazonaws.com[QUESTION]

nizamk, 2021-07-27:
Hi,

I'm planning to stream from RTSP source to kinesis video stream. For a test, I performed streaming from mp4 file to kinesis video stream via following command:
C:\...\amazon-kinesis-video-streams-producer-sdk-cpp\build> gst-launch-1.0 -v filesrc location="sample.mp4" ! qtdemux name=demux ! queue ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="mystream" access-key=".." secret-key=".."

I got the following result & errors:

Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2021-07-28 06:38:07 [8344] INFO - heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2021-07-28 06:38:07 [8344] INFO - heapInitialize(): Creating AIV heap.
2021-07-28 06:38:07 [8344] INFO - heapInitialize(): Heap is initialized OK
2021-07-28 06:38:07 [8344] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-07-28 06:38:07 [8344] DEBUG - getSecurityTokenHandler invoked
2021-07-28 06:38:07 [8344] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 16274254874579082 Expiration: 0
2021-07-28 06:38:07 [8344] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2021-07-28 06:38:07 [8344] INFO - createDeviceResultEvent(): Create device result event.
2021-07-28 06:38:07 [8344] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2021-07-28 06:38:07 [8344] DEBUG - clientReadyHandler invoked
2021-07-28 06:38:07 [8344] INFO - try creating stream
2021-07-28 06:38:07 [8344] INFO - Creating Kinesis Video Stream mystream
2021-07-28 06:38:07 [8344] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Stream name: mystream
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Streaming type: STREAMING_TYPE_REALTIME
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Content type: video/h264
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Max latency (100ns): 600000000
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Fragment duration (100ns): 20000000
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Key frame fragmentation: Yes
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Use frame timecode: Yes
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Absolute frame timecode: Yes
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Nal adaptation flags: 0
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Average bandwith (bps): 4194304
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Framerate: 25
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Buffer duration (100ns): 1200000000
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Replay duration (100ns): 400000000
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Connection Staleness duration (100ns): 600000000
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Store Pressure Policy: 1
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     View Overflow Policy: 1
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Segment UUID: NULL
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Frame ordering mode: 0
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo(): Track list
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Track id: 1
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Track name: kinesis_video
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Codec id: V_MPEG4/ISO/AVC
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Track type: TRACK_INFO_TYPE_VIDEO
2021-07-28 06:38:07 [8344] DEBUG - logStreamInfo():     Track cpd: NULL
2021-07-28 06:38:07 [8344] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-07-28 06:38:07 [8344] WARN - curlCompleteSync(): curl perform failed for url https://kinesisvideo."ap-southeast-1".amazonaws.com/describeStream with result Couldn't resolve host name: Could not resolve host: kinesisvideo."ap-southeast-1".amazonaws.com
2021-07-28 06:38:07 [23064] WARN - curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo."ap-southeast-1".amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=....../20210727/"ap-southeast-1"/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=2dbb0170de4dc540aa30763cdde12623799b6b15d7dea70895d9da797dfd19d9
    content-length: 29
    content-type: application/json
    host: kinesisvideo."ap-southeast-1".amazonaws.com
    user-ag
2021-07-28 06:38:07 [23064] DEBUG - describeStreamCurlHandler(): DescribeStream API response:
2021-07-28 06:38:07 [23064] INFO - describeStreamResultEvent(): Describe stream result event.
2021-07-28 06:38:07 [23064] WARN - curlCompleteSync(): curl perform failed for url https://kinesisvideo."ap-southeast-1".amazonaws.com/describeStream with result Couldn't resolve host name: Could not resolve host: kinesisvideo."ap-southeast-1".amazonaws.com
2021-07-28 06:38:07 [6340] WARN - curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo."ap-southeast-1".amazonaws.com/describeStream
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=...../20210727/"ap-southeast-1"/kinesisvideo/aws4_request, SignedHeaders=host;user-agent;x-amz-date, Signature=2dbb0170de4dc540aa30763cdde12623799b6b15d7dea70895d9da797dfd19d9
    content-length: 29
    content-type: application/json
    host: kinesisvideo."ap-southeast-1".amazonaws.com
    user-ag
2021-07-28 06:38:07 [6340] DEBUG - describeStreamCurlHandler(): DescribeStream API response:
2021-07-28 06:38:07 [6340] INFO - describeStreamResultEvent(): Describe stream result event.
2021-07-28 06:38:07 [6340] WARN - curlCompleteSync(): curl perform failed for url https://kinesisvideo."ap-southeast-1".amazonaws.com/describeStream with result Couldn't resolve host name: Could not resolve host: kinesisvideo."ap-southeast-1".amazonaws.com
2021-07-28 06:38:08 [17832] WARN - curlCompleteSync(): HTTP Error 0 : Response: (null)
R

** Platform **
OS: Windows 10 Enterprise
GStreamer: c:\gstreamer\1.0\msvc_x86_64\
Built tools: x86 Native Tools Command Prompt
Build command: build_windows.bat

** NOTE **
I have built the kvssink & attempted the above command on Ubuntu 18.X successfully.

Thanks
nizam

vattay, 2021-07-28:
Somewhere you are setting the region with quotes and Kinesis doesn't strip them, so it's trying to query https://kinesisvideo."ap-southeast-1".amazonaws.com/describeStream, which doesn't exist.

disa6302, 2021-07-28:
Please set up the region appropriately. As pointed out by @vattay , there shouldnt be double quotes in the URL.

closed 2021-07-28 by disa6302