awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #733: Gstreamer & Pthread [BUG]

rsurati, 2021-08-05:
- OS: Windows-10 (64 bit)

When I try to make the files using  cmake command and the gstream plugin is set to TRUE.
The error logs are telling that <pthread.h> is missing.

ALthough, I installled the pthread library for 64 bit on my windows machine through vcpkg by command:
 **"vcpkg install pthreads:x64-windows".** 

Further I specified the path to <pthread.h> by the following commad:

**cmake .. -G "NMake Makefiles" -DPTHREAD_INCLUDE_DIR="path\to\pthreads_x64-windows\include" -DBUILD_GSTREAMER_PLUGIN=True**

Then also error logs are telling <pthread.h> not found. Although the command prompt is telling
 **'gstreamer-app-1.0' required package not found.** 

Can anyone have the leads to this?

jdelapla, 2021-08-05:
@rsurati could you share the logs including the error message?

rsurati, 2021-08-06:
> @rsurati could you share the logs including the error message?

Yes sure. 

CmakeError.log:

Determining if the include file pthread.h exists failed with the following output:
Change Dir: D:\PATH\TO\amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeTmp

Run Build Command(s):nmake -f Makefile /nologo cmTC_cae66\fast && 	"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\bin\HostX64\x64\nmake.exe"  -f CMakeFiles\cmTC_cae66.dir\build.make /nologo -L                  CMakeFiles\cmTC_cae66.dir\build

Building C object CMakeFiles/cmTC_cae66.dir/CheckIncludeFile.c.obj

	"C:\PATH-TO\CMake\bin\cmake.exe" -E cmake_cl_compile_depends --dep-file=CMakeFiles\cmTC_cae66.dir\CheckIncludeFile.c.obj.d --working-dir=D:\path-to\Repo\amazon-kinesis-video-streams-producer-sdk-cpp\build\CMakeFiles\CMakeTmp --filter-prefix="Note: including file: " -- C:\PROGRA~2\MICROS~2\2019\COMMUN~1\VC\Tools\MSVC\1429~1.300\bin\Hostx64\x64\cl.exe @C:\PATH-To\AppData\Local\Temp\nm5EE3.tmp

CheckIncludeFile.c
D:\PATH-TO\Repo\amazon-kinesis-video-streams-producer-sdk-cpp\build\CMakeFiles\CMakeTmp\CheckIncludeFile.c(1): fatal error C1083: Cannot open include file: 'pthread.h': No such file or directory
NMAKE : fatal error U1077: '"C:\PATH-TO\CMake\bin\cmake.exe"' : return code '0x2'

Stop.

NMAKE : fatal error U1077: '"C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.29.30037\bin\HostX64\x64\nmake.exe"' : return code '0x2'

Stop.

hassanctech, 2021-08-16:
We build for Windows with every commit, here is what we do in our CI:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.travis.yml#L107

And the link the the script is here:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.github/build_windows.bat

Please follow those steps and see if that works for you?

hassanctech, 2021-09-07:
Closing due to inactivity.  If there is still a question please re-open.

closed 2021-09-07 by hassanctech