awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #734: KVS Audio Only Streaming

mudassir-mukhtar-ahmed, 2021-08-08:
Hello,

I'm looking to use KVS to stream audio only. Currently able to stream audio using gstreamer plugin on my system with the below command.

`gst-launch-1.0 -v kvssink name=sink stream-name="teststream" access-key="" secret-key="" aws-region="us-east-1"  osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink.`

However i'm unable to playback the stream using HLS or DASH because the API expects the stream to have a video encoded in H264 to generate a HLS or DASH streaming url. AWS Enterprise support recommended a work around to send a static image along with the audio stream to playback successfully. However i'm unable to figure out how to do that from the gstreamer docs. Appreciate if someone can help on how to stream audio with a static image ( or a sample video? ) to enable HLS or DASH playback ? Also if this is not possible at all, please let me know, so i don't waste my time in trial and error.

M-KH, 2021-08-11:
We are facing the same issue. We couldn't find any documentation on how to send audio only.

hassanctech, 2021-08-11:
You have the pipeline elements reversed so the source element needs to come first and then you can send it to kvssink.  Perhaps you ca use `autovideosrc` or some other dummy video h264 source, then you can merge them together (in your GStreamer pipeline) and then take that result and send it to kvssink. 

Alternatively you can use our C SDK here are samples:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/master/samples/KvsAacAudioVideoStreamingSample.c

https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/master/samples/KvsVideoOnlyStreamingSample.c

But you need to modify this to be audio only. 

But your issue again is for HLS there needs to be a video track, so you would need to use the audio/video sample, see how in our sample we just read the frames from disk, they are in the repo here:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/tree/master/samples

You can do something like that just have the same frame you send over and over.  Keep in mind there are costs associated with the bandwidth so adding video will mean higher bandwidth, even if its very compressible it will be non zero.

In the mean time I'll see if there's a way to get an HLS or DASH stream with audio only.

hassanctech, 2021-08-11:
Yeah I've confirmed that is the case today we do not support audio only for HLS/DASH, it is in our roadmap but we have no ETA.  

If you'd like to use GStreamer then I'd do what I recommended above (you can even use `filesrc` to read h264 encoded static frames, you can keep 1 GOP of low bandwidth frame, low quality that you replay over and over again) there are many ways to go about it.  If you need help with that I would reach out to the GStreamer forums basically your goal is to first construct a pipeline which generates the dummy h264 mixed with your aac audio and try to get it to work with `filesink`.  Once you're able to validate on your end that your pipeline works and you can successfully stream it to a file and play it back with a player like VLC then just substitute `filesink` with `kvssink` and you should be good to go.

mudassir-mukhtar-ahmed, 2021-08-12:
thanks @hassanctech i will try out steps mentioned above and let u know how it works out. Appreciate the detailed response.

hassanctech, 2021-08-16:
Thanks, closing as there is no Kinesis Video Streams, issue.  If you have a specific bug / question about the KVS SDKs, please open a new issue with those details.

closed 2021-08-16 by hassanctech

bml1g12, 2022-03-28:
> I've confirmed that is the case today we do not support audio only for HLS/DASH, it is in our roadmap but we have no ETA.

@mudassir-mukhtar-ahmed  Did you find a working solution for this? Would be grateful if you can share if so, or share what didn't work.