awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #737: [QUESTION] send video to kvs with custom producer timestamp

dorfindoor, 2021-08-18:
Hi!
Is there any way to upload a video to kvs, using gstreamer kvssink with custom producer timestamp?
What I'm trying to resolve, is that sometimes my wifi gets disconnected, and I record   my video locally, I want that after the wifi returns I could upload this video and on the kvs it will be shown as it video tool place at its original time..
Is there anyway to do so?

Thank you

Winneee, 2021-08-30:
[image]

Hi You can try to modify PROP_FILE_START_TIME with your video start time.
Please find attached image. and for more information check gstkvssink.cpp

hassanctech, 2021-09-07:
What you describe is by design the SDK does not change the producer time stamp (producer means when the encoder generated it).  Since the retry happens internal to the SDK there is no way using the gstreamer plugin to do what you describe.  We do not allow the opportunity to intercept the frame when it is being retried.

closed 2021-09-07 by hassanctech