awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #738: kvssink seem to not close when it get GST_STATE_NULL in gstreamer pipeline

waikup83, 2021-08-18:
I have a pipeline with kvssink at the end and when closing a gstreamer by changing it with gst_element_set_state(app->Pipeline, GST_STATE_NULL) it seem that kvssink rest in memory. (i see it with thread number for the process)

i test my pipeline with fakesink at the end and it's ok

**Platform (please complete the following information):**
 - OS: linux
 - Version 1.0

disa6302, 2021-08-18:
@waikup83 , 

The information is not clear. What is it that you see when closing gstreamer? Does the application not terminate? Have you made any changes to the sample?

waikup83, 2021-08-18:
rought exemple included

[Nouveau document texte.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/7009481/Nouveau.document.texte.txt)

when i fakesink i got 2 thread after few second but when im doing the same thing with kvssink with acces code and all setting it remain with 3 thread (seeing with htop)

waikup83, 2021-08-18:
output => 

Playing
Stream is ready
*** Loop
Start video buffer
Stop video buffer
Stopping
*** QUIT

disa6302, 2021-08-19:
Do you see any errors in the log? Are you invoking `stopSync()`. Please run the samples to see if you encounter the issue. As such, the information is not enough to deduce the issue or indicate a kvs issue. As such, it could also be the way in which it is being used and we unfortunately cannot debug custom applications.

waikup83, 2021-08-19:
I'm not invoking stopSync() because i've used Gstreamer element. Maybe it's not a bug but a feature needed because exemple i've seen with gstreamer and kvssink it's just a main() that finish at the end but I don't want to finish my main(). I want kvssink multi streaming and stop when needed. All my code work but when i stop the GStreamer pipeline, the kvssink have a thread that doesn't stop.

The code in file txt earlier have roughly what code does and all.

waikup83, 2021-08-19:
maybe a ready_to_NULL in gst_kvs_sink_change_state can stop the stream entirely

waikup83, 2021-08-19:
mhummm => https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/682

closed 2021-08-19 by waikup83