awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #747: [QUESTION] Cannot download dependent - 'boringssl'

lidong-jimi, 2021-09-01:
When I first command cmake .. 
Occur Error:
  unable to access 'https://boringssl.googlesource.com/boringssl/': Failed to connect to boringssl.googlesource.com port 443: 
  Operation timed out.

I can't download even if I set the agent. 
Which file can I modify to replace the download address of boringssh from **https://boringssl.googlesource.com/boringssl/** to **https://github.com/google/boringssl**

lidong-jimi, 2021-09-01:
Solved

disa6302, 2021-09-13:
@lidong-jimi , 

Can you include more details? 
1. What command are you running?
2. What is the platform being used? 

If it is Windows, re-running CMake seems to have solved the problem in the past. OpenSSL building is flaky on WIndows. For Linux/MacOS, ensure the git checkout operation is able to reach the URL.