awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #749: [QUESTION] unable to build on windows (CURL 32)

Lepm11, 2021-09-07:
HI.
I'm trying to build on windows 10 and I've this error, somebody knows what could be?

[image]

hassanctech, 2021-09-07:
We build on Windows in our Travis CI:

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.travis.yml#L108-L117

Can you please try that in case you're doing something different?

Lepm11, 2021-09-12:
> We build on Windows in our Travis CI:
> 
> https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.travis.yml#L108-L117
> 
> Can you please try that in case you're doing something different?

I still have the same error and I had to reset the pc, currently I was able to execute the command "Nmake MakeFiles" and the files were created correctly, however, when I running the command "cmake .. -DBUILD_GSTREAMER_PLUGIN = ON -DBUILD_JNI = TRUE" it gives me an error this error Could NOT find JNI (missing: JAVA_AWT_LIBRARY JAVA_JVM_LIBRARY

can anybody help me?

disa6302, 2021-09-13:
@Lepm11 , 

We recently updated CMake version in this PR which could solve the issue: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/748

It has not made to the release yet. Can you see if updating CMake helps your case?

Lepm11, 2021-09-14:
So far i have not been able to install on windows 10, i tried on ubuntu 20.04 and it worked fine. Windows generates many errors

Lepm11, 2021-09-27:
in my CMakeList had the following route for the gstreamer:

C:\gstreamer\1.0\x86_64\lib

but my gstramer is installed on C:\gstreamer\1.0\msvc_x86_64\lib

so I changed the route from the CMakeList and its worked on windows 10 !!
I was able to installed on Ubuntu 20.04 and Windows 10.

closed 2021-09-27 by Lepm11