awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #750: [QUESTION] Stale connection retry period and server request timeout setting

crcsylvia, 2021-09-09:
Questions in SDK v3.2.0 [reference to Case1 in #707]: 
We observe that the streaming will continue after 30 mins when stale connection occurred. 
Q1. Is there any method to adjust this time range? like 10 mins?
New question: 
We met HTTP 408 frequently in SDK v3.2.0 and continuously received it for a long time. (restart the program will recover sooner, might be due to the environment limitation).
Q2. How long the server request timeout is?
Q3. Is there any method to adjust the timeout to be more tolerated?

Thank you!

_Originally posted by @crcsylvia in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/707#issuecomment-913304716_

MushMal, 2021-09-09:
There is a similar issue discussed here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/issues/231

Unfortunately, the staleness state machinery in the default callback provider is somewhat broken. Your application could respond to the staleness by re-setting. Alternatively, try to debug the root cause of the staleness and perhaps change the parameters to hit the latency rather than staleness first?

crcsylvia, 2021-09-10:
Thanks for your reply. The producer-c-231 you post is almost the same as the situation we've met.
We will adjust our StreamDefinition (max_latency, buffer_duration, replay_duration, connection_staleness) first and see if there is any improvement.
And what is the correct reset flow for this issue? Can stream->resetConnection() achieve this goal?

MushMal, 2021-09-10:
Yes, it will reset the connection. However, consider having the buffer duration something like 40+ seconds, max latency around 30 and staleness greater than the max latency. This would exercise the latency pressure codepath logic instead of the staleness.

disa6302, 2021-09-13:
Closing assuming question answered.

closed 2021-09-13 by disa6302