awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #753: still search for memory leak

waikup83, 2021-09-10:
i'm still searching for leak with kvssink when just stopping/ restarting gstreamer without terminating main progam - #682

** Any design considerations/constraints **
- i have 2 appsrc that push audio / video
- not terminating the program, i'm stop and restart gstreamer loop

making kvssink object with :
	KvsSink = gst_element_factory_make("kvssink", "sink");
	g_object_set(G_OBJECT(KvsSink), "stream-name", app->Stream.StreamName, NULL);
	g_object_set(G_OBJECT(KvsSink), "access-key", app->Stream.StreamAccessKey, NULL);
	g_object_set(G_OBJECT(KvsSink), "secret-key", app->Stream.StreamSecretKey, NULL);
	g_object_set(G_OBJECT(KvsSink), "aws-region", app->Stream.StreamRegion, NULL);

waikup83, 2021-09-24:
many log4cplus problem with kvssink found by valgrind with two silmultanious thread running gstreamer ans kvssink

[valgrindRaw.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/7225897/valgrindRaw.log)

hassanctech, 2021-11-29:
Haven't had a chance to investigate but I see you opened another issue:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/765

Which is likely why this is happening if it is an issue, I think we need to implement that.  Feel free to submit a PR if you have the change on your side, that is always welcome!

disa6302, 2022-04-24:
Closing as dup of #765 .

closed 2022-04-24 by disa6302