awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #754: Error connecting to RTSP stream

daveisfera, 2021-09-13:
I can connect to the RTSP stream with VLC, but it errors out when using the example in a docker container

** Logging **
Here's the command:
```
gst-launch-1.0 rtspsrc location="rtsp://<username>:<password>@192.168.4.246/live1s1.sdp" short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="KvsTest" storage-size=512 access-key="<access_key>" secret-key="<secret_key>" aws-region="us-east-1"
```

Here's the output:
```
Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2021-09-13 23:39:26 [140513556444992] INFO - heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
2021-09-13 23:39:26 [140513556444992] INFO - heapInitialize(): Creating AIV heap.
2021-09-13 23:39:26 [140513556444992] INFO - heapInitialize(): Heap is initialized OK
2021-09-13 23:39:26 [140513556444992] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-09-13 23:39:26 [140513556444992] DEBUG - getSecurityTokenHandler invoked
2021-09-13 23:39:26 [140513556444992] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1631576366088534600 Expiration: 0
2021-09-13 23:39:26 [140513556444992] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2021-09-13 23:39:26 [140513556444992] INFO - createDeviceResultEvent(): Create device result event.
2021-09-13 23:39:26 [140513556444992] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2021-09-13 23:39:26 [140513556444992] DEBUG - clientReadyHandler invoked
2021-09-13 23:39:26 [140513556444992] INFO - try creating stream
2021-09-13 23:39:26 [140513556444992] INFO - Creating Kinesis Video Stream DaveTest
2021-09-13 23:39:26 [140513556444992] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Stream name: DaveTest 
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Content type: video/h264 
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Max latency (100ns): 600000000
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Fragment duration (100ns): 20000000
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Key frame fragmentation: Yes
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Use frame timecode: Yes
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Absolute frame timecode: Yes
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Nal adaptation flags: 0
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Average bandwith (bps): 4194304
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Framerate: 25
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Buffer duration (100ns): 1200000000
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Replay duration (100ns): 400000000
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Store Pressure Policy: 1
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	View Overflow Policy: 1
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Segment UUID: NULL
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Frame ordering mode: 0
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): Track list
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Track id: 1
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Track name: kinesis_video
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
2021-09-13 23:39:26 [140513556444992] DEBUG - logStreamInfo(): 	Track cpd: NULL
2021-09-13 23:39:26 [140513556444992] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-09-13 23:39:26 [140513556444992] INFO - writeHeaderCallback(): RequestId: 2102de0f-a6dc-4dfb-8c1f-0c405562e7c9
2021-09-13 23:39:26 [140512910898944] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.631568461072E9,"DataRetentionInHours":720,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-1:123456789012:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:234567890123:stream/KvsTest/3456789012345","StreamName":"KvsTest","Version":"45678901234567890123"}}
2021-09-13 23:39:26 [140512910898944] INFO - describeStreamResultEvent(): Describe stream result event.
2021-09-13 23:39:26 [140512910898944] WARN - describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2021-09-13 23:39:26 [140512910898944] WARN - describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2021-09-13 23:39:26 [140512910898944] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2021-09-13 23:39:26 [140512910898944] INFO - writeHeaderCallback(): RequestId: a08797d6-fd69-44e0-9ac3-8c63c02c2601
2021-09-13 23:39:26 [140512902506240] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-12345678.kinesisvideo.us-east-1.amazonaws.com"}
2021-09-13 23:39:26 [140512902506240] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2021-09-13 23:39:26 [140512902506240] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
2021-09-13 23:39:26 [140512902506240] DEBUG - getStreamingTokenHandler invoked
2021-09-13 23:39:26 [140512902506240] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: 1631576366907135100 Expiration: 18446744073709551615
2021-09-13 23:39:26 [140512902506240] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
2021-09-13 23:39:26 [140512902506240] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2021-09-13 23:39:26 [140512902506240] DEBUG - streamReadyHandler invoked
2021-09-13 23:39:26 [140512902506240] Stream is ready
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
[... 298 lines cut ...]
0:00:03.224255200   101 0x7f0f40004ca0 INFO                GST_PADS gstpad.c:4112:gst_pad_peer_query:<rtspsrc0:recv_rtp_src_2_13568361_107> pad has no peer
0:00:03.224356900   101 0x7f0f40004ca0 INFO                    task gsttask.c:318:gst_task_func:<rtpjitterbuffer0:src> Task going to paused
0:00:03.225958900   101 0x7f0f880e0800 INFO                 basesrc gstbasesrc.c:2860:gst_base_src_loop:<udpsrc5> pausing after gst_pad_push() = not-linked
0:00:03.226006200   101 0x7f0f880e0800 WARN                 basesrc gstbasesrc.c:2950:gst_base_src_loop:<udpsrc5> error: Internal data stream error.
0:00:03.226017100   101 0x7f0f880e0800 WARN                 basesrc gstbasesrc.c:2950:gst_base_src_loop:<udpsrc5> error: streaming stopped, reason not-linked (-1)
0:00:03.226037700   101 0x7f0f880e0800 INFO        GST_ERROR_SYSTEM gstelement.c:1890:gst_element_message_full_with_details:<udpsrc5> posting message: Internal data stream error.
0:00:03.226064000   101 0x7f0f880e0800 INFO        GST_ERROR_SYSTEM gstelement.c:1917:gst_element_message_full_with_details:<udpsrc5> posted error message: Internal data stream error.
0:00:03.226114800   101 0x7f0f880e0800 INFO                    task gsttask.c:318:gst_task_func:<udpsrc5:src> Task going to paused
0:00:03.230993000   101 0x7f0f880e0230 INFO               GST_EVENT gstevent.c:808:gst_event_new_caps: creating caps event application/x-rtcp
[... cut ...]
0:00:03.460346500   101      0x2167680 INFO        GST_ERROR_SYSTEM gstelement.c:1917:gst_element_message_full_with_details:<rtspsrc0> posted warning message: Could not read from resource.
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
Additional debug info:
gstrtspsrc.c(4754): gst_rtspsrc_loop_udp (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
The server closed the connection.
```

** Any design considerations/constraints **
We're looking to use KVS with existing cameras that can't send video directly to KVS but that can stream the video using RTSP, so this is doing the mapping for us.

vbbandeira, 2021-09-15:
hello there, try use h264parse after ! rtph264depay !

daveisfera, 2021-09-15:
That worked! Thanks!

closed 2021-09-15 by daveisfera