awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #755: [QUESTION] Recording multiple RTSP streams h265 format using Gstreamer and Kvssink

vbbandeira, 2021-09-14:
Hello there folks,

I need to record 4 RTSP streams into a single stream of the Kinesis Video Streams.
Streams must be placed in the video like this:

[image]

I was able to insert a single stream and make it work perfectly, using the command below:
```
gst-launch-1.0 rtspsrc user-id="admin" user-pw="password" location="rtsp://admin:password@192.168.0.1:554/cam/realmonitor?channel=1&subtype=0" short-header=TRUE ! rtph265depay ! h265parse ! video/x-h265, alignment=au ! kvssink stream-name="test-stream" storage-size=512 access-key="access-key" secret-key="secret-key" aws-region="us-east-1" 
```

However, my goal is to insert an array of streams into the same stream in the Kinesis Video Streams.
For this I found the example with ```videomixer``` that I'll insert below:
```
gst-launch-1.0 -e rtspsrc location=rtsp_url1 ! rtph264depay ! h264parse ! decodebin ! videoconvert! m.sink_0 \
                rtspsrc location=rtsp_url2 ! rtph264depay ! h264parse ! decodebin ! videoconvert! m.sink_1 \
                rtspsrc location=rtsp_url3 ! rtph264depay ! h264parse ! decodebin ! videoconvert! m.sink_2 \
                rtspsrc location=rtsp_url4 ! rtph264depay ! h264parse ! decodebin ! videoconvert! m.sink_3 \
                videomixer name=m sink_1::xpos=1280 sink_2::ypos=720 sink_3::xpos=1280 sink_3::ypos=720 ! x264enc ! mp4mux ! filesink location=./out.mp4 sync=true
```

I adapted the example to just two streams and made it work inside the container, using a command like the one below:
```
gst-launch-1.0 -e rtspsrc user-id="admin" user-pw="password" location="rtsp://password@192.168.0.1:554/cam/realmonitor?channel=1&subtype=0" short-header=TRUE ! rtph265depay ! h265parse ! video/x-h265, alignment=au ! libde265dec ! videoconvert ! m.sink_0 \
        rtspsrc user-id="admin" user-pw="password" location="rtsp://password@192.168.0.2:554/cam/realmonitor?channel=1&subtype=0" short-header=TRUE ! rtph265depay ! h265parse ! video/x-h265, alignment=au ! libde265dec ! videoconvert ! m.sink_1 \
        videomixer name=m sink_0::xpos=1080 sink_1::ypos=1080 ! x265enc ! h265parse ! video/x-h265, alignment=au ! kvssink stream-name="test-stream" storage-size=512 access-key="access-key" secret-key="secret-key" aws-region="us-east-1" 
```
However, when I go into the Kinesis Video Streams and try to download a getClip, I get this error: 

**MissingCodecPrivateDataException
Missing codec private data in fragment for track 1.
Status code: 400**

The logs with GST_DEBUG=1 can be found at https://gist.github.com/vbbandeira/b15ec8af6986237a4cd7e382e4ede261
and the logs with GST_DEBUG=4 can be found at https://gist.github.com/vbbandeira/6bd4b7a014a69da5f46cd036eaf32aec

Can you guys please let me know what is going on there?

Or if possible, help me find the solution to this error.

Thanks!

sergiofigueras, 2021-09-30:
Hello there @disa6302 @prelich! Any thoughts on it? :)

vbbandeira, 2021-09-30:
Hello guys, for those looking for the same solution, I managed to make it work by replacing the "videomixer" which is deprecated by the "composer", below is an example of the command I used and it worked:

```
gst-launch-1.0 rtspsrc location="rtsp://password@192.168.0.1:554/cam/realmonitor?channel=1&subtype=0" short-header=TRUE ! decodebin ! videoconvert ! comp.sink_0 \
rtspsrc location="rtsp://password@192.168.0.2:554/cam/realmonitor?channel=1&subtype=0" short-header=TRUE ! decodebin ! videoconvert ! comp.sink_1 \
compositor name=comp sink_0::xpos=0 sink_1::xpos=1280 ! x264enc ! kvssink stream-name="test-stream" storage-size=512 access-key="access-key" secret-key="secret-key" aws-region="us-east-1" 
```

closed 2021-09-30 by vbbandeira