awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #759: gstreamer plugin in streaming audio and video don't woking

lidong-jimi, 2021-09-19:
I want to upload the audio and video captured by USB camera to the AWS Kinesis Video Streams. 
The platform is MacOS Big Sur 11.4
When I run command in the terminal
`gst-launch-1.0 -v avfvideosrc device-index=0 ! videoconvert ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 ! kvssink name=sink stream-name="" access-key="" secret-key="" osxaudiosrc device=89 ! audioconvert ! avenc_aac ! queue ! sink. `
The outputs in the terminal is normally

> 设置暂停管道 ...
> log4cplus:ERROR could not open file ./kvs_log_configuration
> INFO - createKinesisVideoClient(): Creating Kinesis Video Client
> 2021-09-19 19:12:43 [0x109479e00] INFO - heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
> 2021-09-19 19:12:43 [0x109479e00] INFO - heapInitialize(): Creating AIV heap.
> 2021-09-19 19:12:43 [0x109479e00] INFO - heapInitialize(): Heap is initialized OK
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - getSecurityTokenHandler invoked
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1632049963324980 Expiration: 0
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
> 2021-09-19 19:12:43 [0x109479e00] INFO - createDeviceResultEvent(): Create device result event.
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - clientReadyHandler invoked
> 2021-09-19 19:12:43 [0x109479e00] INFO - try creating stream
> 2021-09-19 19:12:43 [0x109479e00] INFO - Creating Kinesis Video Stream xx
> 2021-09-19 19:12:43 [0x109479e00] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): Kinesis Video Stream Info
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Stream name: xx
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Content type: video/h264,audio/aac 
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Max latency (100ns): 600000000
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Fragment duration (100ns): 20000000
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Key frame fragmentation: Yes
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Use frame timecode: Yes
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Absolute frame timecode: Yes
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Nal adaptation flags: 0
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Average bandwith (bps): 4194304
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Framerate: 100
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Buffer duration (100ns): 1200000000
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Replay duration (100ns): 400000000
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Store Pressure Policy: 1
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	View Overflow Policy: 1
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Segment UUID: NULL
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Frame ordering mode: 5
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): Track list
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Track id: 1
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Track name: kinesis_video
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Track cpd: NULL
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Track id: 2
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Track name: audio
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Codec id: A_AAC
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Track type: TRACK_INFO_TYPE_AUDIO
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - logStreamInfo(): 	Track cpd: NULL
> 2021-09-19 19:12:43 [0x109479e00] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
> 2021-09-19 19:12:43 [0x109479e00] INFO - writeHeaderCallback(): RequestId: 79436f28-465d-411a-a8fc-b05dbd296ba2
> 2021-09-19 19:12:43 [0x70000dc61000] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.631089919189E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:ap-northeast-1:005228416193:alias/aws/kinesisvideo","MediaType":"video/h264,audio/aac","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:ap-northeast-1:005228416193:stream/xx/1631089919189","StreamName":"xx","Version":"YQAsIfhsX77p0KcisZJv"}}
> 2021-09-19 19:12:43 [0x70000dc61000] INFO - describeStreamResultEvent(): Describe stream result event.
> 2021-09-19 19:12:43 [0x70000dc61000] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
> 2021-09-19 19:12:43 [0x70000dc61000] INFO - writeHeaderCallback(): RequestId: 32378a88-a21a-48c2-81b5-f9a46a146c0e
> 2021-09-19 19:12:44 [0x70000dce4000] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-96e4f3e9.kinesisvideo.ap-northeast-1.amazonaws.com"}
> 2021-09-19 19:12:44 [0x70000dce4000] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
> 2021-09-19 19:12:44 [0x70000dce4000] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000020, Next state: 0x0000000000000010
> 2021-09-19 19:12:44 [0x70000dce4000] DEBUG - getStreamingTokenHandler invoked
> 2021-09-19 19:12:44 [0x70000dce4000] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: 1632049964535075 Expiration: 18446744073709551615
> 2021-09-19 19:12:44 [0x70000dce4000] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
> 2021-09-19 19:12:44 [0x70000dce4000] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
> 2021-09-19 19:12:44 [0x70000dce4000] DEBUG - streamReadyHandler invoked
> 2021-09-19 19:12:44 [0x70000dce4000] Stream is ready
> 管道正在使用且不需要 PREROLL ...
> 管道被 PREROLLED ...
> 设置播放管道 ...
[... 195 lines cut ...]
	class : Audio/Sink
	caps  : audio/x-raw, format=F32LE, layout=interleaved, rate=44100, channels=2, channel-mask=0x0000000000000003
	        audio/x-raw, format={ (string)F64LE, (string)F64BE, (string)F32LE, (string)F32BE, (string)S32LE, (string)S32BE, (string)U32LE, (string)U32BE, (string)S24_32LE, (string)S24_32BE, (string)U24_32LE, (string)U24_32BE, (string)S24LE, (string)S24BE, (string)U24LE, (string)U24BE, (string)S20LE, (string)S20BE, (string)U20LE, (string)U20BE, (string)S18LE, (string)S18BE, (string)U18LE, (string)U18BE, (string)S16LE, (string)S16BE, (string)U16LE, (string)U16BE, (string)S8, (string)U8 }, layout=interleaved, rate=[ 1, 2147483647 ], channels=2, channel-mask=0x0000000000000003
	        audio/x-raw, format={ (string)F64LE, (string)F64BE, (string)F32LE, (string)F32BE, (string)S32LE, (string)S32BE, (string)U32LE, (string)U32BE, (string)S24_32LE, (string)S24_32BE, (string)U24_32LE, (string)U24_32BE, (string)S24LE, (string)S24BE, (string)U24LE, (string)U24BE, (string)S20LE, (string)S20BE, (string)U20LE, (string)U20BE, (string)S18LE, (string)S18BE, (string)U18LE, (string)U18BE, (string)S16LE, (string)S16BE, (string)U16LE, (string)U16BE, (string)S8, (string)U8 }, layout=interleaved, rate=[ 1, 2147483647 ], channels=1
	gst-launch-1.0 ... ! osxaudiosink device=77

Device found:

	name  : MacBook Pro麦
	class : Audio/Source
	caps  : audio/x-raw, format=F32LE, layout=interleaved, rate=44100, channels=1
[... cut ...]
	        audio/x-raw, format={ (string)F64LE, (string)F64BE, (string)F32LE, (string)F32BE, (string)S32LE, (string)S32BE, (string)U32LE, (string)U32BE, (string)S24_32LE, (string)S24_32BE, (string)U24_32LE, (string)U24_32BE, (string)S24LE, (string)S24BE, (string)U24LE, (string)U24BE, (string)S20LE, (string)S20BE, (string)U20LE, (string)U20BE, (string)S18LE, (string)S18BE, (string)U18LE, (string)U18BE, (string)S16LE, (string)S16BE, (string)U16LE, (string)U16BE, (string)S8, (string)U8 }, layout=interleaved, rate=48000, channels=1
	gst-launch-1.0 osxaudiosrc device=89 ! ...`

Please help! Thank you!

hassanctech, 2021-09-24:
It looks like you're not setting the KVS region.  By default it will use "us-west-2".  Did you check the console in that specific region?  Do you see a stream created?

lidong-jimi, 2021-09-26:
@hassanctech 
Thanks for you reply, 

Yes. I confirmed the specific region. But there's no video.

The kvssink element has the following required parameters:
  stream-name, 
  storage-size, 
  access-key, 
  secret-key,
  credential-path,

No region parameter.

How can I set it?

disa6302, 2021-09-28:
You can use this: `aws-region`.

Check this file for the available `kvssink` options: 
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp

lidong-jimi, 2021-10-03:
@disa6302 
Thank you so much. 
It's my wrong. I used the wrong access_ key and secret_ key.
awkward！！！😅

lidong-jimi, 2021-10-04:
@hassanctech @disa6302 
Please help me again! Thank you!
I want to customize the resolution and frame rate uploaded to kinesis video stream. The following code does not change the resolution and frame rate of RTSP. What is the correct writing method?

`gst-launch-1.0 rtspsrc location=rtsp://localhost:8554/mystream short-header=TRUE ! rtph264depay ! video/x-h264, width=640,height=480,framerate=15/1, format=avc,alignment=au ! kvssink stream-name="xx" storage-size=512 access-key="xx"  secret-key="xx  aws-region="xx"`

hassanctech, 2021-10-04:
Indeed so the section of the `capsfilter` you have specified here:  `video/x-h264, width=640,height=480,framerate=15/1` does not necessarily *set* those values.  For an rtsp stream to set parameters such as width and heigh you may need to specify them in url itself.  This is specific to your camera and the RTSP server that is implemented on the camera itself which is vending the frames.  You'll need to find out for your camera how to specify the height and width parameters.

lidong-jimi, 2021-10-04:
@hassanctech 
OK. Thank you very much!
I have another question.
For USB Camera, I want to change the resolution and frame rate.
What should I write on the command line?
`gst-launch-1.0 -v avfvideosrc device-index=0 ! videoconvert ! vtenc_h264_hw allow-frame-reordering=FALSE realtime=TRUE max-keyframe-interval=45 ! kvssink name=sink stream-name="" access-key="" secret-key="" osxaudiosrc device=89 ! audioconvert ! avenc_aac ! queue ! sink.`

hassanctech, 2021-10-04:
For `avfvideosrc` you should be able to specify a `capsfilter` after it to set the resolution and frame rate.  Try to do `gst-inspect-1.0 avfvideosrc` to get more information about it as well if you need.  You may need an `h264parse` element as well.  Did your about `rtspsrc` url work correctly?  We found that you need `h264parse` immediately after the `rtph264depay` element for it to work properly.

lidong-jimi, 2021-10-05:
@hassanctech 
Thank you!
I can work without h264parse.