awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #761: [QUESTION] resetConnection behavior

crcsylvia, 2021-09-22:
Hi KVS team,

Follow #750 
We have adjusted our StreamDefinition to 
```
max latency 35s	
buffer duration 45s	
replay duration 0s	
connection staleness 180s
```
And we didn't see staleness callback in these settings.
We observed long-period data loss (more than 10 minutes) in some periods.
(If we reset the process data loss will reduce to less than 1 min)

_**Q1. How can we tell if the stream is upload periodically in log?**_

One of the PutMedia Incoming Data gap observed from aws console monitoring:
(UTC 2021-09-21 23:24-2021-09-22 00:01)
Log for this period:
```
Sep 21 22:05:46.916| INFO|28998|DefaultCallbackProvider.cpp:318|putStreamResultEvent(): Put stream result event. New upload handle 304
Sep 21 22:05:46.972| INFO|28914|DefaultCallbackProvider.cpp:318|getStreamData(): Handle 303 waiting for last persisted ack with ts 16322619432500000
Sep 21 22:05:47.261| INFO|28914|DefaultCallbackProvider.cpp:318|getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 303
Sep 21 22:05:47.265| INFO|28914|DefaultCallbackProvider.cpp:318|postReadCallback(): Reported end-of-stream for stream FD9366-HV-53. Upload handle: 303
Sep 21 22:05:48.679| INFO|29020|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: dcbd291f-b4dc-299e-8579-70632789ac0c
Sep 21 22:44:47.716| INFO|29298|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: 68319be5-b667-4706-b6c9-f96eed9e6d98
Sep 21 22:44:47.741| INFO|29298|DefaultCallbackProvider.cpp:318|getStreamingEndpointResultEvent(): Get streaming endpoint result event.
Sep 21 22:44:50.751| INFO|29298|DefaultCallbackProvider.cpp:318|getStreamingTokenResultEvent(): Get streaming token result event.
Sep 21 22:44:50.754| WARN|29298|ProducerCallback.cpp:26|streamReady : stream handle = 522336
Sep 21 22:44:50.761| INFO|29298|DefaultCallbackProvider.cpp:318|putStreamResultEvent(): Put stream result event. New upload handle 305
Sep 21 22:44:50.803| INFO|29020|DefaultCallbackProvider.cpp:318|getStreamData(): Handle 304 waiting for last persisted ack with ts 16322642869810000
Sep 21 22:44:51.095| INFO|29020|DefaultCallbackProvider.cpp:318|getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 304
Sep 21 22:44:51.101| INFO|29020|DefaultCallbackProvider.cpp:318|postReadCallback(): Reported end-of-stream for stream FD9366-HV-53. Upload handle: 304
Sep 21 22:44:52.412| INFO|29320|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: f4351609-3845-0af0-adf1-74adacb94059
Sep 21 23:24:11.078| INFO|29320|DefaultCallbackProvider.cpp:318|getStreamData(): Handle 305 waiting for last persisted ack with ts 16322666498830000
Sep 21 23:24:11.408| INFO|29320|DefaultCallbackProvider.cpp:318|getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 305
Sep 21 23:24:11.410| INFO|29320|DefaultCallbackProvider.cpp:318|postReadCallback(): Reported end-of-stream for stream FD9366-HV-53. Upload handle: 305
Sep 21 23:24:12.242| INFO|29618|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: f73dc2e2-2a14-4832-8acd-8fc97d7c467c
Sep 21 23:24:12.272| INFO|29618|DefaultCallbackProvider.cpp:318|getStreamingEndpointResultEvent(): Get streaming endpoint result event.
Sep 21 23:24:12.274| INFO|29618|DefaultCallbackProvider.cpp:318|getStreamingTokenResultEvent(): Get streaming token result event.
Sep 21 23:24:12.276| WARN|29618|ProducerCallback.cpp:26|streamReady : stream handle = 522336
Sep 21 23:24:12.281| INFO|29618|DefaultCallbackProvider.cpp:318|putStreamResultEvent(): Put stream result event. New upload handle 306
Sep 21 23:24:13.944| INFO|29620|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: ce988e35-add2-171f-975c-e5eb0c23877e
Sep 22 00:01:40.290| INFO|29818|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: 5da8c59d-b925-460a-b244-367fc53e6da2
Sep 22 00:01:40.319| INFO|29818|DefaultCallbackProvider.cpp:318|getStreamingEndpointResultEvent(): Get streaming endpoint result event.
Sep 22 00:01:40.328| INFO|29818|DefaultCallbackProvider.cpp:318|getStreamingTokenResultEvent(): Get streaming token result event.
Sep 22 00:01:40.344| WARN|29818|ProducerCallback.cpp:26|streamReady : stream handle = 522336
Sep 22 00:01:40.348| INFO|29818|DefaultCallbackProvider.cpp:318|putStreamResultEvent(): Put stream result event. New upload handle 307
Sep 22 00:01:41.719| INFO|29620|DefaultCallbackProvider.cpp:318|getStreamData(): Handle 306 waiting for last persisted ack with ts 16322689004620000
Sep 22 00:01:42.185| INFO|29820|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: df15e551-d005-d4c1-86d1-91f745235966
Sep 22 00:02:17.333| WARN|29620|DefaultCallbackProvider.cpp:318|curlCompleteSync(): curl perform failed for url https://s-d022f287.kinesisvideo.us-east-1.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the last 30 seconds
Sep 22 00:02:17.345| WARN|29620|DefaultCallbackProvider.cpp:318|curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-d022f287.kinesisvideo.us-east-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIA4KF5KTT6BWYAYMA6/20210921/us-east-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=185112babfd71f1645bc21cc903a8946fdc31123b1236a0
Sep 22 00:02:17.353| WARN|29620|DefaultCallbackProvider.cpp:318|putStreamCurlHandler(): Stream with streamHandle 522336 uploadHandle 306 has exited without triggering end-of-stream. Service call result: 599
Sep 22 00:02:17.364| INFO|29620|DefaultCallbackProvider.cpp:318|kinesisVideoStreamTerminated(): Stream 0x7f860 terminated upload handle 306 with service call result 599.
Sep 22 00:02:17.385| WARN|29620|DefaultCallbackProvider.cpp:318|streamTerminatedEvent(): Last fragment with timestamp 16322689004620000 for upload handle 306 might not be fully persisted
Sep 22 00:04:09.354| INFO|29824|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: 5b709d6f-2615-4e47-b3ec-7d3ef7bf0dad
Sep 22 00:04:09.388| INFO|29824|DefaultCallbackProvider.cpp:318|getStreamingEndpointResultEvent(): Get streaming endpoint result event.
Sep 22 00:04:12.393| INFO|29824|DefaultCallbackProvider.cpp:318|getStreamingTokenResultEvent(): Get streaming token result event.
Sep 22 00:04:12.402| WARN|29824|ProducerCallback.cpp:26|streamReady : stream handle = 522336
```

Because of the gap of data lost in Q1, we did experients about calling stream->resetConnection(). (Try to handle stale with resetConnection.)
Observed result: after resetConnection called, it will receive http 408 error when putMedia and can not upload frame in a period.

_**Q2. Please help to explain the expected behavior after resetConnection called. Could we expect the streaming recovered in 30s (or other time range)?**_

Experiment:
[... 23 lines cut ...]
Sep 22 07:43:55.894| INFO|30575|DefaultCallbackProvider.cpp:318|putStreamResultEvent(): Put stream result event. New upload handle 1
Sep 22 07:43:57.272| INFO|30577|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: ebe3ab4a-e178-4fbd-b227-4bb740e206f8
Sep 22 07:44:50.188| WARN|30577|ProducerCallback.cpp:97|Reporting stream stale. Last ACK received 600330000
Sep 22 07:44:50.202| WARN|30577|DefaultCallbackProvider.cpp:318|postReadCallback(): Reported abort-connection for Upload handle: 1
Sep 22 07:44:50.225| WARN|30577|DefaultCallbackProvider.cpp:318|curlCompleteSync(): HTTP Error 408 : Response: (null)
Request URL: https://s-d022f287.kinesisvideo.us-east-1.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIA4KF5KTT6ICVD7MUJ/20210922/us-east-1/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fragment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=325c43c33884eadc14c4d31f6255629891194ad686524
Sep 22 07:44:51.559| INFO|30580|DefaultCallbackProvider.cpp:318|writeHeaderCallback(): RequestId: 4c312c6f-5431-4a1d-809e-26f0a58827b8
Sep 22 07:44:51.570| INFO|30580|DefaultCallbackProvider.cpp:318|getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[... cut ...]
Sep 22 07:44:53.296| WARN|30582|ProducerCallback.cpp:97|Reporting stream stale. Last ACK received 600330000
Sep 22 07:44:53.374| WARN|30582|ProducerCallback.cpp:97|Reporting stream stale. Last ACK received 600330000
Sep 22 07:44:53.402| WARN|30582|ProducerCallback.cpp:97|Reporting stream stale. Last ACK received 600320000
Sep 22 07:44:53.429| WARN|30582|ProducerCallback.cpp:97|Reporting stream stale. Last ACK received 600330000
Sep 22 07:44:53.459| WARN|30582|ProducerCallback.cpp:97|Reporting stream stale. Last ACK received 600330000
```

MushMal, 2021-09-22:
Try to set replay duration to at least a fragment worth. This is important as value of 0 means to not rollback at all and the re-streaming machinery then can "pogo-stick" trying to find a new restart point within the stream skipping the frames until future key-frame while the reset could be keep resetting and advancing the "current".

The other question is about what to do to "know" whether you are streaming and the best answer is to monitor the Fragment ACK callback and ensure you are getting Persisted ACKs (or, in case if you haven't enabled the persistence, then you can process Received ACKs). The ACKs will have timestamps indicating how far back the stream is ACK-ed against your latest frame.

crcsylvia, 2021-09-23:
After we adjust replay_duration, we don't see the same problem. Thank you!

Our new question is "Setting connection_staleness in StreamDescription to CONNECTION_STALENESS_DETECTION_SENTINEL(0)  to skip stale check" an option to avoid running stale code path? Compare to #750 which one is recommended?

Base on the comment in 
https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L988
```
  // Can be CONNECTION_STALENESS_DETECTION_SENTINEL to skip the check.
```

MushMal, 2021-09-23:
The sentinel would indeed disable the staleness all together. If you don't need to check for staleness you could use that value. However, if you still want to ensure you respond to staleness then you might want to chose some larger value than max latency. Again, this is due to the bug that will cause pogo-sticking with staleness state machine if you keep hitting that edge case.

closed 2021-09-28 by crcsylvia