awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #762: [QUESTION]

ragavi9798, 2021-09-22:
Hi,

I've built KVS SDK as Gstreamer Plugin for Windows and it is working fine with Access and Secret Keys. But when I try to provide Certificates (IoT Credential Provider Authentication), it's failing with error code : 0x4000000e([STATUS_DIRECTORY_ENTRY_STAT_ERROR](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html)) 

This is the command I'm using.

`​gst-launch-1.0 rtspsrc location="rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov"  ! rtph264depay ! h264parse ! kvssink stream-name="mystreamname" storage-size=512 iot-certificate="iot-certificate,endpoint=xxxxxxxx.credentials.iot.ap-south-1.amazonaws.com,cert-path=H:\mqtt\certificate.pem,key-path=H:\mqtt\private.pem.key,ca-path=H:\mqtt\cacert.pem,role-aliases=myrolealias" aws-region="ap-south-1"`

Error which I'm getting.
`ERROR - Unable to create Iot Credential provider. Error status: 0x4000000e
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstKvsSink:sink: Could not initialize supporting library.
Additional debug info:
/amazon-kinesis-video-streams-producer-sdk-cpp/src/gstreamer/gstkvssink.cpp(1375): gst_kvs_sink_change_state (): /GstPipeline:pipeline0/GstKvsSink:sink:
Failed to init kvs producer. Error: Unable to create Iot Credential provider. Error status: 0x4000000e
2020-11-02 23:45:00 [139930241742592] Setting pipeline to NULL ...
Freeing pipeline ...`

I'm not sure what does this error mean. Is there any issue with the path to certificate on windows?

Note: I tried the verification step given in AWS documentation.

`curl  --cert ./certificate.pem --key ./private.pem.key https://IOT_GET_CREDENTIAL_ENDPOINT/role-aliases/myrolealias/credentials --cacert ./cacert.pem`

This is working. It gives Accesss, Secret Access and Session token which I exported an Environment variable and tested streaming video using gst-launch command. It worked. Only issue is with IoT credential provider. Please help on the same.

ragavi9798, 2021-09-28:
I tried all combinations. Used local path to file, used absolute path to file, used path alone. But the issue is same. Also tried,
`H:\\mqtt\\certificate.pem,key-path=H:\\mqtt\\private.pem.key,ca-path=H:\\mqtt\\cacert.pem,role-aliases=myrolealias"`. This resulted in "couldn't read file error"

disa6302, 2021-09-28:
Please ensure the file is valid. Test out the application with default path to see if it works and that the file has the permissions to be read.

ragavi9798, 2021-09-29:
Yeah, so test the validity of the file, I used the below command which gives temporary credentials for the same. It's working fine and when I export the temporary credentials, I can stream the video. So, I hope the issue is with SDK's IoT credential provider compatibility for Windows.
`curl --cert .\certificate.pem --key .\private.pem.key https://IOT_GET_CREDENTIAL_ENDPOINT/role-aliases/myrolealias/credentials --cacert .\cacert.pem`

disa6302, 2021-09-29:
@ragavi9798 , 

The sample is available [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/windows.md). For visibility, this is an example. You might have to modify this based on camera configurations or your requirements:

`gst-launch-1.0 rtspsrc location="rtsp://YourCameraRtspUrl" short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name="iot-stream" iot-certificate="iot-certificate,endpoint=endpoint,cert-path=/path/to/certificate,key-path=/path/to/private/key,ca-path=/path/to/ca-cert,role-aliases=role-aliases"`