awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #766: [QUESTION] An error occurred when incoming cpd

Aylaasia-LiYiyang, 2021-09-28:
when calling the kinesis_video_stream->start(cpd) interface . I get error  "Failed to get the size of the buffer for hex decoding the codec private data with: 2"。 
Is there a problem with the input cpd?
How to generate it ?
[image]

hassanctech, 2021-09-28:
The error code you see 2 is `STATUS_INVALID_ARG` as defined here:

https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/common/include/com/amazonaws/kinesis/video/common/CommonDefs.h#L465

This is returned here in `hexDecode`:  https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/src/utils/src/Hex.c#L159

This error is thrown is the value is not properly hex encoded.  Perhaps you can print out your buffer and see what characters are wrong.

disa6302, 2021-09-28:
@Aylaasia-LiYiyang ,

What commit are you using? Can you reference [this](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/395) ticket for details?

Aylaasia-LiYiyang, 2021-09-29:
The hexencode is incorrect, I can see the video after I fix it。

closed 2021-09-29 by Aylaasia-LiYiyang