awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #768: [BUG] Error streaming H.265

daveisfera, 2021-09-30:
**Logging**
```
Progress: (request) Sent PLAY request
0:00:01.581412800    68 0x7f6f68081a30 INFO        GST_ELEMENT_PADS gstelement.c:913:gst_element_get_static_pad: no such pad 'sink' in element "udpsrc1"
0:00:01.581532800    68 0x7f6f68081a30 INFO               GST_EVENT gstevent.c:808:gst_event_new_caps: creating caps event application/x-rtcp
0:00:02.640976700    68 0x7f6f68081990 INFO               GST_EVENT gstevent.c:808:gst_event_new_caps: creating caps event application/x-rtcp
0:00:02.641060700    68 0x7f6f68081990 INFO               GST_EVENT gstevent.c:889:gst_event_new_segment: creating segment event time segment start=0:00:00.000000000, offset=0:00:00.000000000, stop=99:99:99.999999999, rate=1.000000, applied_rate=1.000000, flags=0x00, time=0:00:00.000000000, base=0:00:00.000000000, position 0:00:00.000000000, duration 99:99:99.999999999
0:00:06.587481700    68       0x83a770 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<rtpssrcdemux0> current PLAYING pending VOID_PENDING, desired next PAUSED
0:00:06.587618500    68       0x83a770 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<rtpssrcdemux0> completed state change to PAUSED
0:00:06.587668300    68       0x83a770 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<rtpssrcdemux0> notifying about state-changed PLAYING to PAUSED (VOID_PENDING pending)
0:00:06.587802100    68       0x83a770 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<manager> child 'rtpssrcdemux0' changed state to 3(PAUSED) successfully
0:00:06.588030100    68       0x83a770 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<rtpsession0> current PLAYING pending VOID_PENDING, desired next PAUSED
0:00:06.588124700    68       0x83a770 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<rtpsession0> completed state change to PAUSED
0:00:06.588163900    68       0x83a770 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<rtpsession0> notifying about state-changed PLAYING to PAUSED (VOID_PENDING pending)
0:00:06.588277300    68       0x83a770 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<manager> child 'rtpsession0' changed state to 3(PAUSED) successfully
0:00:06.588334300    68       0x83a770 INFO              GST_STATES gstelement.c:2394:gst_element_continue_state:<manager> committing state from PLAYING to PAUSED, pending READY, next READY
0:00:06.588387700    68       0x83a770 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<manager> notifying about state-changed PLAYING to PAUSED (READY pending)
0:00:06.588447900    68       0x83a770 INFO              GST_STATES gstelement.c:2401:gst_element_continue_state:<manager> continue state change PAUSED to READY, final READY
0:00:06.588551800    68       0x83a770 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<rtpssrcdemux0> current PAUSED pending VOID_PENDING, desired next READY
0:00:06.588592200    68       0x83a770 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<rtpssrcdemux0> completed state change to READY
0:00:06.588644200    68       0x83a770 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<rtpssrcdemux0> notifying about state-changed PAUSED to READY (VOID_PENDING pending)
0:00:06.588677700    68       0x83a770 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<manager> child 'rtpssrcdemux0' changed state to 2(READY) successfully
0:00:06.588700300    68       0x83a770 INFO              GST_STATES gstbin.c:2521:gst_bin_element_set_state:<rtpsession0> current PAUSED pending VOID_PENDING, desired next READY
0:00:06.588778000    68       0x83a770 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<rtpsession0> completed state change to READY
0:00:06.588896900    68       0x83a770 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<rtpsession0> notifying about state-changed PAUSED to READY (VOID_PENDING pending)
0:00:06.588930200    68       0x83a770 INFO              GST_STATES gstbin.c:2969:gst_bin_change_state_func:<manager> child 'rtpsession0' changed state to 2(READY) successfully
0:00:06.588962200    68       0x83a770 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<manager> completed state change to READY
0:00:06.589006100    68       0x83a770 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<manager> notifying about state-changed PAUSED to READY (VOID_PENDING pending)
0:00:06.589209900    68 0x7f6f680819e0 INFO                 basesrc gstbasesrc.c:2743:gst_base_src_loop:<udpsrc0> pausing after gst_base_src_get_range() = flushing
0:00:06.589300300    68 0x7f6f680819e0 INFO                    task gsttask.c:318:gst_task_func:<udpsrc0:src> Task going to paused
0:00:06.589388100    68       0x83a770 INFO              GST_STATES gstelement.c:2394:gst_element_continue_state:<udpsrc0> committing state from PLAYING to PAUSED, pending READY, next READY
0:00:06.589423100    68       0x83a770 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<udpsrc0> notifying about state-changed PLAYING to PAUSED (READY pending)
0:00:06.589530000    68       0x83a770 INFO              GST_STATES gstelement.c:2401:gst_element_continue_state:<udpsrc0> continue state change PAUSED to READY, final READY
0:00:06.589749000    68 0x7f6f680819e0 INFO                    task gsttask.c:320:gst_task_func:<udpsrc0:src> Task resume from paused
0:00:06.590093900    68       0x83a770 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<udpsrc0> completed state change to READY
0:00:06.590162400    68       0x83a770 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<udpsrc0> notifying about state-changed PAUSED to READY (VOID_PENDING pending)
0:00:06.590413000    68 0x7f6f68081a30 INFO                 basesrc gstbasesrc.c:2743:gst_base_src_loop:<udpsrc1> pausing after gst_base_src_get_range() = flushing
0:00:06.590569500    68       0x83a770 INFO              GST_STATES gstelement.c:2394:gst_element_continue_state:<udpsrc1> committing state from PLAYING to PAUSED, pending READY, next READY
0:00:06.590609800    68       0x83a770 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<udpsrc1> notifying about state-changed PLAYING to PAUSED (READY pending)
0:00:06.590653400    68 0x7f6f68081a30 INFO                    task gsttask.c:318:gst_task_func:<udpsrc1:src> Task going to paused
0:00:06.590685900    68       0x83a770 INFO              GST_STATES gstelement.c:2401:gst_element_continue_state:<udpsrc1> continue state change PAUSED to READY, final READY
0:00:06.591235900    68 0x7f6f68081a30 INFO                    task gsttask.c:320:gst_task_func:<udpsrc1:src> Task resume from paused
0:00:06.591450300    68       0x83a770 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<udpsrc1> completed state change to READY
0:00:06.591500400    68       0x83a770 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<udpsrc1> notifying about state-changed PAUSED to READY (VOID_PENDING pending)
0:00:06.639706700    68       0x83a770 INFO              GST_STATES gstelement.c:2419:gst_element_continue_state:<udpsrc0> completed state change to NULL
0:00:06.639753000    68       0x83a770 INFO              GST_STATES gstelement.c:2324:_priv_gst_element_state_changed:<udpsrc0> notifying about state-changed READY to NULL (VOID_PENDING pending)
0:00:06.639808600    68       0x83a770 INFO        GST_ELEMENT_PADS gstpad.c:2071:gst_pad_unlink: unlinking udpsrc0:src(0x8a7260) and manager:recv_rtp_sink_0(0x7f6f68014060)
0:00:06.639906000    68       0x83a770 INFO        GST_ELEMENT_PADS gstpad.c:2125:gst_pad_unlink: unlinked udpsrc0:src and manager:recv_rtp_sink_0
0:00:06.639926900    68       0x83a770 INFO           GST_PARENTAGE gstbin.c:1822:gst_bin_remove_func:<rtspsrc0> removed child "udpsrc0"
0:00:06.639960900    68       0x83a770 INFO         GST_REFCOUNTING gstelement.c:2994:gst_element_dispose:<udpsrc0> 0x7f6f68072b40 dispose
0:00:06.639988000    68       0x83a770 INFO        GST_ELEMENT_PADS gstelement.c:785:gst_element_remove_pad:<udpsrc0> removing pad 'src'
[... cut ...]
2. Change camera configuration to H.265
3. Verify it working in VLC
4. Run produce with H.265 parsing

**Expected behavior**
H.265 should work like H.264

**SDK version number**
Current version available in the docker image (how do I check this?)

**Open source building**
Using the docker image at `546150905175.dkr.ecr.us-west-2.amazonaws.com/kinesis-video-producer-sdk-cpp-amazon-linux`

**Platform (please complete the following information):**
 - OS: Docker Desktop 4.0.1 on macOS 11.6
 - Processor: 2.4 GHz 8-Core Intel Core i9

disa6302, 2021-10-29:
There seems to be an issue with your source. I see this message in the logs:

```WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource..```

Are you sure your source is sending out frames? Can you test this with another sink like filesink? This will help pinpoint if it is a KVS issue or an issue with your pipeline

daveisfera, 2021-10-29:
That same log happens when using a H.264 stream with kvssink and I can view the stream in VLC with both H.264 and H.265.

I tried replacing `kvssink` with `filesink` and it wrote a file but I couldn't open it with VLC or Quicktime, so how can I test that?

daveisfera, 2021-10-29:
Played with it a bit and realized that I had to add `matroskamux` to the pipeline so the video would be encoded and now it's working. So I have been able to have `filesink` work with both H.264 and H.265.

And here's the output that was the same in both cases (other than the duration that it prints at the end):
```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://admin:<password>@192.168.5.53
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
Additional debug info:
gstrtspsrc.c(4904): gst_rtspsrc_reconnect (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.
^Chandling interrupt.
Interrupt: Stopping pipeline ...
Execution ended after 0:00:25.759780000
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

disa6302, 2021-10-29:
When you say it is working, do you mean you see streaming? Does it stream with kvssink? 
If not, can you attach debug logs of the SDK alone (without GST_DEBUG=<x>)

daveisfera, 2021-10-29:
No, it's still not working with `kvssink` when using H.265. `filesink` works with both H.264 and H.265, but `kvssink` only works with H.264 and is still running into the same problem with H.265.

daveisfera, 2021-10-29:
Here's the output when running `kvssink` with the same H.265 stream that works with `filesink`:
```
log4cplus:ERROR could not open file ./kvs_log_configuration
Setting pipeline to PAUSED ...
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2021-10-29 19:17:16 [140713391449920] INFO - heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
2021-10-29 19:17:16 [140713391449920] INFO - heapInitialize(): Creating AIV heap.
2021-10-29 19:17:16 [140713391449920] INFO - heapInitialize(): Heap is initialized OK
2021-10-29 19:17:16 [140713391449920] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-10-29 19:17:16 [140713391449920] DEBUG - getSecurityTokenHandler invoked
2021-10-29 19:17:16 [140713391449920] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1635535036479816200 Expiration: 0
2021-10-29 19:17:16 [140713391449920] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2021-10-29 19:17:16 [140713391449920] INFO - createDeviceResultEvent(): Create device result event.
2021-10-29 19:17:16 [140713391449920] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2021-10-29 19:17:16 [140713391449920] DEBUG - clientReadyHandler invoked
2021-10-29 19:17:16 [140713391449920] INFO - try creating stream
2021-10-29 19:17:16 [140713391449920] INFO - Creating Kinesis Video Stream DaveTest
2021-10-29 19:17:16 [140713391449920] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Stream name: DaveTest 
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Content type: video/h265 
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Max latency (100ns): 600000000
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Fragment duration (100ns): 20000000
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Key frame fragmentation: Yes
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Use frame timecode: Yes
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Absolute frame timecode: Yes
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Nal adaptation flags: 0
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Average bandwith (bps): 4194304
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Framerate: 25
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Buffer duration (100ns): 1200000000
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Replay duration (100ns): 400000000
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Store Pressure Policy: 1
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	View Overflow Policy: 1
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Segment UUID: NULL
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Frame ordering mode: 0
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): Track list
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Track id: 1
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Track name: kinesis_video
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Codec id: V_MPEGH/ISO/HEVC
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
2021-10-29 19:17:16 [140713391449920] DEBUG - logStreamInfo(): 	Track cpd: NULL
2021-10-29 19:17:16 [140713391449920] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-10-29 19:17:16 [140713391449920] INFO - writeHeaderCallback(): RequestId: c2f3138b-7825-4422-9188-8f384d5ce8df
2021-10-29 19:17:16 [140712677934848] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.63485781205E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:us-east-1:286047911707:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:286047911707:stream/DaveTest/1634857812050","StreamName":"DaveTest","Version":"D5nZhjYZciFuvi3TPYR8"}}
2021-10-29 19:17:16 [140712677934848] INFO - describeStreamResultEvent(): Describe stream result event.
2021-10-29 19:17:16 [140712677934848] WARN - describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2021-10-29 19:17:16 [140712677934848] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2021-10-29 19:17:16 [140712677934848] INFO - writeHeaderCallback(): RequestId: 6a05d782-c537-4060-8599-b66a2cd0bed8
2021-10-29 19:17:17 [140712669542144] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-808fdb44.kinesisvideo.us-east-1.amazonaws.com"}
[... cut ...]
2021-10-29 19:17:52 [140713391449920] INFO - freeKinesisVideoClient(): Freeing Kinesis Video Client
2021-10-29 19:17:53 [140713391449920] DEBUG - curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
2021-10-29 19:17:53 [140713391449920] DEBUG - freeKinesisVideoClientInternal(): Total allocated memory 0
2021-10-29 19:17:54 [140713391449920] WARN - curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down
2021-10-29 19:17:54 [140713391449920] %                                                                                
```

disa6302, 2021-10-29:
My bad I clsoed the issue with a comment. Will investigate the logs and get back. Thank you.

daveisfera, 2021-10-29:
Thanks and let me know if there's anything I can help with.

On a semi-related note, could/should the `Dropping frame with flag: 1024` log message be at `ERROR` or at least `WARN` level?

daveisfera, 2021-11-09:
Anything I can do to help further test/debug the problem?

daveisfera, 2021-11-10:
Is there any sort of bug bounty or such that we could do to help prioritize this getting fixed?

MushMal, 2021-11-10:
Just looking at the returned error: 0x5200005e - this is defined in https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html 

<html>
<body>

0x5200005e | STATUS_ACK_ERR_FRAGMENT_DURATION_REACHED | The maximum fragment duration was reached. | The max fragment duration is defined in Producer SDK Limits. This error indicates that there are either very low frames per second or there are no key frames to create manageable duration fragments. Check the encoder settings and ensure that key frames are being produced properly at the regular intervals.
-- | -- | -- | --

</body>
</html>

This is an indicator that your encoder is not producing regular key-frames. I am not familiar with the encoder settings but can you readup a little on it and set it to cut an Idr frame every 2-4 seconds?

daveisfera, 2021-11-10:
Yes, it's generating an I-frame every 2 seconds just like with the H.264 setup. The resolution and frame rate are identical with both the H.264 and H.265 setup. VLC shows the stream and using `filesink` records it to a file and then QuickTime can display it, so it appears to be a standard/compliant stream of video and there's something going wrong in the handling of the stream in the `kvssink` plugin.

MushMal, 2021-11-10:
KVS is attempting to have no assumption about the underlying streams - as such, it has a concept of a key-frame and non-key-frame instead of Idr and non-Idr frames. There are areas where things can be encoding specific. Those are in the area of CPD extraction/processing and in the backend, the playback (HLS/DASH/Console playback) but other than that there is no assumption made. In your case, the issue here is to 1) ensure the I-frames are produced at a regular intervals from the encoder and 2) the I-frames are properly interpreted by the KVS sink as key-frames. 

If you are sure your source (encoder) produces proper GoPs then the issue must be how kvs sink GStreamer plugin interprets the flags. 

Perhaps someone from KVS team can chime whether they have a test based on some H265 (perhaps software) encoder that successfully gets ingested?

daveisfera, 2021-11-11:
Yes, there appears to be a key frame every 2 seconds, because here's the output of checking that on the output of `filesink` with `ffprobe`:
```
$ ffprobe -show_frames /tmp/dlj/output.mp4 | grep -i key_frame | uniq -c
Input #0, matroska,webm, from '/tmp/dlj/output.mp4':
  Metadata:
    encoder         : GStreamer matroskamux version 1.10.4
    creation_time   : 2021-11-11T00:27:47.000000Z
  Duration: 00:00:04.84, start: 5.355000, bitrate: 1057 kb/s
    Stream #0:0(eng): Video: hevc (Main), yuv420p(tv), 1280x720, SAR 1:1 DAR 16:9, 30 fps, 30 tbr, 1k tbn, 1k tbc (default)
    Metadata:
      title           : Video
      1 key_frame=1
     59 key_frame=0
      1 key_frame=1
     59 key_frame=0
      1 key_frame=1
     25 key_frame=0
```
You can see that there are 59 frames between each key-frame and the video is at 30 fps, so that should be a key-frame every 2 seconds.

We really want to get this working, so I'm glad to help out with anything I can to figure this out. Also, is there anyone that's worked on the `kvssink` code that could be prioritized through a bug bounty or something similar?

MushMal, 2021-11-11:
If I were to guess the flag value of 1024 is GST_BUFFER_FLAG_HEADER

Do you mind replacing the following line:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/2b9436dbc7152b2f0374b795c4705e91fa9a5460/src/gstreamer/gstkvssink.cpp

`(GST_BUFFER_FLAG_IS_SET(buf, GST_BUFFER_FLAG_HEADER) && (!GST_BUFFER_PTS_IS_VALID(buf) || !GST_BUFFER_DTS_IS_VALID(buf)));`

with the following line:

`(GST_BUFFER_FLAG_IS_SET(buf, GST_BUFFER_FLAG_HEADER) && (!GST_BUFFER_PTS_IS_VALID(buf) && !GST_BUFFER_DTS_IS_VALID(buf)));`

Basically, stating that if the buffer contains a header and no PTS and DTS then skip it

daveisfera, 2021-11-11:
Ok, I got it to work by upgrading the base image to `rockylinux/rockylinux:8` or `fedora:35`. It appears the fix is from the newer version of `gstreamer` that's being used there (i.e. `amazonlinux:2` is pretty old now and I wish there was an `amazonlinux:3` based on RHEL 8 😢 )

disa6302, 2021-11-11:
@daveisfera ,

Apologies for not getting back sooner. But glad updating the base image helped. I am resolving this ticket. Feel free to reach out if you face any more issues. Thank you!

closed 2021-11-11 by disa6302

daveisfera, 2021-11-12:
Is there a plan to make the official images be something other than `amazonlinux:2`? If not, is there a recommended base image?

disa6302, 2021-11-12:
@daveisfera ,
There is not a plan to support anything other than amazonlinux:2 at the moment.

Other than amazonlinux, I would look into ubuntu latest image and see how that works out. I have personally not tested it out though for this SDK.

daveisfera, 2021-11-12:
Is support for H.265 going to be removed from the documentation then? This is what's pointed to in the official AWS documentation and it doesn't actually support H.265 so it seems that at least a disclaimer should be added.

daveisfera, 2021-11-15:
Here's the status of images that I tested with:
`Fedora 34`: drops with flag 1536 on H.264
`Fedora 35`: won't start
`Debug buster`: works with H.265
`Debian bullseye`: drops with flag 1536 on H.264
`Rocky Linux 8`: works with H.265