awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #780: [BUG] fragment-duration appears to not be used

daveisfera, 2021-11-01:
**Logging**
Output when running with `fragment-duration=1000` and it outputs that the values was loaded (i.e. `DEBUG - logStreamInfo(): 	Fragment duration (100ns): 10000000`)
```
INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
DEBUG - logStreamInfo(): Kinesis Video Stream Info
DEBUG - logStreamInfo(): 	Stream name: DaveTest 
DEBUG - logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME 
DEBUG - logStreamInfo(): 	Content type: video/h264 
DEBUG - logStreamInfo(): 	Max latency (100ns): 600000000
DEBUG - logStreamInfo(): 	Fragment duration (100ns): 10000000
DEBUG - logStreamInfo(): 	Key frame fragmentation: Yes
DEBUG - logStreamInfo(): 	Use frame timecode: Yes
DEBUG - logStreamInfo(): 	Absolute frame timecode: Yes
DEBUG - logStreamInfo(): 	Nal adaptation flags: 0
DEBUG - logStreamInfo(): 	Average bandwith (bps): 4194304
DEBUG - logStreamInfo(): 	Framerate: 25
DEBUG - logStreamInfo(): 	Buffer duration (100ns): 1200000000
DEBUG - logStreamInfo(): 	Replay duration (100ns): 400000000
DEBUG - logStreamInfo(): 	Connection Staleness duration (100ns): 600000000
DEBUG - logStreamInfo(): 	Store Pressure Policy: 1
DEBUG - logStreamInfo(): 	View Overflow Policy: 1
DEBUG - logStreamInfo(): 	Segment UUID: NULL
DEBUG - logStreamInfo(): 	Frame ordering mode: 0
DEBUG - logStreamInfo(): Track list
DEBUG - logStreamInfo(): 	Track id: 1
DEBUG - logStreamInfo(): 	Track name: kinesis_video
DEBUG - logStreamInfo(): 	Codec id: V_MPEG4/ISO/AVC
DEBUG - logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
DEBUG - logStreamInfo(): 	Track cpd: NULL
DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
```

**Describe the bug**
`fragment-duration` appears to be loaded but unused.

**To Reproduce**
Steps to reproduce the behavior:
1. Run with `fragment-duration` set to a value other than the default
2. Get an HLS URL for the stream
3. Watch the stream and observe that fragments are always 3 seconds long

**Expected behavior**
`fragment-duration` would be used and would set the fragment duration for the HLS URL

**SDK version number**
3.1.0

**Open source building**
N/A (using current Docker setup)

**Platform (please complete the following information):**
N/A (using current Docker setup)

MushMal, 2021-11-01:
fragment-duration is used to set StreamInfo.StreamCaps.fragmentDuration property. This property is used only when the fragmentation mode is based on non-fey-frame fragmentation. By default, the SDK is configured to "cut" fragments whenever the upstream marks a fragment as a key-frame. Assuming your scenario is based on GStreamer then the upstream encoder owns the setting of the I-frames which then get marked as key-frames which is what is driving the fragmentation so in your case the property doesn't have any impact. For video, you would rather set the fragment duration in the encoder itself and leave the SDK to cut the fragments on the key-frames.

Fragment duration is used in streams which don't really have a concept of GoP - for example an AAC audio or similar. For that, all frames are key-frames and the SDK cuts a fragment on the first key-frame after the duration has elapsed.

Hope this helps.

daveisfera, 2021-11-02:
I initially thought that might be the case, but the HLS stream has a fragment size of 3 seconds and the current key frame setting for the encoder is 2 seconds, so it feels like something must be happening in `kvssink` or `gstreamer`. Anything I can do to help figure out what's going on here?

daveisfera, 2021-11-09:
Yes, I verified that it's using the fragment duration from the encoder and it was just an issue with the way that the HLS player was handling its buffer that made it appear to be 3 seconds. Might be worth putting the info that you referenced in the documentation so future people won't have questions like I did.

closed 2021-11-09 by daveisfera

MushMal, 2021-11-09:
Some of the information is available in the documentations here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/master/docs

Would be great if people could do PRs for documentation too.