awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #781: kvssink with a different stream-name not working

clogwog, 2021-11-03:
I have an IoT Greengrass device (with name smartdvr-1423019132001 )
on the device I stream video to kinesis using the kvssink gstreamer plugin like:

`gst-launch-1.0 .... video source ... ! kvssink name=bob storage-size=512 iot-certificate="iot-certificate,endpoint=xxxxxxxxx.credentials.iot.ap-southeast-2.amazonaws.com,cert-path=/greengrass/v2/thingCert.crt,key-path=/greengrass/v2/privKey.key,ca-path=/greengrass/v2/rootCA.pem,role-aliases=KvsCameraIoTRoleAlias" aws-region="ap-southeast-2" log-config="/etc/mtdata/kvssink-log.config" stream-name=smartdvr-1423019132001`

and with the instructions on how to setup impersonation via the certificate it uses the following alias: KvsCameraIoTRoleAlias which points to

KVSCameraCertificateBasedIAMRole :
```
{
"Version": "2012-10-17",
"Statement": [
  {
    "Sid": "VisualEditor0",
    "Effect": "Allow",
    "Action": [
      "kinesisvideo:PutMedia",
      "kinesisvideo:GetDataEndpoint",
      "kinesisvideo:DescribeStream",
      "kinesisvideo:TagStream",
      "kinesisvideo:CreateStream"
    ],
    "Resource": "arn:aws:kinesisvideo:::stream/${credentials-iot:ThingName}/*"
  }
]
}
```

..this works.. and it creates a smartdvr-1423019132001 kinesis video stream and is streaming the video.

now i want to change to have multiple streams (may have more than 1 camera that i want to stream)
so i've changed the stream-name to smartdvr-1423019132001-video1
and changed the impersonation policy to
`"Resource": "arn:aws:kinesisvideo:::stream/${credentials-iot:ThingName}-*/*"`

but i'm getting errors in the kvs.log file like:

```
[DEBUG] [03-11-2021 09:28:24:182.950 GMT] Creating IoT auth callbacks.
[ERROR] [03-11-2021 09:28:24:413.866 GMT] blockingCurlCall(): Curl call response failed with http status 403
[ERROR] [03-11-2021 09:28:24:418.225 GMT] Unable to create Iot Credential provider. Error status: 0x15000011
```

if i switch it back it starts working again.

also tried
`"Resource": "*"`
still the same
any suggestions where i can look ?

note: this is a repost of https://forums.aws.amazon.com/thread.jspa?threadID=347747&tstart=0 as i noticed that most support for this is being given here instead of on the aws forum.

clogwog, 2021-11-04:
diving into the code a bit, it looks like the stream-name is being passed into 

`createCurlIotCredentialProviderWithTime` as the thingName 

https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/blob/a8091d1a140fbdcb260ba3c8302aa2e4712d2d91/src/source/IotAuthCallback.c#L39

is there a reason for this ? 

would this whole 1 stream per thingname be solved by adding an extra 'thingname' to the paramters and using that to get the authentication token, then using stream-name for..you know.. the name of the stream ?

then letting the policy on the backend limit access to whatever you like.. but don't enforce the stream-name at this stage ?

disa6302, 2021-11-15:
Closing since PR is merged.

closed 2021-11-15 by disa6302

clogwog, 2022-05-26:
@Coldplayer1995 i added a new parameter called iot-thing-name 
If you specify it with your exact thing-name then you can use anything else as your stream-name. I think there is an example in the description of the pull request.