awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #782: Building Gstreamer with -DBUILD_STATIC=ON not working

karshee, 2021-11-04:
We are trying to build Gstreamer with the following cmake command - 
cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=OFF -DBUILD_STATIC=ON -DALIGNED_MEMORY_MODEL=ON -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DBUILD_OPENSSL_PLATFORM=linux-armv4

and then when we run make - we are given an error.

We are building on an arm64 architecture but we are building inside a docker container with raspian image - FROM resin/rpi-raspbian:stretch .

** Logging **

[image]

** Any design considerations/constraints **

We are building this on a custom yocto platform on a modalai VOXL that will be streaming via kinesis and communicating with IOTCore, all inside a container - the end goal is to build with -DBUILD_STATIC=ON so we can just extract the gst-launch-1.0 binary file. We can then include just this in our dockerfile for a light and compact container. It would be great if you have the binary files already or were open to any other suggestions to achieve this as the whole SDK is too large to run on the IOT device.

disa6302, 2021-11-08:
@karshee ,

Can you ensure STATIC build works on a standard platform first?
I see it is unable to find any of the libraries in the paths.

karshee, 2021-11-09:
Hey, 
since we have got it working with -DBUILD_STATIC=ON - we first encountered errors of samples failing so we commented out the following lines in CMakeLists.txt 
[image]

and the video stream worked with the following outcome after the build -
[image]
 
but when we try to run the gst-launch-1.0 binary file outside the container we are missing dependencies, we tried copying everything in the build folder over too but x264enc and kvssink are still seen as missing. 
[image]

karshee, 2021-11-09:
we used this CMAKE command - 
cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=OFF -DBUILD_STATIC=ON -DALIGNED_MEMORY_MODEL=ON -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON -DBUILD_OPENSSL_PLATFORM=linux-armv4

disa6302, 2021-11-11:
No element "x264enc" indicates that the gstreamer plugins have not been installed successfully. To verify it is a linking issue, can you try to build the SDK without  `-DBUILD_STATIC=ON` and see if it works?

karshee, 2021-11-15:
Yes it works without -DBUILD_STATIC=ON but our goal is to minimize the overall size of the docker container. We wanted to use -DBUILD_STATIC=ON to try and build the binary files then extract them and just use them in our own dockerfile. Will this work ? or do you have any other suggestions to minimizing the size of the overall kinesis sdk docker container ? as were running it on an IOT device so require minimal storage usage.

disa6302, 2022-04-24:
As long as your docker container has access to the library paths, it would work. I am closing this now. Please feel free to reach out if you have any more questions.

closed 2022-04-24 by disa6302