awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #786: [BUG]

savidha777, 2021-11-08:
Hi,

For RPI 3B+ platform, we are integrated KVS SDK and we used kvssink plugin to push the live streaming content into AWS KVS service.

Used below pipeline for our validation:
**gst-launch-1.0 v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,width=640,height=480 ! omxh264enc ! h264parse ! kvssink stream-name="test_camera_stream" storage-size=100 access-key="XXXXXXXXX" secret-key="YYYYYYYYY" aws-region="ZZZZZZZZZ"**

With above pipeline we able to view and record RPI streaming content in KVS service with 2.0.2 KVS SDK tagged revision, but with latest 3.1.1/3.2.0 tagged kvs sdk revision we are getting below core-dump issue after seeing single frame in kvs service.

Core-dump:
#0  0x76ce6588 in removeChainedBlock (pAivHeap=pAivHeap@entry=0x72e10, pBlock=pBlock@entry=0x75d005c5)
    at /usr/src/debug/kvs/3.1.1-r0/git/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/heap/src/AivHeap.c:879
#1  0x76ce67ba in aivHeapFree (pHeap=0x72e10, handle=<optimized out>)
    at /usr/src/debug/kvs/3.1.1-r0/git/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/heap/src/AivHeap.c:325
#2  0x76cd41b0 in viewItemRemoved (pContentView=<optimized out>, customData=<optimized out>, pViewItem=0x74c58, currentRemoved=0)
    at /usr/src/debug/kvs/3.1.1-r0/git/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/client/src/Client.c:1621
#3  0x76ce4de0 in contentViewTrimTail (pContentView=0x74b78, itemIndex=<optimized out>)
    at /usr/src/debug/kvs/3.1.1-r0/git/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/view/src/ContentView.c:587
#4  0x76cd82d8 in streamFragmentPersistedAck (pKinesisVideoStream=pKinesisVideoStream@entry=0x4ec10, timestamp=<optimized out>,
    pUploadHandleInfo=pUploadHandleInfo@entry=0x2d9b0)
    at /usr/src/debug/kvs/3.1.1-r0/git/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/client/src/Stream.c:2535
#5  0x76cda6fa in streamFragmentAckEvent (pKinesisVideoStream=pKinesisVideoStream@entry=0x4ec10, uploadHandle=<optimized out>,
    pFragmentAck=pFragmentAck@entry=0x54d20)
    at /usr/src/debug/kvs/3.1.1-r0/git/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/client/src/StreamEvent.c:875
#6  0x76ce55a4 in processParsedAck (pKinesisVideoStream=0x4ec10)
    at /usr/src/debug/kvs/3.1.1-r0/git/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/client/src/AckParser.c:278
#7  0x76ce587e in parseFragmentAck (pKinesisVideoStream=pKinesisVideoStream@entry=0x4ec10, uploadHandle=<optimized out>,
    ackSegment=ackSegment@entry=0x8157c "{\"EventType\":\"PERSISTED\",\"FragmentTimecode\":3691104463,\"FragmentNumber\":\"91343852333184962997674104545554217272768984578\"}\n\r\n", ackSegmentSize=<optimized out>, ackSegmentSize@entry=123)
    at /usr/src/debug/kvs/3.1.1-r0/git/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/client/src/AckParser.c:239
#8  0x76cd3d5e in kinesisVideoStreamParseFragmentAck (streamHandle=<optimized out>, uploadHandle=<optimized out>,
    ackSegment=0x8157c "{\"EventType\":\"PERSISTED\",\"FragmentTimecode\":3691104463,\"FragmentNumber\":\"91343852333184962997674104545554217272768984578\"}\n\r\n",
    ackSegmentSize=123) at /usr/src/debug/kvs/3.1.1-r0/git/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/client/src/Client.c:1417
#9  0x76ccd5ba in postWriteCallback (pBuffer=<optimized out>, size=<optimized out>, numItems=123, customData=0x7bc50)
    at /usr/src/debug/kvs/3.1.1-r0/git/dependency/libkvscproducer/kvscproducer-src/src/source/Response.c:593
#10 0x76a809d8 in ?? () from /usr/lib/libcurl.so.4

Requesting to give your input for this core-dump issue.

disa6302, 2021-11-08:
@savidha777 , 
Thank you for the coredump. Would you mind attaching the verbose log file as well? Running `export AWS_KVS_LOG_LEVEL=1` will generate the logs.

savidha777, 2021-11-09:
Hi @disa6302 

Please find attached 3.1.1.KVS SDK core-dump log file.
[kvs_core_dump.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/7503988/kvs_core_dump.txt)

disa6302, 2021-11-15:
@savidha777 , 

I have not been able to reproduce this on Mac. And the stack trace does not seem to make sense either. A few clarifications:

1. Were there any changes/firmware updates that were done on the Raspberry Pi apart from updating the producer SDK?
2. Is your device memory aligned? Can you try building with `-DALIGNED_MEMORY_MODEL=TRUE` flag?

savidha777, 2021-11-16:
HI @disa6302 

Thanks for your support

After enabled this ALIGNED_MEMORY_MODEL flag, we able to view streaming content in AWS service without any issue.

disa6302, 2021-11-16:
Great. I am resolving this issue now. Feel free to reopen/open a new issue if you have any more questions.

closed 2021-11-16 by disa6302