awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #787: WARNING: erroneous pipeline: no element "kvssink"

gorleabhinav, 2021-11-09:
I have cross compiled amazon kinesis cpp sdk for aarch64 system. the gstreamer pipeline failed to execute with some warnings.

**gstreamer pipeline:** 
gst-launch-1.0 rtspsrc location="rtsp url" short-header=TRUE ! rtph264depay ! video/x-h264, format=avc,alignment=au ! kvssink stream-name="my_stream_name" storage-size=512 access-key="my_access_key" secret-key="my_secrete_key" aws-region="ap-south-1"

**error logs:** 
(gst-plugin-scanner:2467): GStreamer-WARNING **: 15:05:05.692: Failed to load plugin '/root/amazon-kinesis-video-streams-producer-sdk-cpp/build/libgstkvssink.so': /lib/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /root/amazon-kinesis-video-streams-producer-sdk-cpp/build/libgstkvssink.so)

(gst-plugin-scanner:2467): GStreamer-WARNING **: 15:05:05.714: Failed to load plugin '/root/amazon-kinesis-video-streams-producer-sdk-cpp/build/libKinesisVideoProducer.so': /lib/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /root/amazon-kinesis-video-streams-producer-sdk-cpp/build/libKinesisVideoProducer.so)

(gst-plugin-scanner:2469): GStreamer-WARNING **: 15:05:05.811: Failed to load plugin '/root/amazon-kinesis-video-streams-producer-sdk-cpp/build/libKinesisVideoProducer.so': /lib/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /root/amazon-kinesis-video-streams-producer-sdk-cpp/build/libKinesisVideoProducer.so)

(gst-plugin-scanner:2470): GStreamer-WARNING **: 15:05:05.934: Failed to load plugin '/usr/lib/gstreamer-1.0/libcurl.so': /lib/libc.so.6: version `GLIBC_2.28' not found (required by /usr/lib/gstreamer-1.0/libcurl.so)

(gst-plugin-scanner:2470): GStreamer-WARNING **: 15:05:05.943: Failed to load plugin '/usr/lib/gstreamer-1.0/libgstkvssink.so': /lib/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /usr/lib/gstreamer-1.0/libgstkvssink.so)

(gst-plugin-scanner:2470): GStreamer-WARNING **: 15:05:05.947: Failed to load plugin '/usr/lib/gstreamer-1.0/libKinesisVideoProducer.so': /lib/libstdc++.so.6: version `GLIBCXX_3.4.26' not found (required by /usr/lib/gstreamer-1.0/libKinesisVideoProducer.so)
WARNING: erroneous pipeline: no element "kvssink"

i set the below environment variable path as per my system.
export GST_PLUGIN_PATH=`pwd`/build
export LD_LIBRARY_PATH=`pwd`/open-source/local/lib

unable to understand what is going wrong? 
any help would appreciated. let me know if need more info.

disa6302, 2021-11-12:
@gorleabhinav , 

It is possible libstdc++6 is not available on your machine? Can you try installing libstdc++6 and see if that helps?

disa6302, 2021-11-16:
Closing due to no response. Feel free to reopen if you have any updates.

closed 2021-11-16 by disa6302

gorleabhinav, 2021-11-17:
the problem with the libstdc++6 library version, previously the system has libstdc++6.0.24 but i updated it with libstdc++6.0.27. now issue get resolved. Thanks.