awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #788: cross compilation error

gorleabhinav, 2021-11-09:
**config.log file results**
```{r}
make[3]: Entering directory '/home/adminuser/workspace/git/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/liblog4cplus/build/src/project_log4cplus'
make[3]: Leaving directory '/home/adminuser/workspace/git/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/liblog4cplus/build/src/project_log4cplus'
this is the am__doit target
configure:3277: $? = 0
configure:3296: result: yes (GNU style)
configure:3366: checking for gcc
configure:3393: result: /home/adminuser/workspace/toolchain/amlogic/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-gcc
configure:3622: checking for C compiler version
configure:3631: /home/adminuser/workspace/toolchain/amlogic/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-gcc --version >&5
aarch64-linux-gnu-gcc (Linaro GCC 7.3-2018.05) 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701]
Copyright (C) 2017 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
configure:3642: $? = 0
configure:3631: /home/adminuser/workspace/toolchain/amlogic/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-gcc -v >&5
Using built-in specs.
COLLECT_GCC=/home/adminuser/workspace/toolchain/amlogic/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-gcc
COLLECT_LTO_WRAPPER=/home/adminuser/workspace/toolchain/amlogic/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/../libexec/gcc/aarch64-linux-gnu/7.3.1/lto-wrapper
Target: aarch64-linux-gnu
Configured with: '/home/tcwg-buildslave/workspace/tcwg-make-release/builder_arch/amd64/label/tcwg-x86_64-build/target/aarch64-linux-gnu/snapshots/gcc.git~linaro-7.3-2018.05/configure' SHELL=/bin/bash --with-mpc=/home/tcwg-buildslave/workspace/tcwg-make-release/builder_arch/amd64/label/tcwg-x86_64-build/target/aarch64-linux-gnu/_build/builds/destdir/x86_64-unknown-linux-gnu --with-mpfr=/home/tcwg-buildslave/workspace/tcwg-make-release/builder_arch/amd64/label/tcwg-x86_64-build/target/aarch64-linux-gnu/_build/builds/destdir/x86_64-unknown-linux-gnu --with-gmp=/home/tcwg-buildslave/workspace/tcwg-make-release/builder_arch/amd64/label/tcwg-x86_64-build/target/aarch64-linux-gnu/_build/builds/destdir/x86_64-unknown-linux-gnu --with-gnu-as --with-gnu-ld --disable-libmudflap --enable-lto --enable-shared --without-included-gettext --enable-nls --with-system-zlib --disable-sjlj-exceptions --enable-gnu-unique-object --enable-linker-build-id --disable-libstdcxx-pch --enable-c99 --enable-clocale=gnu --enable-libstdcxx-debug --enable-long-long --with-cloog=no --with-ppl=no --with-isl=no --disable-multilib --enable-fix-cortex-a53-835769 --enable-fix-cortex-a53-843419 --with-arch=armv8-a --enable-threads=posix --enable-multiarch --enable-libstdcxx-time=yes --enable-gnu-indirect-function --with-build-sysroot=/home/tcwg-buildslave/workspace/tcwg-make-release/builder_arch/amd64/label/tcwg-x86_64-build/target/aarch64-linux-gnu/_build/sysroots/aarch64-linux-gnu --with-sysroot=/home/tcwg-buildslave/workspace/tcwg-make-release/builder_arch/amd64/label/tcwg-x86_64-build/target/aarch64-linux-gnu/_build/builds/destdir/x86_64-unknown-linux-gnu/aarch64-linux-gnu/libc --enable-checking=release --disable-bootstrap --enable-languages=c,c++,fortran,lto --build=x86_64-unknown-linux-gnu --host=x86_64-unknown-linux-gnu --target=aarch64-linux-gnu --prefix=/home/tcwg-buildslave/workspace/tcwg-make-release/builder_arch/amd64/label/tcwg-x86_64-build/target/aarch64-linux-gnu/_build/builds/destdir/x86_64-unknown-linux-gnu
Thread model: posix
gcc version 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701] (Linaro GCC 7.3-2018.05) 
configure:3642: $? = 0
configure:3631: /home/adminuser/workspace/toolchain/amlogic/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-gcc -V >&5
aarch64-linux-gnu-gcc: error: unrecognized command line option '-V'
aarch64-linux-gnu-gcc: fatal error: no input files
compilation terminated.
configure:3642: $? = 1
configure:3631: /home/adminuser/workspace/toolchain/amlogic/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-gcc -qversion >&5
aarch64-linux-gnu-gcc: error: unrecognized command line option '-qversion'; did you mean '--version'?
aarch64-linux-gnu-gcc: fatal error: no input files
compilation terminated.
configure:3642: $? = 1
configure:3662: checking whether the C compiler works
configure:3684: /home/adminuser/workspace/toolchain/amlogic/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-gcc    conftest.c  >&5
configure:3688: $? = 0
configure:3736: result: yes
configure:3739: checking for C compiler default output file name
configure:3741: result: a.out
configure:3747: checking for suffix of executables
configure:3754: /home/adminuser/workspace/toolchain/amlogic/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-gcc -o conftest    conftest.c  >&5
configure:3758: $? = 0
configure:3780: result: 
configure:3802: checking whether we are cross compiling
configure:3810: /home/adminuser/workspace/toolchain/amlogic/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu/bin/aarch64-linux-gnu-gcc -o conftest    conftest.c  >&5
configure:3814: $? = 0
configure:3821: ./conftest
/home/adminuser/workspace/git/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/liblog4cplus/build/src/project_log4cplus/configure: line 3823: ./conftest: cannot execute binary file: Exec format error
configure:3825: $? = 126
configure:3832: error: in `/home/adminuser/workspace/git/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/liblog4cplus/build/src/project_log4cplus':
configure:3834: error: cannot run C compiled programs.
If you meant to cross compile, use `--host'.
See `config.log' for more details
```
**Description**
i am trying to cross compile kinesis sdk for aarch64 platform. if you see in the logs conftest.c file cross compiled for aarch64(target system) but trying to execute on x86_64(host system).

**To Reproduce**
Steps to reproduce the behavior:
1. clone the amazon-kinesis-video-streams-producer-sdk-cpp
2. [download the cross-compiler tool-chain](http://releases.linaro.org/components/toolchain/binaries/7.3-2018.05/aarch64-linux-gnu/gcc-linaro-7.3.1-2018.05-x86_64_aarch64-linux-gnu.tar.xz)
3. set the CC and CXX path according to the cross-compiler tool-chain.
4. run  the command: cmake .. -DBUILD_GSTREAMER_PLUGIN=ON
5. See error

How to resolve this cross compilation issue ?

wilx, 2021-11-11:
This looks like an issue in the CMake script which tries to compile log4cplus using its Autotools build system. The [`CONFIGURE_COMMAND`](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/CMake/Dependencies/liblog4cplus-CMakeLists.txt#L6) variable is set up only for non-cross compilation. If you want to cross compile, you will need to give it also `--build=x86_64-pc-linux-gnu --host=aarch64-linux-gnu` parameters.

gorleabhinav, 2021-11-21:
it worked. 
Thanks @wilx.

closed 2021-11-21 by gorleabhinav