awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #795: WARNING: erroneous pipeline: no element "kvssink"

THESOCK1999, 2021-11-16:
I am trying to stream my webcam to aws kinesis using gstreamer.

I am using the command "gst-launch-1.0 -v v4l2src device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! x264enc  bframes=0 key-int-max=45 bitrate=500 tune=zerolatency ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name=.. storage-size=128 access-key=.. secret-key=.."

It then prints the error "WARNING: erroneous pipeline: no element "kvssink"".

I have looked at other solutions on here but I cannot get it to work on my device.

disa6302, 2021-11-16:
Please ensure the following 2 things:

1. KVS Sink Library exists in the build directory
2. You have run `export GST_PLUGIN_PATH=<path to build folder>`

Please follow the instructions in the README. You can search other past issues as well to get some answers. Closing.

closed 2021-11-16 by disa6302

THESOCK1999, 2021-11-17:
I keep seeing a reference to the build directory. What does this refer to?

THESOCK1999, 2021-11-18:
Could someone advise me on how I can find/create the build directory?

disa6302, 2021-11-18:
Again, please follow the readme. 

1. Create a build directory in the root of the repository
2. Move to build directory
3. Run `cmake.. -DBUILD_GSTREAMER_PLUGIN=ON`
4. Run `make`
5. Export the library path
6. Run the sample

THESOCK1999, 2021-11-18:
When you say root repository, what are you referring to?

THESOCK1999, 2021-11-18:
I am very new to these programs, and I would really appreciate the help

THESOCK1999, 2021-11-18:
Does this refer to the /usr directory or is it something else?

THESOCK1999, 2021-11-18:
Do you need to create your own repository or are you referring to an existing repository?

THESOCK1999, 2021-11-19:
Could somebody please assist me in understanding this?

THESOCK1999, 2021-11-19:
I would really appreciate some help on the matter as I cannot find any information to assist me with the instructions that you have given

niyatim23, 2021-11-19:
By root of the directory, we mean the root directory of this GitHub repository on your system. The file structure should be like 
`amazon-kinesis-video-streams-producer-sdk-cpp/build` where 
`amazon-kinesis-video-streams-producer-sdk-cpp` is the folder created when you cloned the repository. Then, you `cd` into the build directory and perform steps 3 to 6

THESOCK1999, 2021-11-22:
I did every step like you advised and yet when I run the command "gst-inspect-1.0 kvssink" I get the response "No such element or plugin 'kvssink'" are there any other steps that I can follow to ensure that I didn't do something wrong?