awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #799: Stuck with RTSP Audio + Video

luisberga, 2021-11-22:
Hello! 

Below is my gts command: 

`gst-launch-1.0 rtspsrc name=src location="rtsp://192.168.14.5:5554/playlist.m3u" short-header=true do-timestamp=true ! rtph264depay ! h264parse ! kvssink name=sink stream-name="name" access-key="key" secret-key="secret" aws-region="sa-east-1" src. ! rtpmpadepay ! mpg123audiodec ! audioresample ! audioconvert ! voaacenc ! aacparse ! queue ! sink.`

however, i'm getting this error: 

`...Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:src/GstUDPSink:udpsink3: Error sending UDP packets
Additional debug info:
gstmultiudpsink.c(726): gst_multiudpsink_send_messages (): /GstPipeline:pipeline0/GstRTSPSrc:src/GstUDPSink:udpsink3:
client 192.168.15.3:0, reason: Error sending message: Invalid argument
INFO - kinesisVideoStreamFormatChanged(): Stream format changed.
2021-11-22 18:26:31 [140587632580352] WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:src: Delayed linking failed.
Additional debug info:
./grammar.y(506): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstRTSPSrc:src:
failed delayed linking some pad of GstRTSPSrc named src to some pad of GstRtpPcmaDepay named rtppcmadepay0
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:src/GstUDPSink:udpsink1: Error sending UDP packets
Additional debug info:
gstmultiudpsink.c(726): gst_multiudpsink_send_messages (): /GstPipeline:pipeline0/GstRTSPSrc:src/GstUDPSink:udpsink1:
client 192.168.14.5:0, reason: Error sending message: Invalid argument
`

And I've got stuck in this part... Anyone could help me? 

Thanks!!

hassanctech, 2021-11-22:
Can you please post which version of `kvssink` you are using?  In addition as a first debugging step have you tried to change the sink to something like filesink and see if that works?  This way we can help to narrow down if your issue is upstream near your source or truly with the plugin. I would first try the simplest pipeline and once you get it working try to add more of what you need.  Also are you sure what is coming from the source is in fact h264 encoded?

luisberga, 2021-11-22:
Hello! 

I'm using version 1.0. 

When I try just: 

`gst-launch-1.0 rtspsrc name=src location=rtsp://192.168.15.3:5554/playlist.m3u short-header=TRUE ! rtph264depay ! h264parse ! video/x-h264, format=avc,alignment=au ! kvssink name=sink stream-name=Teste1 storage-size=512 access-key=AKIATY2QOGFD6DC4W2PB secret-key=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 aws-region=sa-east-1 `

The video works like a charm, but just the video... without audio. 

I did try with filesink, but i got this error: 

`gst-launch-1.0 rtspsrc name=src location=rtsp://192.168.15.3:5554/playlist.m3u short-header=TRUE ! rtph264depay ! h264parse ! video/x-h264, format=avc,alignment=au ! filesink name=sink stream-name=Teste1 storage-size=512 access-key=AKIATY2QOGFD6DC4W2PB secret-key=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 aws-region=sa-east-1 

WARNING: erroneous pipeline: no property "stream-name" in element "sink"
`

Yes, what's coming from the source is h264 (the encoder that the source uses is OMX.google.h264.encoder)

hassanctech, 2021-11-22:
The latter it is because `filesink` does not have those parameters you only need to supply it `location` parameter which is where to write the file.

hassanctech, 2021-11-22:
Try that and let's see if you get audio kn the file.  Then we can debug the audio part.

hassanctech, 2021-11-22:
Also sorry I should have been more specific for the version, which version of the producer CPP SDK you're using to build the `kvssink` plugin?

luisberga, 2021-11-22:
Hello! 

Since i'm streaming to Kinesis stream, how can I use the `filesink`? I mean... How can I pass he `stream-name` with `filesink`? 

I'm using the last release (3.2.0). 

Thanks!

luisberga, 2021-11-22:
The strange part that the filesink produces the output file... But I can't open. Maybe the format is not h264?

luisberga, 2021-11-23:
I saw in VLC, the stream information is: 

[image]

luisberga, 2021-11-23:
I did include `mpegtsmux = mux` and succeed to record the video from rtsp source... But how can I do it to misc the audio too? 

`gst-launch-1.0 rtspsrc name=src location=rtsp://192.168.15.3:5554/playlist.m3u ! rtph264depay ! h264parse ! mpegtsmux name=mux ! filesink location="test.ts"`

luisberga, 2021-11-23:
Hello! 

I suceed to misc audio and video: 

`gst-launch-1.0 rtspsrc name=src location=rtsp://192.168.15.3:5554/playlist.m3u short-header=TRUE ! queue ! rtph264depay ! h264parse ! kvssink name=sink stream-name=test storage-size=512 access-key=key secret-key=key aws-region=sa-east-1 src. ! queue ! rtpmp4gdepay ! aacparse ! sink.`

However, I'm getting a lot of Warnings. I think that these warning can cause to my stream to get stuck in some points, frozing and start to loading to start again (resulting in delay).

`WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:src/GstUDPSink:udpsink1: Error sending UDP packets
Additional debug info:
gstmultiudpsink.c(726): gst_multiudpsink_send_messages (): /GstPipeline:pipeline0/GstRTSPSrc:src/GstUDPSink:udpsink1:
client 192.168.15.3:0, reason: Error sending message: Invalid argument
[DEBUG] [23-11-2021 01:03:32:778.782 GMT] postReadCallback(): Wrote 14456 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG] [23-11-2021 01:03:32:839.333 GMT] postReadCallback(): Wrote 22526 bytes to Kinesis Video. Upload stream handle: 0
[WARN ] [23-11-2021 01:03:32:851.661 GMT] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 5125166961 presentation timestamp: 16376294148540335
[WARN ] [23-11-2021 01:03:33:006.363 GMT] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 5125597686 presentation timestamp: 16376294148883004
[DEBUG] [23-11-2021 01:03:33:140.054 GMT] postReadCallback(): Wrote 35945 bytes to Kinesis Video. Upload stream handle: 0
[WARN ] [23-11-2021 01:03:33:213.371 GMT] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 5128383837 presentation timestamp: 16376294151100114
[DEBUG] [23-11-2021 01:03:33:249.056 GMT] Kinesis Video client and stream metrics
	>> Overall storage byte size: 536870912
	>> Available storage byte size: 536465842
	>> Allocated storage byte size: 405070
	>> Total view allocation byte size: 576080
	>> Total streams elementary frame rate (fps): 4908
	>> Total streams transfer rate (bps): 1110096 (1084 Kbps)
	>> Current view duration (ms): 89
	>> Overall view duration (ms): 2887
	>> Current view byte size: 16868
	>> Overall view byte size: 400606
	>> Current elementary frame rate (fps): 30.0155
	>> Current transfer rate (bps): 1110096 (1084 Kbps)
[WARN ] [23-11-2021 01:03:33:263.550 GMT] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 5129207318 presentation timestamp: 16376294152380004
`

Do you know how to fix that?

hassanctech, 2021-11-23:
The error you're seeing `0x30000005` means `STATUS_CONTENT_VIEW_INVALID_TIMESTAMP` and looking at your DTS and PTS values they are not in the correct scale, please correct your timestamps that should resolve this issue.

luisberga, 2021-11-23:
Good... how can I change my timestamp? xD

luisberga, 2021-11-23:
When I use this: 

`gst-launch-1.0 rtspsrc name=src location=rtsp://192.168.15.3:5554/playlist.m3u ! queue ! rtph264depay ! h264parse ! mpegtsmux name=mux ! filesink location="test.ts" src. ! queue ! rtpmp4gdepay ! aacparse ! mux.`

The video and audio work like a charm! However, when I try to send to Kinesis, with this pipe: 

`gst-launch-1.0 rtspsrc name=src location=rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mp4 short-header=TRUE ! queue ! rtph264depay ! h264parse ! mpegtsmux name=mux ! kvssink name=sink stream-name=test storage-size=512 access-key=key secret-key=key aws-region=sa-east-1 src. ! queue ! rtpmp4gdepay ! aacparse ! mux.`

I got this error:

`WARNING: erroneous pipeline: could not link mux to kvssink0`

hassanctech, 2021-11-23:
Can you run with `GST_DEBUG=5` ?

hassanctech, 2021-11-23:
I examined your timestamps the diff between the successive pts timestamps is:

853300
853400
400400
824200
1249300
420000
1372700
707200

It looks like your encoder is not consistent these diffs should roughly be the same value this will cause errors because the frames have durations so when a fragment ends if you try to insert a frame with a time code overlapping you will see other errors (I do not see indication of that right now) but it will most likely happen with an encoder which is not consistent.

You cannot send mux'd video to kvssink it will accept raw h264 (either annex b or avcc) so if you delete `mpegtsmux name=mux` from the pipeline with `kvssink` it should work.

luisberga, 2021-11-23:
Hi! 

With `GST_DEBUG=5`:

```
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2021-11-23 17:18:04 [139871915054144] INFO - heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
2021-11-23 17:18:04 [139871915054144] INFO - heapInitialize(): Creating AIV heap.
2021-11-23 17:18:04 [139871915054144] INFO - heapInitialize(): Heap is initialized OK
2021-11-23 17:18:04 [139871915054144] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-11-23 17:18:04 [139871915054144] DEBUG - getSecurityTokenHandler invoked
2021-11-23 17:18:04 [139871915054144] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1637698684874218140 Expiration: 0
2021-11-23 17:18:04 [139871915054144] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2021-11-23 17:18:04 [139871915054144] INFO - createDeviceResultEvent(): Create device result event.
2021-11-23 17:18:04 [139871915054144] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2021-11-23 17:18:04 [139871915054144] DEBUG - clientReadyHandler invoked
2021-11-23 17:18:04 [139871915054144] INFO - try creating stream
2021-11-23 17:18:04 [139871915054144] INFO - Creating Kinesis Video Stream Teste1
2021-11-23 17:18:04 [139871915054144] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Stream name: Teste1 
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Streaming type: STREAMING_TYPE_REALTIME 
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Content type: video/h264,audio/aac 
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Max latency (100ns): 600000000
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Fragment duration (100ns): 20000000
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Key frame fragmentation: Yes
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Use frame timecode: Yes
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Absolute frame timecode: Yes
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Nal adaptation flags: 0
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Average bandwith (bps): 4194304
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Framerate: 100
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Buffer duration (100ns): 1200000000
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Replay duration (100ns): 400000000
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Connection Staleness duration (100ns): 600000000
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Store Pressure Policy: 1
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  View Overflow Policy: 1
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Segment UUID: NULL
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Frame ordering mode: 5
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo(): Track list
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Track id: 1
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Track name: kinesis_video
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Codec id: V_MPEG4/ISO/AVC
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Track type: TRACK_INFO_TYPE_VIDEO
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Track cpd: NULL
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Track id: 2
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Track name: audio
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Codec id: A_AAC
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Track type: TRACK_INFO_TYPE_AUDIO
2021-11-23 17:18:04 [139871915054144] DEBUG - logStreamInfo():  Track cpd: NULL
2021-11-23 17:18:04 [139871915054144] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2021-11-23 17:18:04 [139871915054144] INFO - writeHeaderCallback(): RequestId: 9c1d4e0d-f78a-450d-b4ee-eca7afeb7e7d
2021-11-23 17:18:04 [139871345846016] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.637259545601E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:sa-east-1:259477352775:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:sa-east-1:259477352775:stream/Teste1/1637259545601","StreamName":"test","Version":"dasdasdqwd222"}}
2021-11-23 17:18:04 [139871345846016] INFO - describeStreamResultEvent(): Describe stream result event.
2021-11-23 17:18:04 [139871345846016] WARN - describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2021-11-23 17:18:04 [139871345846016] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000020
2021-11-23 17:18:04 [139871345846016] INFO - writeHeaderCallback(): RequestId: 4f35e68a-f8f5-4dd1-8439-5dd26460ea18
[... cut ...]
        >> Current elementary frame rate (fps): 24.4999
        >> Current transfer rate (bps): 27377848 (26736 Kbps)
2021-11-23 17:18:10 [139871337453312] DEBUG - postReadCallback(): Wrote 3774 bytes to Kinesis Video. Upload stream handle: 1
```

Yes, if i delete `mpegtsmux name=mux` it works, but I got the `0x30000005` error... What should I do, in this case? Since my video stream has this infos: 

[image]

Which codec did you suggest and how can I implement? Sorry, I'm newbie in this audio/video world =(

hassanctech, 2021-11-23:
Yes we're back to the original problem you have to figure out your video source and why it's not producing frames with the proper pts values and why they are all over the place and not consistent.  If it's 24fps your frames should be ~approx 42ms apart. Here it looks like you're feeding from some publicly available test video source, what is your actual production use case where will the video come from, if the video source there doesn't have this issue then we might be wasting time debugging why this public source doesn't have correct pts/dts values.

One way to correct it is to have an `appsink` element then in your code you implement a callback (you will not be able to do this with gst-launch you have to write some code to do it) which modifies the pts values and then an `appsrc` to pick up the changed timestamp in the buffer, the callback would be trivial all it does is set a correct pts.  That's ok if you want to test but really your source needs to produce the correct pts value.  You can probably use ffmpeg to rewrite those values as well.

luisberga, 2021-11-24:
Hi Hassan. 

Unfortunately, I don't have control of my video source... How can I use ffmpeg to rewrite those values?

Thanks

hassanctech, 2021-11-29:
There is another issue, I looked at your more detailed logs and I see failures with reason:  `FRAMES_MISSING_FOR_TRACK` you can see documentation here explaining this error:  https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/API_dataplane_PutMedia.html

This will happen if you have both audio/video tracks and one of the tracks is missing frames for a fragment.  This will be rejected by the backend which is what is happening in your case as shown in the logs.

Unfortunately we cannot provide support how to rewrite those values but if you search ffmpeg or GStreamer forums you should be able to get help from there.

hassanctech, 2021-11-29:
Also please make sure you're using the latest release 3.2.0:  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/releases

luisberga, 2021-11-29:
Thank you, Hassan! 

I'm using the last version.

Appreciated.

closed 2021-11-29 by luisberga