awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #800: [BUG] Doesn't start with current versions of gstreamer

daveisfera, 2021-11-30:
**Logging**
Fedora 29 with 1.14.4 (this version of gstreamer works with Debian Buster):
```
Progress: (connect) Connecting to rtsp://admin:<password>@192.168.5.91/profile3/media.smp
GLib-GIO-Message: 01:24:19.588: Using the 'memory' GSettings backend.  Your settings will not be saved or shared with other applications.
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (request) SETUP stream 1
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
0:00:01.174766500     8 0x562439d35a00 WARN                     bin gstbin.c:2787:gst_bin_do_latency_func:<pipeline0> did not really configure latency of 0:00:00.000000000
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
0:00:01.175375100     8 0x7f057c0a8a80 FIXME                default gstutils.c:3981:gst_pad_create_stream_id_internal:<fakesrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
0:00:01.175402800     8 0x7f057c0a8b70 FIXME                default gstutils.c:3981:gst_pad_create_stream_id_internal:<fakesrc1:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
Progress: (request) Sent PLAY request
0:00:06.208421500     8 0x562439ba9320 WARN                 rtspsrc gstrtspsrc.c:5423:gst_rtspsrc_reconnect:<rtspsrc0> warning: Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
Additional debug info:
gstrtspsrc.c(5423): gst_rtspsrc_reconnect (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.
0:00:06.417056400     8 0x7f057c0a8e30 WARN                 default grammar.y:510:gst_parse_no_more_pads:<rtspsrc0> warning: Delayed linking failed.
0:00:06.417219400     8 0x7f057c0a8e30 WARN                 default grammar.y:510:gst_parse_no_more_pads:<rtspsrc0> warning: failed delayed linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstRtpH264Depay named rtph264depay0
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Delayed linking failed.
Additional debug info:
./grammar.y(510): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
failed delayed linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstRtpH264Depay named rtph264depay0
EOS on shutdown enabled -- waiting for EOS after Error
Waiting for EOS...
0:00:06.418495700     8 0x562439ba9320 WARN                 rtspsrc gstrtspsrc.c:5649:gst_rtspsrc_loop:<rtspsrc0> error: Internal data stream error.
0:00:06.418543100     8 0x562439ba9320 WARN                 rtspsrc gstrtspsrc.c:5649:gst_rtspsrc_loop:<rtspsrc0> error: streaming stopped, reason not-linked (-1)
ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Internal data stream error.
Additional debug info:
gstrtspsrc.c(5649): gst_rtspsrc_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
streaming stopped, reason not-linked (-1)
0:00:31.498202100     8 0x7f055c003e80 WARN                 rtspsrc gstrtspsrc.c:3151:on_timeout:<rtspsrc0> source a12de4a5, stream a12de4a5 in session 1 timed out
0:00:31.545092300     8 0x7f055c003d90 WARN                 rtspsrc gstrtspsrc.c:3151:on_timeout:<rtspsrc0> source 20c66031, stream 20c66031 in session 0 timed out
```

Fedora 31 and Fedora 32 with 1.16.2 (Fedora 30 has 1.16 and has the same sort of error):
```
Progress: (connect) Connecting to rtsp://admin:<password>@192.168.5.91/profile3/media.smp
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
0:00:03.574829600     8 0x563c82e7a8c0 WARN                  udpsrc gstudpsrc.c:1445:gst_udpsrc_open:<udpsrc0> warning: Could not create a buffer of requested 524288 bytes (Operation not permitted). Need net.admin privilege?
0:00:03.574885600     8 0x563c82e7a8c0 WARN                  udpsrc gstudpsrc.c:1455:gst_udpsrc_open:<udpsrc0> have udp buffer of 212992 bytes while 524288 were requested
Progress: (request) SETUP stream 0
0:00:03.589912600     8 0x563c82e7a8c0 WARN                  udpsrc gstudpsrc.c:1445:gst_udpsrc_open:<udpsrc2> warning: Could not create a buffer of requested 524288 bytes (Operation not permitted). Need net.admin privilege?
0:00:03.589959000     8 0x563c82e7a8c0 WARN                  udpsrc gstudpsrc.c:1455:gst_udpsrc_open:<udpsrc2> have udp buffer of 212992 bytes while 524288 were requested
Progress: (request) SETUP stream 1
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
0:00:03.606490400     8 0x563c82d63ea0 WARN                     bin gstbin.c:2786:gst_bin_do_latency_func:<pipeline0> did not really configure latency of 0:00:00.000000000
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
0:00:03.607275500     8 0x7fa4a40830c0 FIXME                default gstutils.c:3980:gst_pad_create_stream_id_internal:<fakesrc0:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
0:00:03.607318700     8 0x7fa4a4083060 FIXME                default gstutils.c:3980:gst_pad_create_stream_id_internal:<fakesrc1:src> Creating random stream-id, consider implementing a deterministic way of creating a stream-id
Progress: (request) Sent PLAY request
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
Additional debug info:
gstrtspsrc.c(5768): gst_rtspsrc_reconnect (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.
0:00:08.633012100     8 0x563c82e7a8c0 WARN                 rtspsrc gstrtspsrc.c:5768:gst_rtspsrc_reconnect:<rtspsrc0> warning: Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.
0:00:08.837401000     8 0x7fa4a40832a0 WARN                 default grammar.y:506:gst_parse_no_more_pads:<rtspsrc0> warning: Delayed linking failed.
0:00:08.837446000     8 0x7fa4a40832a0 WARN                 default grammar.y:506:gst_parse_no_more_pads:<rtspsrc0> warning: failed delayed linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstRtpH264Depay named rtph264depay0
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Delayed linking failed.
Additional debug info:
[... 140 lines cut ...]
WORKDIR /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/
RUN cmake3 .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=OFF -DCMAKE_VERBOSE_MAKEFILE:BOOL=ON && \
    make

FROM fedora:35

RUN yum install -y \
    gstreamer1-plugins-base \
    gstreamer1-plugins-good \
    gstreamer1-plugins-bad-free \
    gstreamer1-plugins-ugly-free \
    libcurl \
    log4cplus \
    openssl-libs \
    && \
    yum clean all

ENV GST_PLUGIN_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/:$GST_PLUGIN_PATH

WORKDIR /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/
COPY --from=0 /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/ .
COPY cmd.sh log_config ./

CMD ["./cmd.sh"]
```

disa6302, 2021-12-06:
@daveisfera , 

Thank you for reporting this @daveisfera . We will look into what needs to be done to get this updated.

daveisfera, 2021-12-18:
To try and isolate the problem a bit more, I built `gstreamer` from source on Debian Bullseye and 1.16.3 is working just fine, but 1.18.0 doesn't work and doesn't log anything, but then 1.18.5 doesn't work either but outputs an error.

There's only a few tags for 1.17, but I can't build 1.17.1 and 1.17.2 has the same problem as 1.18.0 and then 1.17.90 is basically just 1.18.0, so that didn't reveal much other than I assume the tag numbers mean that it was a change made very early in the development of 1.17 that caused the problem.

Here's the logs from 1.18.5 with `GST_DEBUG=3`:
```
0:00:06.581641656     7 0x55fec649b120 WARN                 rtspsrc gstrtspsrc.c:5886:gst_rtspsrc_reconnect:<rtspsrc0> warning: Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.
WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
Additional debug info:
../gst/rtsp/gstrtspsrc.c(5886): gst_rtspsrc_reconnect (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Could not receive any UDP packets for 5.0000 seconds, maybe your firewall is blocking it. Retrying using a tcp connection.
Redistribute latency...
0:00:07.139854649     7 0x55fec635aa70 WARN                     bin gstbin.c:2811:gst_bin_do_latency_func:<pipeline0> did not really configure latency of 0:00:00.000000000
0:00:07.140734066     7 0x55fec635aa70 WARN                     bin gstbin.c:2811:gst_bin_do_latency_func:<pipeline0> did not really configure latency of 0:00:00.000000000
Redistribute latency...
0:00:07.232383789     7 0x7fbd2402a5e0 FIXME           rtph265depay gstrtph265depay.c:1284:gst_rtp_h265_depay_process:<rtph265depay0> Assuming DONL field is not present
... <DONL log repeated a lot>
INFO - kinesisVideoStreamFormatChanged(): Stream format changed.
2021-12-18 05:20:23 [140450061088512] WARN - Dropping frame with flag: 1600
2021-12-18 05:20:23 [140450061088512] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000040, Next state: 0x0000000000000080
2021-12-18 05:20:23 [140450061088512] INFO - putStreamResultEvent(): Put stream result event. New upload handle 0
2021-12-18 05:20:23 [140450061088512] WARN - notifyDataAvailable(): Failed to un-pause curl with error: 43. Curl object 0x7fbd0801acd0
2021-12-18 05:20:23 [140450061088512] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000080, Next state: 0x0000000000000100
2021-12-18 05:20:23 [140450061088512] INFO - writeHeaderCallback(): RequestId: cfef91a9-b7c7-c3b8-95d8-e0717744d22d
```

disa6302, 2022-02-02:
@daveisfera , 

I have not had the chance to get to this ticket. Will update once I have something.

daveisfera, 2022-02-04:
You may have already noticed, but 1.20 was released:
https://gstreamer.freedesktop.org/releases/1.20/

ajith-karkera, 2022-08-05:
Is this issue fixed?

jdelapla, 2023-01-20:
@ajith-karkera your pads are incorrect, you should not link decodebin to kvssink, remove that element and try again.
@daveisfera Your issue is "failed delayed linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstRtpH264Depay named rtph264depay0" 

As always, please test your gstreamer pipelines with a `filesink`. If it works with a filesink, but doesn't work with kvssink, then please reopen the ticket and we can help.

closed 2023-01-20 by jdelapla

jdelapla, 2023-02-22:
Can you share logs & pipeline that works with `filesink` but doesn't work with `kvssink` ? All the logs you've shared prior show a failure to link `rtspsrc` to `rtph264depay`, which `kvssink` doesn't play any role in.

daveisfera, 2023-02-23:
I just rebuilt with 3.3.1 and Debian as a base, and using `filesink` and `kvssink` is working with 1.18.6 and 1.22.0. The playback on the test page pauses and won't resume at times, but it appears that it's at least building and running now.