awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #801: [BUG] Not able to build KVS plugin in release mode

RajPrasanna-Quantiphi, 2021-12-06:
We are using AWS KVS in our production application to ingest Video from Camera to AWS. Our edge pipeline is built with GStreamer + KVS sink on a Windows 10 machine. Given this context, we are able to build the KVS sink plugin in both debug & release mode but the plugin built in release mode is not working. As it is a production system, release mode will be optimal & convenient. Need support in this regard.

- Could you provide more information about what errors are occurring when attempting to use the plugin when it is built in release mode?
During build process there are no errors but for a kvs sink plugin built in release mode, when we run gst-inspect-1.0 kvssink is not showing any plugin information. But in debug mode it shows plugin information. 

- Could you provide the exact commands used to build and to run the plugin in both debug and release mode?
cmake -DCMAKE_BUILD_TYPE=Release -DBUILD_GSTREAMER_PLUGIN=TRUE ..

disa6302, 2021-12-06:
@RajPrasanna-Quantiphi , 

We do not use the flag while building `kvssink`. `kvssink` releases with our CPP SDK and you can use our official release (Release 3.2.0) to use `kvssink`.

closed 2021-12-06 by disa6302

RajPrasanna-Quantiphi, 2021-12-08:
@disa6302 Thanks for the response. Can you please provide the link to download the kvssink 3.2.0 binary?

sharadrj, 2022-02-09:
Sorry for the delay in getting back. We don't provide binaries for our releases. Suggest you build the plugin from source but without using the release flag.