awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #803: No Video Streaming Producer support for IOS

bfwizard, 2021-12-09:
I understand this is not the best place to put this (producer-sdk-cpp), but we are using mobile phones as cameras and wish to process those videos in AI/ML models we have in AWS.

It would be great if your team could work with aws-amplify/aws-sdk-ios teams to get this working by creating a swift to c bridge so we can use PutMedia calls

Currently there is no solution available to IOS as producer and so we have no solution to this business use-case.

hassanctech, 2021-12-09:
Indeed there is no Producer SDK available today for iOS.  Our recommendation has been to build the C Libraries for your platform and wrap those at a higher level (Swift for example) so you can use them in your Swift applications.  Similar to the usual C++/JNI/Java model for C++ to Java.  We are looking into offering something along the lines of what you have suggested but do not have a timeline for that unfortunately!

bfwizard, 2021-12-10:
I think AWS management is missing a big opportunity here to serve this business case.  I understand the use-case for Kinesis video streams was for webcams and the like. But there are literally billions of cellphones with cameras on them on the planet.  If you build the "last mile" capability to get those camera's videos into the cloud, then this creates an enormous business opportunities for companies and startups to process those videos with AI/ML....and of course AWS could capitalize on on this.

jdelapla, 2023-01-20:
Although I agree that this would be an excellent project for KVS to undertake, it is not currently planned. Since this is not a feature  request to the cpp sdk and rather a product request I will be closing this ticket.

closed 2023-01-20 by jdelapla