awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #804: [BUG] Not closing after EOS

daveisfera, 2021-12-11:
**Logging**
```
03:53:24.591 postReadCallback(): Reported end-of-stream for stream my_stream. Upload handle: 262
03:53:24.591 writeHeaderCallback(): RequestId: ee1ff8b1-16be-4411-ac60-6c4c665d2dcc
03:53:24.591 getStreamingEndpointResultEvent(): Get streaming endpoint result event.
03:53:24.591 getStreamingTokenResultEvent(): Get streaming token result event.
03:53:24.591 putStreamResultEvent(): Put stream result event. New upload handle 264
03:53:24.591 writeHeaderCallback(): RequestId: ff327f9d-d0f4-2a99-a511-915a0d3cbc3c
03:53:24.591 getStreamData(): Handle 263 waiting for last persisted ack with ts 16385260865620000
03:53:24.591 getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 263
03:53:24.591 postReadCallback(): Reported end-of-stream for stream my_stream. Upload handle: 263
03:53:24.591 writeHeaderCallback(): RequestId: abcc507d-6372-4136-b8a0-eac62e16c79c
03:53:24.591 getStreamingEndpointResultEvent(): Get streaming endpoint result event.
03:53:24.591 getStreamingTokenResultEvent(): Get streaming token result event.
03:53:24.591 putStreamResultEvent(): Put stream result event. New upload handle 265
03:53:24.591 writeHeaderCallback(): RequestId: ec0db659-4e25-bb11-b62e-4128b78bae7b
03:53:24.591 getStreamData(): Handle 264 waiting for last persisted ack with ts 16385283064830000
03:53:24.591 getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 264
03:53:24.591 postReadCallback(): Reported end-of-stream for stream my_stream. Upload handle: 264
03:53:24.591 stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 000055a65914ac38.
03:53:24.591 postReadCallback(): Reported end-of-stream for stream my_stream. Upload handle: 265
03:53:24.591 WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
03:53:24.591 Additional debug info:
03:53:24.591 gstrtspsrc.c(5289): gst_rtspsrc_loop_udp (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
03:53:24.591 Unhandled return value -7.
03:53:24.591 EOS on shutdown enabled -- waiting for EOS after Error
03:53:24.591 Waiting for EOS...
03:53:24.593 ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
03:53:24.593 Additional debug info:
03:53:24.593 gstrtspsrc.c(5357): gst_rtspsrc_loop_udp (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
03:53:24.593 Could not receive message. (System error)
03:53:24.593 ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Internal data stream error.
03:53:24.593 Additional debug info:
03:53:24.593 gstrtspsrc.c(5649): gst_rtspsrc_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
03:53:24.593 streaming stopped, reason error (-5)
03:54:45.983 stopStreamSync(): Failed to stop Kinesis Video Stream - timed out.
03:54:45.984 stopKinesisVideoStreamSync(): operation returned status code: 0x0000000f
03:54:45.984 Failed to stop the stream with: 15
03:54:45.984 Sending eos
03:54:45.984 Got EOS from element "pipeline0".
03:54:45.984 EOS received - stopping pipeline...
03:54:45.984 Execution ended after 168:00:40.115686242
03:54:45.984 Setting pipeline to PAUSED ...
03:54:45.984 Setting pipeline to READY ...
03:54:45.985 Setting pipeline to NULL ...
03:54:45.986 Freeing pipeline ...
```

**Describe the bug**
Occasionally, the process will remain open when there's an EOS even though things have closed down

**To Reproduce**
Don't have a consistent reproducer yet, but it does happen occasionally

**Expected behavior**
Process would close when EOS happens

**SDK version number**
3.2.0

**Open source building**
Docker container

**Platform (please complete the following information):**
 - OS: Debian
 - Processor: x86-64
 - Version: Buster

niyatim23, 2022-01-26:
Hi, can you please answer the following questions for me to help you:

1. By "the process will remain open", do you mean it hangs?
2. Do you see errors/anything out of the blue in your GStreamer logs?

daveisfera, 2022-01-27:
1. Yes, it stays running waiting for the EOS and doesn't close (even after hours or days of waiting)
2. The errors shown above are the majority of the times that I've seen this happen

MushMal, 2022-01-27:
03:54:45.984 stopKinesisVideoStreamSync(): operation returned status code: 0x0000000f
states that you were unable to properly stream out all of the remaining bits in the buffer and receive a final ack. The streaming had timed out. Check your network connection

daveisfera, 2022-01-27:
Yes, I agree that there was an error and the data wasn't sent correctly, but then gstreamer tries to turn off and waits for the EOS but never exits, so a new process won't start since it's waiting for this one to finish.

MushMal, 2022-01-27:
Sounds like bug in the error handler for the gstreamer plugin. Not sure if it's handled in the demo repository for the combined webrtc/streamer plugin. @niyatim23 any idea?

niyatim23, 2022-02-01:
@daveisfera, the log that you have shared does not show the application hanging. Can you please share the logs where this can be seen?
@MushMal, I looked at the code to handle EOS [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/gstreamer/gstkvssink.cpp#L1059-L1070) but nothing looks problematic at the moment. I think the logs reproducing the issue will be helpful

daveisfera, 2022-02-01:
That's the last output that we see until we hard kill the process. If there's any additional info that we can provide, then please let us know and we'll do our best to help out.

hassanctech, 2022-02-01:
````
03:54:45.984 Setting pipeline to PAUSED ...
03:54:45.984 Setting pipeline to READY ...
03:54:45.985 Setting pipeline to NULL ...
03:54:45.986 Freeing pipeline ...
```

After this output it still hangs that's what you're saying? Just want to make sure I understand.  Have you tried with setting `GST_DEBUG` to see any other output? This doesn't seem related to our plugin if it made it this far.

MushMal, 2022-02-01:
Looking at the code @niyatim23  linked, it seems that the result of the stop is being ignored. There should be an error printed out which we see

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/src/KinesisVideoStream.cpp#L158

Per @hassanctech I would also recommend setting a debug level for GStreamer to try to get more info

daveisfera, 2022-02-01:
Which debug level should I run with? 2? 3?

daveisfera, 2022-02-03:
Here's another set of output:
```
01:14:19.359 ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
01:14:19.359 Additional debug info:
01:14:19.359 gstrtspsrc.c(5705): gst_rtspsrc_loop_udp (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
01:14:19.359 Could not receive message. (System error)
01:14:19.359 INFO getStreamingEndpointResultEvent(): Get streaming endpoint result event.
01:14:19.359 ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Internal data stream error.
01:14:19.359 Additional debug info:
01:14:19.359 INFO getStreamingTokenResultEvent(): Get streaming token result event.
01:14:19.359 gstrtspsrc.c(6057): gst_rtspsrc_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
01:14:19.359 INFO putStreamResultEvent(): Put stream result event. New upload handle 88
01:14:19.359 streaming stopped, reason error (-5)
01:14:19.359 INFO writeHeaderCallback(): RequestId: d3f398e6-b2d8-b9b2-8986-ee7a72f63d60
01:14:19.359 INFO getStreamData(): Handle 87 waiting for last persisted ack with ts 16438692796760000
01:14:19.359 INFO getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 87
01:14:19.359 INFO postReadCallback(): Reported end-of-stream for stream pitchaware-prod-creekside_6_3-security. Upload handle: 87
01:14:19.359 INFO writeHeaderCallback(): RequestId: 2689b6b2-e9a1-4bf0-8213-fb8ffc766037
01:14:19.359 INFO getStreamingEndpointResultEvent(): Get streaming endpoint result event.
01:14:19.359 INFO getStreamingTokenResultEvent(): Get streaming token result event.
01:14:19.359 INFO putStreamResultEvent(): Put stream result event. New upload handle 89
01:14:19.359 INFO writeHeaderCallback(): RequestId: f5de2dbd-d5f5-a3cd-afab-52322bcd6cc9
01:14:19.359 INFO getStreamData(): Handle 88 waiting for last persisted ack with ts 16438715696700000
01:14:19.359 INFO getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 88
01:14:19.359 INFO postReadCallback(): Reported end-of-stream for stream pitchaware-prod-creekside_6_3-security. Upload handle: 88
01:14:19.359 INFO writeHeaderCallback(): RequestId: 6446a4fa-a72f-416e-bdf3-537d7a3045e6
01:14:19.359 INFO getStreamingEndpointResultEvent(): Get streaming endpoint result event.
01:14:19.359 INFO getStreamingTokenResultEvent(): Get streaming token result event.
01:14:19.359 INFO putStreamResultEvent(): Put stream result event. New upload handle 90
01:14:19.359 INFO writeHeaderCallback(): RequestId: e1857fb8-1f0b-36ab-bbf0-f781b2aa9d9a
01:14:19.359 INFO getStreamData(): Handle 89 waiting for last persisted ack with ts 16438737846570000
01:14:19.359 INFO getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 89
01:14:19.359 INFO postReadCallback(): Reported end-of-stream for stream pitchaware-prod-creekside_6_3-security. Upload handle: 89
01:14:19.359 INFO stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 000055ba94589ff8.
01:14:19.359 INFO postReadCallback(): Reported end-of-stream for stream pitchaware-prod-creekside_6_3-security. Upload handle: 90
01:14:19.359 WARNING: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Could not read from resource.
01:14:19.359 Additional debug info:
01:14:19.359 gstrtspsrc.c(5637): gst_rtspsrc_loop_udp (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
01:14:19.359 Unhandled return value -7.
01:14:19.359 EOS on shutdown enabled -- waiting for EOS after Error
01:14:19.359 Waiting for EOS...
01:15:42.364 ERROR stopStreamSync(): Failed to stop Kinesis Video Stream - timed out.
01:15:42.364 ERROR stopKinesisVideoStreamSync(): operation returned status code: 0x0000000f
01:15:42.364 ERROR Failed to stop the stream with: 15
01:15:42.364 INFO Sending eos
01:15:42.364 Got EOS from element "pipeline0".
01:15:42.364 EOS received - stopping pipeline...
01:15:42.367 Execution ended after 57:02:08.867008355
01:15:42.367 Setting pipeline to NULL ...
01:15:42.367 Freeing pipeline ...
```

The logs that stand out to me are:
```
streaming stopped, reason error (-5)
```
and
```
Unhandled return value -7.
```

hassanctech, 2022-02-03:
Is it always the case when this happens that the `stopKinesisVideoStreamSync()` times out? 

Will look into this perhaps curl thread is not exiting and is blocked for some reason.

Hassan

daveisfera, 2022-02-03:
Yes, that same output was there in all of the instances that I checked out in the logs.

niyatim23, 2022-02-28:
Hi @daveisfera, can you please share TRACE logs for the SDK? To get those logs, please set the log level to `TRACE` [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_log_configuration#L1). Additionally, please make sure that the log config file path is correctly specified [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/0121f2de342104c982171aa9bbb09c6a1a4fe652/src/gstreamer/gstkvssink.cpp#L106)

daveisfera, 2022-03-02:
I've made that change and I will keep an eye out for logs from this issue when it does

daveisfera, 2022-03-17:
I had to turn this off, because running with `TRACE` produces a LOT of logs and this issue takes a LONG time to happen (usually hours or even days). What else can I do to help debug this?

niyatim23, 2023-05-07:
Are you still seeing this issue? If there are no steps to reproduce this, unfortunately, we cannot do much to help

niyatim23, 2023-05-12:
Closing since this issue is stale. Feel free to open / reopen an issue

closed 2023-05-12 by niyatim23