awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #805: No such element or plugin 'kvssink'

abhishekPurandare-obs, 2022-01-03:
I am still getting the same error after executing the following commands.
```
In repo's root dir
>> mkdir -p build
>> cd build
>> sudo apt-get install libssl-dev libcurl4-openssl-dev liblog4cplus-dev libgstreamer1.0-dev libgstreamer-plugins-base1.0-dev
gstreamer1.0-plugins-base-apps gstreamer1.0-plugins-bad gstreamer1.0-plugins-good
gstreamer1.0-plugins-ugly gstreamer1.0-tools
>> cmake -DBUILD_GSTREAMER_PLUGIN=TRUE ..
>> make
In repo's root dir
>> export GST_PLUGIN_PATH=`pwd`/build
>> export LD_LIBRARY_PATH=`pwd`/open-source/local/lib
 
```

Build directory contents

```
CMakeCache.txt       dependency                          kvs_gstreamer_multistream_sample  libKinesisVideoProducer.so
CMakeFiles           kvs_gstreamer_audio_video_sample    kvs_gstreamer_sample              Makefile
cmake_install.cmake  kvs_gstreamer_file_uploader_sample  libgstkvssink.so
```

It seems like the shared objects were created. But I am still getting 

No such element or plugin 'kvssink' after executing `gst-inspect-1.0 kvssink`

Versions 
**CMake 3.22.1** on Ubuntu 20.04.3

hassanctech, 2022-01-03:
You need to set `GST_PLUGIN_PATH` and `LD_LIBRARY_PATH`.  `GST_PLUGIN_PATH` is where `gst-inspect-1.0` will look for the plugin .so file.  And the `LD_LIBRARY_PATH` is where `ld` will look for the dependencies for the plugin (like libproducer etc.) make sure those are both set

hassanctech, 2022-01-03:
Also you can try it on the `.so` directly so `gst-inspect-1.0 libgstkvssink.so`

abhishekPurandare-obs, 2022-01-04:
> Also you can try it on the `.so` directly so `gst-inspect-1.0 libgstkvssink.so`

This worked I guess. But getting blacklisted though

```
Plugin Details:
  Name                     libgstkvssink.so
  Description              Plugin for blacklisted file
  Filename                 /home/laptop-obs-65/kinesis/amazon-kinesis-video-streams-producer-sdk-cpp/build/libgstkvssink.so
  Version                  0.0.0
  License                  BLACKLIST
  Source module            BLACKLIST
  Binary package           BLACKLIST
  Origin URL               BLACKLIST
```

hassanctech, 2022-01-04:
Yeah ok so you just need to clear your blacklist it's possible during your setup you inadvertently blacklisted it, until you remove it from the blacklist the `gst-inspect-1.0` will likely keep reporting that it can't find it.

disa6302, 2022-01-19:
Closing due to inactivity.