awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #808: [QUESTION] Unable to clone libopenssl

Radagan, 2022-01-18:
When running CMake on Linux (Armbian Focus for an SoC called Pinecube), it locks up on "Cloning into 'project_libopenssl':

```
[100%] Built target libkvscproducer-download
-- Kinesis Video Producer path is /home/radagan/dev/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Begin building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/radagan/dev/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl
Scanning dependencies of target project_libopenssl
[ 11%] Creating directories for 'project_libopenssl'
[ 22%] Performing download step (git clone) for 'project_libopenssl'
Cloning into 'project_libopenssl'...
error: index-pack died of signal 9
fatal: index-pack failed
Cloning into 'project_libopenssl'...
error: index-pack died of signal 9
fatal: index-pack failed
Cloning into 'project_libopenssl'...
error: index-pack died of signal 9
fatal: index-pack failed
-- Had to git clone more than once:
          3 times.
CMake Error at /home/radagan/dev/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/tmp/project_libopenssl-gitclone.cmake:31 (message):
  Failed to clone repository: 'https://github.com/openssl/openssl.git'

make[2]: *** [CMakeFiles/project_libopenssl.dir/build.make:92: build/src/project_libopenssl-stamp/project_libopenssl-download] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/project_libopenssl.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMake/Utilities.cmake:93 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:66 (build_dependency)
```

Any idea what I'm doing wrong. I did clone with submodules.

kateyanurag, 2022-01-19:
@Radagan
This typically happens when the on-disk memory is not enough to clone. I cannot think of any other reason for clone to fail.

Radagan, 2022-01-19:
That could definitely be the case. It does build without dependencies, but then it fails with "Could not initialize supporting library", but without naming the missing lib. I think I have them on my system, but passing /usr/lib doesn't work. Is there an easy way to get the list of dependencies built so I could just copy them to open-source/local/lib?

kateyanurag, 2022-01-20:
@Radagan there are the list of dependencies - https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/master/CMake/Dependencies
I myself haven't separately built them and the copied to open-source/local/lib but I'm guessing that should work.

Radagan, 2022-01-20:
@kateyanurag Thank you! Will give it a go.

closed 2022-01-20 by Radagan

bml1g12, 2022-04-05:
What was the solution? Getting on this same issue on a Ubuntu host using the [official docker image](https://github.com/aws-samples/amazon-kinesis-video-streams-demos/blob/master/producer-cpp/docker-rtsp/Dockerfile) with lots of RAM and HD space available

himadrii27, 2024-06-06:
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMake/Utilities.cmake:93 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:74(build_dependency)
  
  I am also getting the same error can you please help me with this?