awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #809: [BUG] C++ Producer SDK doesn't install (copy files to /usr)

daveisfera, 2022-01-19:
**Logging**
Trying to build the C++ Producer SDK and it doesn't actually install as expected
```
/usr/bin/cmake -S/opt/amazon-kinesis-video-streams-producer-sdk-cpp -B/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build --check-build-system CMakeFiles/Makefile.cmake 0
/usr/bin/cmake -E cmake_progress_start /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build//CMakeFiles/progress.marks
make  -f CMakeFiles/Makefile2 all
make[1]: Entering directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
make  -f dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicUtils.dir/build.make dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicUtils.dir/depend
make[2]: Entering directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
cd /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /opt/amazon-kinesis-video-streams-producer-sdk-cpp /opt/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicUtils.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
make  -f dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicUtils.dir/build.make dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicUtils.dir/build
make[2]: Entering directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
make[2]: Nothing to be done for 'dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspicUtils.dir/build'.
make[2]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
[ 23%] Built target kvspicUtils
make  -f dependency/libkvscproducer/kvscproducer-src/CMakeFiles/kvsCommonCurl.dir/build.make dependency/libkvscproducer/kvscproducer-src/CMakeFiles/kvsCommonCurl.dir/depend
make[2]: Entering directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
cd /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /opt/amazon-kinesis-video-streams-producer-sdk-cpp /opt/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/dependency/libkvscproducer/kvscproducer-src /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/dependency/libkvscproducer/kvscproducer-src/CMakeFiles/kvsCommonCurl.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
make  -f dependency/libkvscproducer/kvscproducer-src/CMakeFiles/kvsCommonCurl.dir/build.make dependency/libkvscproducer/kvscproducer-src/CMakeFiles/kvsCommonCurl.dir/build
make[2]: Entering directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
make[2]: Nothing to be done for 'dependency/libkvscproducer/kvscproducer-src/CMakeFiles/kvsCommonCurl.dir/build'.
make[2]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
[ 33%] Built target kvsCommonCurl
make  -f dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspic.dir/build.make dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspic.dir/depend
make[2]: Entering directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
cd /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /opt/amazon-kinesis-video-streams-producer-sdk-cpp /opt/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspic.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
make  -f dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspic.dir/build.make dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspic.dir/build
make[2]: Entering directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
make[2]: Nothing to be done for 'dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/CMakeFiles/kvspic.dir/build'.
make[2]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
[ 75%] Built target kvspic
make  -f dependency/libkvscproducer/kvscproducer-src/CMakeFiles/cproducer.dir/build.make dependency/libkvscproducer/kvscproducer-src/CMakeFiles/cproducer.dir/depend
make[2]: Entering directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
cd /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /opt/amazon-kinesis-video-streams-producer-sdk-cpp /opt/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/dependency/libkvscproducer/kvscproducer-src /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/dependency/libkvscproducer/kvscproducer-src/CMakeFiles/cproducer.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
make  -f dependency/libkvscproducer/kvscproducer-src/CMakeFiles/cproducer.dir/build.make dependency/libkvscproducer/kvscproducer-src/CMakeFiles/cproducer.dir/build
make[2]: Entering directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
make[2]: Nothing to be done for 'dependency/libkvscproducer/kvscproducer-src/CMakeFiles/cproducer.dir/build'.
make[2]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
[ 89%] Built target cproducer
make  -f CMakeFiles/KinesisVideoProducer.dir/build.make CMakeFiles/KinesisVideoProducer.dir/depend
make[2]: Entering directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
cd /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build && /usr/bin/cmake -E cmake_depends "Unix Makefiles" /opt/amazon-kinesis-video-streams-producer-sdk-cpp /opt/amazon-kinesis-video-streams-producer-sdk-cpp /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/KinesisVideoProducer.dir/DependInfo.cmake --color=
make[2]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
make  -f CMakeFiles/KinesisVideoProducer.dir/build.make CMakeFiles/KinesisVideoProducer.dir/build
make[2]: Entering directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
make[2]: Nothing to be done for 'CMakeFiles/KinesisVideoProducer.dir/build'.
make[2]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
[100%] Built target KinesisVideoProducer
make[1]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
/usr/bin/cmake -E cmake_progress_start /opt/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles 0
make  -f CMakeFiles/Makefile2 preinstall
make[1]: Entering directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
make[1]: Nothing to be done for 'preinstall'.
make[1]: Leaving directory '/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build'
Install the project...
/usr/bin/cmake -P cmake_install.cmake
-- Install configuration: ""
```

**Describe the bug**
`make install` doesn't install to `CMAKE_INSTALL_PREFIX` like it should

**To Reproduce**
Steps to reproduce the behavior:
1. `cmake -DBUILD_GSTREAMER_PLUGIN=OFF -DBUILD_DEPENDENCIES=OFF -DCMAKE_INSTALL_PREFIX=/usr ..`
2. `make`
3. `make install`
4. Notice that it didn't copy any files to `/usr`

**Expected behavior**
Shared libraries and headers copied to `/usr`

**SDK version number**
3.2.0

**Open source building**
Built with Debian Bullseye

**Platform (please complete the following information):**
 - OS: Debian
 - Processor: x64
 - Version: Bullseye

niyatim23, 2022-01-26:
Hi, can you please check if you face the same issue with `-DBUILD_DEPENDENCIES=ON`?

daveisfera, 2022-01-27:
Still has the same problem

CY-1992, 2022-03-29:
Facing the same issue. Any progress on it?

disa6302, 2022-09-16:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/905 --- this merged into develop. Please give it a try and reach out to us if it does not work. Thanks!

closed 2022-09-16 by disa6302

RaistlinD2x, 2023-03-09:
```
[  2%] Built target kvs_gstreamer_file_uploader_sample
[ 39%] Built target kvspic
[ 59%] Built target kvspicUtils
[ 68%] Built target kvsCommonCurl
[ 79%] Built target cproducer
[ 88%] Built target KinesisVideoProducer
[ 89%] Built target kvs_gstreamer_audio_video_sample
[ 91%] Built target kvs_gstreamer_multistream_sample
[ 92%] Built target kvs_gstreamer_sample
[ 96%] Built target gstkvssink
[100%] Built target KinesisVideoProducerJNI
```

Is the output of make but when I look in the `build` dir the plugin doesn't exist.