awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #810: [BUG] Stability issue - Streaming stops after 40 mins with a crash.when streaming

vaishna-11, 2022-01-21:
**Logging**
Add relevent SDK logging.  IMPORTANT NOTE:  Please make sure to NOT share AWS access credentials under any circumstance!  Please make sure they are not in the logs.

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**SDK version number**
Include the SDK version you are running. If it is a specific commit on master branch, include that

**Open source building**
If it is a build issue, include 3rd party library version and steps to how you are building it

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - OS: [e.g. Linux, Windows]
 - Processor: [e.g. ARMv5]
 - Version [e.g. 18.04]

**Additional context**
Add any other context about the problem here.

kateyanurag, 2022-01-21:
@vaishna-11 a few questions - 
(1) Is this consistently reproducible? 
(2) Have you noticed this crash each time the sample is run for ~40mins or it was just a one-off occurrence?  
(3) How much memory does your device have? Have you confirmed that this is not due to device memory exhaustion?

vaishna-11, 2022-01-22:
Please find the answers.
1. yes, the issue is consistently reproducible.
2. Issue happens almost most of the times after streaming for approx 40mins. it is not one-off occurence.
3. yes, this is not a memory issue. we have monitored the memory usage while the tests were run.

vaishna-11, 2022-01-23:
Update:
We are able to recreate the issue much faster when we change DEFAULT_CREDENTIAL_ROTATION_SECONDS to smaller value.

disa6302, 2022-01-23:
@vaishna-11 ,

What credential provider are you using?

vaishna-11, 2022-01-24:
This test was based on access secret key. how do we confirm the same for the certificate files?

kateyanurag, 2022-01-24:
@vaishna-11 I tried reproducing this issue but was not able to see the failure.
Did you make any other code changes in the sample code?

niyatim23, 2022-01-24:
By "credential provider" we're trying to understand which of the providers listed [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/docs/auth.md#kvs-sdk-auth-integration) are you using?

MushMal, 2022-01-25:
@vaishna-11 rotating credential provider is designed for test purposes only. If you have static credentials such as AWS access key and secret key then you should be using static credentials provider. Most likely, your provider is setting the credential expiration to an expired date/time. What I don't understand is whether this is causing a crash and what's crashing.

* Have the software stabilized on a commodity platform prior flushing on your device
* Ensure the stock samples work on your device before modifying so you can debug your own issues
* Include the detailed crash stack in verbose mode so someone could help perhaps looking at your stack
* For devices, either use IoT credential provider or something similar - don't hard-code static credentials on the device (and hence use Static credential provider).

vaishna-11, 2022-01-25:
> @vaishna-11 I tried reproducing this issue but was not able to see the failure. Did you make any other code changes in the sample code OR anywhere in the SDK?
> 
> Also what is your rtsp source? I used file as a source in my testing.

@kateyanurag 
No changes were done on the sample app for this testing. 
Source we are trying to use is the rtsp url in the network, as mentioned in the steps to reproduce.
**./kvs_gstreamer_sample stream_name src_url**
**./kvs_gstreamer_sample kvs_example_camera_stream rtsp://192.168.137.1:8554/mystream**

vaishna-11, 2022-01-25:
@MushMal  @niyatim23 

From our application, we are using IoT based credential provider which makes use of Curl client.
We observed the crash in our application every ~40mins while streaming. We tried playing around the value MAX_ENFORCED_TOKEN_EXPIRATION_DURATION to 3mins and could see that application crashed in 3mins.

So tried similar approach on the sample app too with access/secret key and could recreate the issue and the same happened there as well. There were no changes done at the sample app side.

disa6302, 2022-02-17:
@vaishna-11 , 

The stack trace you provided seems to be a binary of sorts. It does not have any information to go with it. I have also run the sample as is with default credentials locally and I am unable to reproduce the crash. 
Can you please provide a stack backtrace? You can use `bt` in `gdb` to generate this. Thanks!

disa6302, 2022-03-07:
Cloding due to no response. We are unable to reproduce this. Please feel free to reopen.

closed 2022-03-07 by disa6302