awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #813: Log4cplus and curl are not built in release mode

Need-4-Speed, 2022-01-29:
**Describe the bug**
Log4cplus and curl's  release dlls are not generated when SDK is built with -DCMAKE_BUILD_TYPE=Release flag.

**To Reproduce**
Steps to reproduce the behavior:
1. Run cmake ..  -DCMAKE_BUILD_TYPE=Release  -DBUILD_GSTREAMER_PLUGIN=ON  -DBUILD_TEST=TRUE
2. Check kvs\open-source\local\bin folder for log4cplus and curl dlls.
3. You can notice that release dlls are not generated. Check for -D postfix notation.

**Expected behavior**
Should generate the dlls based on the release type.

**SDK version number**
Latest Master 

**Possible Fix**
Release build is issue is fixed by making to Cmake\Utilities.cmake file. I have corrected the code in fetch_repo and build_dependency functions by adding build type flag.
 
Before:
  execute_process(
    COMMAND ${CMAKE_COMMAND} --build . 
    RESULT_VARIABLE result
    WORKING_DIRECTORY ${DEPENDENCY_DOWNLOAD_PATH}/lib${lib_name})
 
After:
  execute_process(
    COMMAND ${CMAKE_COMMAND} --build . --config ${CMAKE_BUILD_TYPE}
    RESULT_VARIABLE result
    WORKING_DIRECTORY ${DEPENDENCY_DOWNLOAD_PATH}/lib${lib_name})

niyatim23, 2022-02-14:
Hi, thank you for reporting it. We do welcome contributions, feel free to open a PR with the changes and the team will review it.

niyatim23, 2022-02-26:
Closing due to inactivity, please open a PR for this contribution. Feel free to raise any further issues/questions.

Need-4-Speed, 2022-04-02:
I tried to create a branch before creating a pull request. I am not able to create a branch. I am receiving 403 error