awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #814: [BUG] Hanging on SIGSEGV

daveisfera, 2022-01-31:
**Logging**
```
Caught SIGSEGV
exec gdb failed: No such file or directory
Spinning.  Please run 'gdb gst-launch-1.0 8' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
```

**Describe the bug**
`gstreamer` is crashing and then hanging 

**To Reproduce**
Happens occasionally after it's been running for a while and I don't have a specific reproducer yet

**Expected behavior**
It wouldn't crash, but at least if it did crash, then it wouldn't hang trying to start `gdb`

**SDK version number**
3.2.0

**Platform (please complete the following information):**
 - OS: Debian
 - Processor: x64
 - Version: Bullseye

**Additional context**
I think this might be related to #804

MushMal, 2022-01-31:
I doubt this is a related issue to the one mentioned. In this case, it's likely that your OS is failing to start the gdb due to memory pressures. Attempt to run the application under the gdb from the start in order to capture it. Alternatively, set your OS to capture the mini-dump so you can examine it later on a commodity platform instead of your device. As for the origin of the crash - unless you have a trace with the stack it would be impossible to trace.

daveisfera, 2022-01-31:
It's running in a docker container and `gdb` isn't available, so I'll add it to see what info it can shed on this issue.

daveisfera, 2022-02-01:
`gdb` is now inside of the docker container and this is the output that we see:
```
Caught SIGSEGV
Spinning.  Please run 'gdb gst-launch-1.0 7' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
```

Is there an ENV or flag we can set for it to do the core dump instead of spinning inside of `gdb`?

daveisfera, 2022-02-10:
Any update on this? What can I do to help get this fixed?

niyatim23, 2022-03-10:
Hi @daveisfera, it would be really difficult to investigate this without logs since we have not been able to reproduce it. Can you please share trace logs for this?

daveisfera, 2022-03-17:
I'm looking to get it to at least not spin in `gdb`. How do I do that? It would be extra nice, if it would dump the stack trace and then exit, but for now, I'd take it just exiting rather than spinning and creating a zombie process.

daveisfera, 2022-05-09:
Is there a way to make it not start `gdb`? Currently being stuck there is a much bigger problem than the crash itself, because the service would just restart and start working again, but being stuck in `gdb` incorrectly keeps it alive. Ideally, it `gdb` would dump the stack trace and exit, but I'll take it not starting or exiting immediately so things aren't totally stuck.

daveisfera, 2022-08-11:
Any advice on this? I tried removing `gdb` to prevent it from spinning, but it still happens. This is the output when it happens:
```
Caught SIGSEGV
exec gdb failed: No such file or directory
Spinning.  Please run 'gdb gst-launch-1.0 8' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
```

jdelapla, 2023-01-20:
There is not enough information for the team to assist you with. Please re-open the ticket with a stack trace, otherwise a reported crash in gstreamer with no steps to reproduce could be from any number of factors, even factors unrelated to the SDK.

daveisfera, 2023-01-20:
How can I make it print a stacktrace when it happens?

stephenhibbert, 2023-03-03:
I think I'm seeing the same thing here...

I'm running on an M1 pro mac using rancher desktop for docker incase that helps...
```

root@lima-rancher-desktop:/opt/amazon-kinesis-video-streams-producer-sdk-cpp/build# gst-launch-1.0 -v rtspsrc location=rtsp://localhost:8554/mystream short-header=TRUE ! rtph264depay ! h264parse ! kvssink stream-name=mystream storage-size=128 aws-region=eu-west-1 access-key=*** secret-key=***
Setting pipeline to PAUSED ...
[INFO ] [03-03-2023 15:39:39:445.813 GMT] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ] [03-03-2023 15:39:39:445.850 GMT] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ] [03-03-2023 15:39:39:445.856 GMT] heapInitialize(): Creating AIV heap.
[INFO ] [03-03-2023 15:39:39:445.864 GMT] heapInitialize(): Heap is initialized OK
[DEBUG] [03-03-2023 15:39:39:445.948 GMT] getSecurityTokenHandler invoked
[DEBUG] [03-03-2023 15:39:39:445.956 GMT] Refreshing credentials. Force refreshing: 0 Now time is: 1677857979445954854 Expiration: 0
[INFO ] [03-03-2023 15:39:39:445.964 GMT] createDeviceResultEvent(): Create device result event.
[DEBUG] [03-03-2023 15:39:39:445.974 GMT] clientReadyHandler invoked
[INFO ] [03-03-2023 15:39:39:445.994 GMT] try creating stream
[INFO ] [03-03-2023 15:39:39:446.023 GMT] Creating Kinesis Video Stream mystream
[INFO ] [03-03-2023 15:39:39:446.031 GMT] createKinesisVideoStream(): Creating Kinesis Video Stream.
[INFO ] [03-03-2023 15:39:39:446.038 GMT] logStreamInfo(): SDK version: 8867047d8d8f56645009ed73788a57f0a6107772
[DEBUG] [03-03-2023 15:39:39:446.045 GMT] logStreamInfo(): Kinesis Video Stream Info
[DEBUG] [03-03-2023 15:39:39:446.050 GMT] logStreamInfo():      Stream name: mystream 
[DEBUG] [03-03-2023 15:39:39:446.055 GMT] logStreamInfo():      Streaming type: STREAMING_TYPE_REALTIME 
[DEBUG] [03-03-2023 15:39:39:446.060 GMT] logStreamInfo():      Content type: video/h264 
[DEBUG] [03-03-2023 15:39:39:446.065 GMT] logStreamInfo():      Max latency (100ns): 600000000
[DEBUG] [03-03-2023 15:39:39:446.070 GMT] logStreamInfo():      Fragment duration (100ns): 20000000
[DEBUG] [03-03-2023 15:39:39:446.074 GMT] logStreamInfo():      Key frame fragmentation: Yes
[DEBUG] [03-03-2023 15:39:39:446.079 GMT] logStreamInfo():      Use frame timecode: Yes
[DEBUG] [03-03-2023 15:39:39:446.083 GMT] logStreamInfo():      Absolute frame timecode: Yes
[DEBUG] [03-03-2023 15:39:39:446.088 GMT] logStreamInfo():      Nal adaptation flags: 0
[DEBUG] [03-03-2023 15:39:39:446.093 GMT] logStreamInfo():      Average bandwith (bps): 4194304
[DEBUG] [03-03-2023 15:39:39:446.098 GMT] logStreamInfo():      Framerate: 25
[DEBUG] [03-03-2023 15:39:39:446.104 GMT] logStreamInfo():      Buffer duration (100ns): 1200000000
[DEBUG] [03-03-2023 15:39:39:446.109 GMT] logStreamInfo():      Replay duration (100ns): 400000000
[DEBUG] [03-03-2023 15:39:39:446.115 GMT] logStreamInfo():      Connection Staleness duration (100ns): 600000000
[DEBUG] [03-03-2023 15:39:39:446.120 GMT] logStreamInfo():      Store Pressure Policy: 1
[DEBUG] [03-03-2023 15:39:39:446.125 GMT] logStreamInfo():      View Overflow Policy: 1
[DEBUG] [03-03-2023 15:39:39:446.130 GMT] logStreamInfo():      Segment UUID: NULL
[DEBUG] [03-03-2023 15:39:39:446.136 GMT] logStreamInfo():      Frame ordering mode: 0
[DEBUG] [03-03-2023 15:39:39:446.141 GMT] logStreamInfo(): Track list
[DEBUG] [03-03-2023 15:39:39:446.146 GMT] logStreamInfo():      Track id: 1
[DEBUG] [03-03-2023 15:39:39:446.152 GMT] logStreamInfo():      Track name: kinesis_video
[DEBUG] [03-03-2023 15:39:39:446.157 GMT] logStreamInfo():      Codec id: V_MPEG4/ISO/AVC
[DEBUG] [03-03-2023 15:39:39:446.162 GMT] logStreamInfo():      Track type: TRACK_INFO_TYPE_VIDEO
[DEBUG] [03-03-2023 15:39:39:446.168 GMT] logStreamInfo():      Track cpd: NULL
[INFO ] [03-03-2023 15:39:39:663.877 GMT] writeHeaderCallback(): RequestId: c7871843-84c1-4288-948c-18aa4f69f702
[DEBUG] [03-03-2023 15:39:39:664.017 GMT] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.677856090885E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:eu-west-1:***:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:eu-west-1:***:stream/mystream/1677856090885","StreamName":"mystream","Version":"BXMLRZ923W1qKQ4UyLpK"}}
[INFO ] [03-03-2023 15:39:39:664.889 GMT] describeStreamResultEvent(): Describe stream result event.
[INFO ] [03-03-2023 15:39:39:853.670 GMT] writeHeaderCallback(): RequestId: 904b8eb1-f2e0-476d-a7b7-a0755a622b47
[DEBUG] [03-03-2023 15:39:39:854.397 GMT] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-5ac677c0.kinesisvideo.eu-west-1.amazonaws.com"}
[INFO ] [03-03-2023 15:39:39:859.535 GMT] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG] [03-03-2023 15:39:39:860.403 GMT] getStreamingTokenHandler invoked
[DEBUG] [03-03-2023 15:39:39:860.473 GMT] Refreshing credentials. Force refreshing: 1 Now time is: 1677857979860457105 Expiration: 18446744073709551615
[INFO ] [03-03-2023 15:39:39:860.574 GMT] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG] [03-03-2023 15:39:39:860.686 GMT] streamReadyHandler invoked
Stream is ready
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
[... cut ...]
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc4: caps = application/x-rtcp
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
ERROR: pipeline doesn't want to play.

(gst-launch-1.0:17): GStreamer-CRITICAL **: 15:39:41.918: gst_message_parse_error: assertion 'GST_MESSAGE_TYPE (message) == GST_MESSAGE_ERROR' failed
Caught SIGSEGV
exec gdb failed: No such file or directory
Spinning.  Please run 'gdb gst-launch-1.0 17' to continue debugging, Ctrl-C to quit, or Ctrl-\ to dump core.
```

niyatim23, 2023-05-07:
@stephenhibbert, can you elaborate further on your setup? Are you trying to run an amd64 docker image on an arm64 system? I don't think that would work because the base architectures are different

niyatim23, 2023-05-12:
Closing since this issue is stale. Feel free to open / reopen an issue

daveisfera, 2023-05-12:
Definitely still an issue. Fixing the SIGSEGV would be nice, but really I'm just asking for a way to have it not hang forever waiting for user input. Is there a way to do that?

stephenhibbert, 2023-05-12:
> @stephenhibbert, can you elaborate further on your setup? Are you trying to run an amd64 docker image on an arm64 system? I don't think that would work because the base architectures are different

My setup is an arm64 M1 MacBook Pro. The Dockerfile is pasted in my original comment above - in summary the arm64 Ubuntu base image causes the SIGSEV. The amd64 image running on Rosetta works as expected. I would like to have the option to run on arm64 to enable Graviton or RPi support.

niyatim23, 2023-05-12:
I see you mention multiple systems. I'm still not clear about your setup. What is your base system and which docker image are you trying to run on it? Are you able to reproduce this issue when using an Ubuntu image on an Ubuntu base system? Have you changed anything in the sample application?

niyatim23, 2023-07-10:
Closing due to no response

closed 2023-07-10 by niyatim23

trishmapow, 2024-01-30:
For the benefit of future readers: the option to set is `gst-launch-1.0 -f` which disables the fault handler so it doesn't hang on GDB. 
https://gstreamer.freedesktop.org/documentation/tools/gst-launch.html?gi-language=c