awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #822: [BUG] - The requested version of GStreamer is not available

adomigar, 2022-02-15:
**Logging**
`[INFO] (Copier) aws.iot.EdgeConnectorForKVS: stdout. 8102 [pool-6-thread-1] [WARN] com.aws.iot.edgeconnectorforkvs.videorecorder.util.GstDao - The requested version of GStreamer is not available. Requested : 1.18.4, Available : 1.16.2. {scriptName=services.aws.iot.EdgeConnectorForKVS.lifecycle.Run.Script, serviceName=aws.iot.EdgeConnectorForKVS, currentState=RUNNING}`

**Describe the bug**
I am trying to ingest a live camera feed connected locally to my greengrass core device. At the moment I deploy the aws.iot.EdgeConnectorForKVS component it complains about the version of GStreamer. I have tried to update GStreamer with apt without success. However I have managed to install GStreamer 1.18.5 with brew. I currently have this version installed in `/home/linuxbrew/.linuxbrew/bin/`. How can I point this new directory to the EdgeConnectorForKVS component so that it finds this new version?

**Platform (please complete the following information):**
 - OS: Ubuntu
 - Version 20.04

disa6302, 2022-02-17:
@adomigar ,

I dont think I understand your set up. Are you building this SDK following the build instructions? We do not have any Greengrass integration, so not sure how we can help here.

disa6302, 2022-02-21:
Closing due to inactivity and lack of relevance. Feel free to reach out if you have any KVS specific question.

closed 2022-02-21 by disa6302