awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #823: [BUG] KVS Stream Latency of more than 10sec

Winneee, 2022-02-16:
Hi,
I am using kvssink on producer side for streaming with below pipeline
cmd =  "gst-launch-1.0 --gst-debug-level=1 rtspsrc do-timestamp=TRUE location=\"rtsp://admin:admin@localhost/live.sdp\" 
short-header=TRUE name=demux demux. ! rtph265depay ! h265parse ! video/x-h265,
 format=havc,alignment=au, width=1920, height=1080 ! kvssink name=sink stream-name=" + 
data['stream_name'] + " " + "access-key=" +data['access_key'] + " " + 
"secret-key=" + data['secret_key'] + " " + "aws-region=" + data['aws_region']
 + " " + "restart_on_error=false" + " " + "connection_staleness=40" +" "+"key-frame-fragmentation=false" 
+ " demux. ! queue ! rtpmp4adepay ! sink."

With this pipeline i can see delay of more than 10sec to get frames on kinesis playback.
Could you please let me know if there any parameters, i can tweak to get low latency or is this expected latency with kvs.

disa6302, 2022-02-16:
@Winneee ,

Have you tried your pipeline using sinks other than kvssink? Do you see the same latency in that?

Winneee, 2022-02-16:
No, latency is with kvssink only

hassanctech, 2022-02-16:
How do you measure latency?  Are you using the console or HLS/DASH playback?  In this case for in region you can expect 2-3x your GOP size due to the nature of HLS/DASH protocols.  The KVS console player would be similar to that.  If you have a key frame every 4s I can easily see latency hitting 10s, if you are sending data cross region then expect more latency because round trip time becomes non-negligible.  

If you want to retrieve the stream in low latency you need to call the GetMedia APIs.

disa6302, 2022-02-21:
Closing assuming answered. Feel free to reach out if you have more questions.

closed 2022-02-21 by disa6302