awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #826: [BUG]

ZTAP0011, 2022-02-18:
**Logging**
CMake Error: The source directory "D:/projects/ikashana/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/NMake Makefiles" does not exist.

**Describe the bug**
Not able to build the sdk, getting the issue CMake Error: The source directory "D:/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/NMake Makefiles" does not exist.

**To Reproduce**
Steps to reproduce the behavior:
1. Cloned hte repo.
2. Installed visual studio 2019
3. Go to visual studio developer terminal and run cmake .. -G "NMake Makefiles"

**Expected behavior**
Should build successfully

**SDK version number**
current master branch - 

**Open source building**
Running cmake .. -G "NMake Makefiles" in visual studio developer console.

**Screenshots**
[image]

**Platform (please complete the following information):**
 - OS: Windows 10
 - Intel i5
 - Windows 10 Pro

disa6302, 2022-02-21:
This could be happening due to past failures. Ensure you follow the steps on this. https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.github/build_windows.bat

Also, please use our official releases instead of master branch. https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/releases

niyatim23, 2022-02-26:
Closing assuming answered. Feel free to reach out if you have more questions.

closed 2022-02-26 by niyatim23