awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #827: Cross-compile to ARM aarch64 cmake failure

gi12345, 2022-02-25:
cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=ON
fails with:
----
checking whether we are cross compiling... configure: error: in `/home/gi/kvssdk_feb18/amazon-kinesis-video-streams-producer-sdk-cpp/open- source/liblog4cplus/build/src/project_log4cplus':
configure: error: cannot run C compiled programs.
If you meant to cross compile, use `--host'.
-----
There is no parameter in CMake to pass the --host flag. There is further no setting in CMake.txt nor CMakeCache.txt. Cross-compile, toolchain info, CXX-flags can be configured in CMakeCache.txt and are accepted by CMake, but the configure command in liblog4cplus does not get invoked with the --host parameter.
Even manually changing the configure command to include the --host parameter does not help; CMake overrides the configure command and drops the --host parameter.

niyatim23, 2022-03-01:
Hi @gi12345, we are looking into this issue. I'll update this issue once we have a fix.

niyatim23, 2022-03-02:
Hi @gi12345, can you please share the following details for the platform that you are trying to build this on:
 - OS: 
 - Processor: 
 - Version

gi12345, 2022-03-02:
OS: Ubuntu 20.04.3 LTS
Processor: Intel(R) Core(TM) i7-8850H CPU @ 2.60GHz,
A "standard" X86_64 PC with VMware running Ubuntu 20.04.3 LTS.
Building the KVS SDK for this target works.

But this is a cross-compile problem. The target is an ARM-64 processor (aarm64). An ARM supplied cross-compiler is used for cross-compiling. The host target is ARM64.
The CMake must support a target host architecture setting and pass it on to log4cpp component's configure command.

gi12345, 2022-03-02:
An alternative approach to cross-compiling might be to checkout and build the KVS-SDK on an ARM-64 platform, like a raspberry PI-4 with Ubuntu-20-ARM64 release.

niyatim23, 2022-03-02:
1. To fix the log4cplus host error, can you try replacing the line [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/CMake/Dependencies/liblog4cplus-CMakeLists.txt#L6) with `SET(CONFIGURE_COMMAND ${CMAKE_CURRENT_BINARY_DIR}/build/src/project_log4cplus/configure "CFLAGS=${CMAKE_C_FLAGS}" --prefix=${OPEN_SRC_INSTALL_PREFIX} --host=aarch64-linux-gnu)`. Does this fix the issue?
2. Are you cross-compiling gstreamer separately?

gi12345, 2022-03-02:
I will tryout the fix above.
On your question: Are you cross-compiling gstreamer separately?
My goal was the checkout the KVS with all dependencies (gstreamer, curl, openss;,...) and cross-compile everything. This guarantees that all libraries and include file are compatible.

gi12345, 2022-03-02:
This command fails:
SET(CONFIGURE_COMMAND ${CMAKE_CURRENT_BINARY_DIR}/build/src/project_log4cplus/configure "CFLAGS=${CMAKE_C_FLAGS}" --prefix=${OPEN_SRC_INSTALL_PREFIX} --host=aarch64-linux-gnu)
bash: syntax error near unexpected token `CONFIGURE_COMMAND'
------------------------------------
But this one works:
${CMAKE_CURRENT_BINARY_DIR}/build/src/project_log4cplus/configure "CFLAGS=${CMAKE_C_FLAGS}" --prefix=${OPEN_SRC_INSTALL_PREFIX} --host=aarch64-linux-gnu
checking build system type... x86_64-pc-linux-gnu
checking host system type... aarch64-unknown-linux-gnu
checking target system type... aarch64-unknown-linux-gnu
...
checking if /usr/local/linaro-aarch64-2020.09-gcc10.2-linux5.4/bin/aarch64-linux-gnu-g++ supports -c -o file.o... (cached) yes
checking whether the /usr/local/linaro-aarch64-2020.09-gcc10.2-linux5.4/bin/aarch64-linux-gnu-g++ linker (/usr/local/linaro-aarch64-2020.09-gcc10.2-linux5.4/aarch64-linux-gnu/bin/ld) supports shared libraries... yes
checking dynamic linker characteristics... (cached) GNU/Linux ld.so
checking how to hardcode library paths into programs... immediate
..
---------------------------
This is most likely just a shell problem; I am using bash and I guess you use sh or csh.
--------------------------
I had this library with manual configure command invocation built before. Now the key is to get the command integrated into the KVS-SDK so it can pass the arm64 parameter and build the entire KVS-SDK lib, like:
cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=ON .....host=arm4

or through a CMAKE parameter in CMakeCache.txt like:
CMAKE_CXX_FLAGS_DEBUG:STRING=-g

niyatim23, 2022-03-02:
We are working on the integration. Do the `cmake` and the `make` succeed for you after including this change? Are you seeing any further issues with the build?

gi12345, 2022-03-02:
Do the cmake and the make succeed for you after including this change?
No. Even when I made the liblog4cplus "by hand", (as just now) as soon as I start the:
cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=ON ..
It wipes out the existing build. I actually even changed the "configure" file to hard-code the arm64 parameter, didn't work because all the files in this directory get overridden likely on a new git-checkout. I even changed the file permission and owner to root and read-only, still wiped out by cmake.
---------------------------
 Are you seeing any further issues with the build?
I can't get further than the  liblog4cplus. When it fails the whole KSV-SDK build fails and everything has to start again. 
I peeked ahead and used a previous non-cross compiled library to see if it would cross-compile. It failed very early in an include file data-type definition which seems not arm64 compliant. That raises some warning light, but we have to overcome the liblog4cplus problem first.
---
In non-cross-compile case  it built fine, created all KVS-SDK .so libs.

gi12345, 2022-03-03:
Just a "heads-up". Applied your change above, passes liblog4cplus, fails in building openssl:
-----------------
...
[ 66%] Performing build step for 'project_libopenssl'
aarch64-linux-gnu-gcc: error: unrecognized command-line option '-m64'
...
... log details below ...
------------------
Looks like a similar configure problem as in liblog4cplus. From quickly scanning the configure options the likely parameter should be linux-aarch64.
--------------------------------------------------------
I will checkout brand new and build everything from scratch. It will take quite a while because every checkout wipes out all cross-compile info in CmakeCache.txt and requires manual re-edit.

=============== Log =============================
-- Kinesis Video Cpp Producer path is /home/gi/kvssdk_feb18/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /home/gi/kvssdk_feb18/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- autoconf already built
-- automake already built
-- log4cplus already built
-- Finished building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gi/kvssdk_feb18/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer
[ 11%] Performing update step for 'libkvscproducer-download'
[ 22%] No configure step for 'libkvscproducer-download'
[ 33%] No build step for 'libkvscproducer-download'
[ 44%] No install step for 'libkvscproducer-download'
[ 55%] No test step for 'libkvscproducer-download'
[ 66%] Completed 'libkvscproducer-download'
[100%] Built target libkvscproducer-download
-- Kinesis Video Producer path is /home/gi/kvssdk_feb18/amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src
-- Begin building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: /home/gi/kvssdk_feb18/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl
Scanning dependencies of target project_libopenssl
[ 11%] Creating directories for 'project_libopenssl'
[ 22%] Performing download step (git clone) for 'project_libopenssl'
Cloning into 'project_libopenssl'...
Note: switching to 'OpenSSL_1_1_1g'.

You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.

If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:

  git switch -c <new-branch-name>

Or undo this operation with:

  git switch -

Turn off this advice by setting config variable advice.detachedHead to false

HEAD is now at e2e09d9fba Prepare for 1.1.1g release
Submodule 'boringssl' (https://boringssl.googlesource.com/boringssl) registered for path 'boringssl'
Submodule 'krb5' (https://github.com/krb5/krb5) registered for path 'krb5'
Submodule 'pyca.cryptography' (https://github.com/pyca/cryptography.git) registered for path 'pyca-cryptography'
Cloning into '/home/gi/kvssdk_feb18/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/boringssl'...
Cloning into '/home/gi/kvssdk_feb18/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/krb5'...
Cloning into '/home/gi/kvssdk_feb18/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/libopenssl/build/src/project_libopenssl/pyca-cryptography'...
Submodule path 'boringssl': checked out '2070f8ad9151dc8f3a73bffaa146b5e6937a583f'
Submodule path 'krb5': checked out '890ca2f401924cdcb88f493950b04700bbe52db3'
Submodule path 'pyca-cryptography': checked out '09403100de2f6f1cdd0d484dcb8e620f1c335c8f'
[ 33%] No patch step for 'project_libopenssl'
[ 44%] Performing update step for 'project_libopenssl'
[ 55%] Performing configure step for 'project_libopenssl'
Operating system: x86_64-whatever-linux2
[... 8 lines cut ...]
***                                                                ***
***   If you encounter a problem while building, please open an    ***
***   issue on GitHub <https://github.com/openssl/openssl/issues>  ***
***   and include the output from the following command:           ***
***                                                                ***
***       perl configdata.pm --dump                                ***
***                                                                ***
***   (If you are new to OpenSSL, you might want to consult the    ***
***   'Troubleshooting' section in the INSTALL file first)         ***
***                                                                ***
**********************************************************************
[ 66%] Performing build step for 'project_libopenssl'
aarch64-linux-gnu-gcc: error: unrecognized command-line option '-m64'
make[4]: *** [Makefile:708: apps/app_rand.o] Error 1
make[3]: *** [Makefile:174: all] Error 2
make[2]: *** [CMakeFiles/project_libopenssl.dir/build.make:113: build/src/project_libopenssl-stamp/project_libopenssl-build] Error 2
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/project_libopenssl.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
CMake Error at dependency/libkvscproducer/kvscproducer-src/CMake/Utilities.cmake:93 (message):
  CMake step for libopenssl failed: 2
Call Stack (most recent call first):
  dependency/libkvscproducer/kvscproducer-src/CMakeLists.txt:66 (build_dependency)

-- Configuring incomplete, errors occurred!
See also "/home/gi/kvssdk_feb18/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".

gi12345, 2022-03-03:
Tried with brand new checkout from scratch. Same result as above. Log:
[result1.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/8180986/result1.txt)
.

gi12345, 2022-03-07:
Any idea when the OpenSSL problem, logged above, will be addressed?  I am stuck am on it. This is really part of the ARM aarch64 build problem.

gi12345, 2022-03-07:
I can't see a linked PR. 
I applied above fix, it fixed the project_libopenssl problem, then it failed in openssl as logged twice above.

disa6302, 2022-03-07:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/831

disa6302, 2022-03-07:
@gi12345 ,

Yes it is an old article. I included the link because it has some details in it. Please try out the changes in the PR linked. Openssl is a dependency of KVS build, but it is difficult to provide targeted instructions for all the different toolchains available out there. If there is something different with a toolchain that we discover, we would include instructions in the README related to it. However, given the PR builds pass with `linux-generic64`, we are pointing you to the steps we have used to cross compile for ARM. I checked the log file you attached but it does not have any details to point out the source of the error. Kindly try out both the toolchains (linux-generic64 and linux-aarch64). You can use the branch in the PR and clean build). Run it in VERBOSE mode, it might yield more logs.

For your reference, I also built cross compiled the SDK using `linux-aarch64` on an Ubuntu EC2 instance for verification and it works fine. I verified it the following way:

```
$file libKinesisVideoProducer.so
libKinesisVideoProducer.so: ELF 64-bit LSB shared object, ARM aarch64, version 1 (GNU/Linux), dynamically linked, BuildID[sha1]=b0997722692123f4afe5e9b8ea1e969148c33d74, not stripped
```

gi12345, 2022-03-07:
"...For your reference, I also built cross compiled the SDK using linux-aarch64 on an Ubuntu EC2 instance for verification and it works fine. I verified it the following way:..."
What Ubuntu operating system is running on EC2? Is it a ARM version?

disa6302, 2022-03-07:
It is x86_64

niyatim23, 2022-03-10:
Closing since the issue has been fixed, feel free to reopen / ask further questions if needed

closed 2022-03-10 by niyatim23