awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #829: [QUESTION] How to use the sdk as dependency

CY-1992, 2022-03-01:
There is [kvs_gstreamer_sample.cpp](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/samples/kvs_gstreamer_sample.cpp) in samples directory. I want to customize this sample and integrate it with my other project. So I am looking for a way to use the sdk as dependency. **But I don't find any header files of SDK in /usr/local/include or /usr/include after installation.** 

And Is there anyway we can use the sdk in CMAKE as shown in below

```cmake
cmake_minimum_required(VERSION 3.1)
# note: cxx-17 requires cmake 3.8, cxx-20 requires cmake 3.12
project(basic-pub-sub CXX)

file(GLOB SRC_FILES
       "*.cpp"
)

add_executable(${PROJECT_NAME} ${SRC_FILES})

set_target_properties(${PROJECT_NAME} PROPERTIES
    CXX_STANDARD 14)

find_package(aws-crt-cpp REQUIRED)
find_package(KinesisVideoProducer REQUIRED)

target_link_libraries(${PROJECT_NAME} KinesisVideoProducer AWS::aws-crt-cpp)
```

disa6302, 2022-03-21:
@CY-1992 , 

We do not install to /usr/include by default. It is a bug in our build system. https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/issues/809. The workaround would be install in the open-source directory by default and setting the search path to that. Or, you can set up your own path and cache CMAKE_PREFIX_PATH.

closed 2022-03-21 by disa6302

CY-1992, 2022-03-24:
@disa6302 I do not understand completely. It would be great if you could explain it in more detail. Also I try to use kinesis producer sdk as git submodule. Added as sub directory in cmakelists.txt.

`add_subdirectory(${CMAKE_CURRENT_SOURCE_DIR}/kinesisProducer EXCLUDE_FROM_ALL)`

I have also included the directories so that I could use the sdk in my project.

`include_directories(
    "${CMAKE_SOURCE_DIR}/src/kinesisProducer/include",
    "${CMAKE_SOURCE_DIR}/src/kinesisProducer/src",
    "${CMAKE_SOURCE_DIR}/src/kinesisProducer/src/credential-providers",
    "${CMAKE_SOURCE_DIR}/src/kinesisProducer/src/common",
    "${CMAKE_SOURCE_DIR}/src/kinesisProducer/src/JNI/include"
)`

But compilation throws following error 
`fatal error: com/amazonaws/kinesis/video/client/Include.h: No such file or directory
    5 | #include "com/amazonaws/kinesis/video/client/Include.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.`

BestITUserEUW, 2022-03-30:
@CY-1992 having the same issue here. It can't find the client because it doesn't exist. 
`com/amazonaws/kinesis/video/client/Include.h`
is a dependency of: 
https://github.com/awslabs/amazon-kinesis-video-streams-producer-c/tree/v1.3.0/src/include/com/amazonaws/kinesis/video
but as you can see in the Picture below there is no client folder. @disa6302 how should we include files that doesn't exist?
[image]

BestITUserEUW, 2022-03-30:
@CY-1992 you have to add the header files from kvspic too.
[image]