awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #830: [QUESTION] How to reduce latency without re-encoding stream (i.e. disregarding keyframes)?

mattfeury, 2022-03-02:
Hello,

When working with an RTSP stream and pumping to KVS, it's frequent to have a desire for minimal latency. Looking at the settings now, it seems that fragments are shipped up to the cloud based on fragment_size and keyframe frequency. For all intents and purposes, I view latency as a time from when the stream is ingested to the pipeline to when it's available in the KVS server/console.

When dealing with an h264 stream, it's often more performant to not decode the stream and re-encode it before sending to KVS, when one could just send it to KVS directly. however, this means we are subject to whatever the keyframe interval on the stream is.  Is there any recommendation for how to force the fragment size and/or modify the key frame interval of the originating stream in the pipeline?

For example, a simple pipeline could look like:
```
gst-launch-1.0 rtspsrc location="$url" ! rtph264depay ! h264parse ! kvssink stream-name="$name" storage-size=512 access-key="$access_key" secret-key="$secret_key" aws-region="$region"
```

In this instance, regardless of how I modify parameters like max-latency, fragment-duration, buffer-duration, etc, the stream is only going to upload to the cloud when a keyframe occurs, which I have no control over.

I could obviously decodebin -> x264enc, but this seems like a waste of CPU.

are there any thoughts around accomplishing this?

thanks!

MushMal, 2022-03-02:
The HLS/MPEG-DASH and console playback are what's known as live playbacks and not realtime playbacks. It would inherently at least the duration of the fragment latency built in. If you are looking at a realtime playback perhaps you can look at GetMedia API which can give you bit-level granularity but you need to implement your own playback engine or use an elementary stream player from somewhere,.

mattfeury, 2022-03-02:
hey @MushMal thanks for the quick response! would you mind going into a little more re live vs realtime? to me they sound fairly synonymous, so if there's a distinction i've love to learn more about it! 

i figured that the kvssink didn't upload things until a fragment was finished (aka a keyframe occurred), so regardless of how it's played back by the web player, the latency of all uploads would still be bottlenecked by the keyframe interval, right?

MushMal, 2022-03-03:
kvssink (and the producer SDK which it sits on top of) are like a filter in the media pipeline. On the left, it receives frames (frame granularity) and on the right (output) it sends at a byte granularity. It tries to stream the bits as fast as possible given the network conditions (modulo the upper bounds on the throughput on the backend servers which should be very high and last time I checked it was in 100Mbps).

The RTSP stream operates on a frame/byte granularity by default so I would assume your RTSP upstream streams/receives as soon as the frame is produced and encoded on the encoder on the camera side. From that moment on, it would produce the frame into the kvssink/producer and the frame will be packaged into a streamable MKV format (zero latency) and streamed out as soon as possible. If you have high bandwidth with low latency, the frame data will spend no time in the KVS producer buffer.

The main latency comes on the playback side. live playback is based on fragmented playback which means that the player should have at least a fragment worth of frames in order to begin the decoding and playback. If you have a fragment which is 4 seconds long then there will be at least 4 seconds delay on the playback side. 

On the contrary, the realtime playback is not based on the fragment granularity but rather on a frame/slice granularity. The MKV packaging of the KVS stream is "streamable" which allows production and consumption on an MKV element granularity which in this case is a frame. Given the realtime path in the KVS end-to-end, the produced frames into the KVS producer SDK (kvssink) and high bandwidth with low latency, the frame bits will be sent to the ingress endpoint and with very low latency will be available at the egress endpoint for the realtime consumption. GetMedia API is used to access the realtime stream.

Not sure if this covers your question - please chime in.

disa6302, 2022-03-09:
Closing assuming answered

closed 2022-03-09 by disa6302

mattfeury, 2022-03-23:
thanks @MushMal. appreciate the context. it looks like most of the [sample codes](https://github.com/aws-samples/amazon-kinesis-video-streams-web-viewer) leverage `getHLSStreamingSessionURL` to show a "realtime" feed. do you have any hints/examples/pro-tips about how one could leverage getMedia via a web frontend?

MushMal, 2022-03-23:
Unfortunately, there are no ready APIs for elementary stream playback in the web sandbox. Web browsers expose MSE interface which operates at a fragment granularity. You might want to consider RTP type of streaming. Either your source could act as a WebRTC master or you can have an application in the cloud that can call GetMedia and act as a webrtc master peer for your browser app.