awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #832: [QUESTION] errors occurred in the this command "cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI-TRUE"

SeowonPark001, 2022-03-05:
According to the explanation [READ ME], an error occurred while conducting the kinesis video stream producer tutorial, but I asked a question because I couldn't see a solution even after googling for a long time.
Please let me know how to solve the following errors.

Command:
$ cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI-TRUE

Errors:
Glibc 2.9 - 2.12 and GNU M4 1.4.11 - 1.4.15 have another strstr bug.
make[2]: *** [CMakeFiles/project_libautoconf.dir/build.make:111: build/src/project_libautoconf-stamp/project_libautoconf-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/project_libautoconf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
CMake Error at CMake/Utilities.cmake:99 (message):
  CMake step for libautoconf failed: 2
Call Stack (most recent call first):
  CMakeLists.txt:73 (build_dependency)

For your information, before executing this command, the sudo apt upgrade was finished. And a cmake_cxx_compiler error occurred so the command "sudo apt install g++" was executed.

Below is the result of executing the command.

-- The CXX compiler identification is GNU 9.3.0
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Kinesis Video Cpp Producer path is /mnt/c/Users/user/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp
-- dependencies install path is /mnt/c/Users/user/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/open-source
-- Begin building dependencies.
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/c/Users/user/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/libautoconf
Scanning dependencies of target project_libautoconf
[ 11%] Creating directories for 'project_libautoconf'
[ 22%] Performing download step (download, verify and extract) for 'project_libautoconf'
-- Downloading...
   dst='/mnt/c/Users/user/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/libautoconf/build/src/autoconf-2.69.tar.gz'
   timeout='none'
-- Using src='http://ftp.gnu.org/gnu/autoconf/autoconf-2.69.tar.gz'
-- [download 100% complete]
-- Downloading... done
-- extracting...
     src='/mnt/c/Users/user/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/libautoconf/build/src/autoconf-2.69.tar.gz'
     dst='/mnt/c/Users/user/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/libautoconf/build/src/project_libautoconf'
-- extracting... [tar xfz]
-- extracting... [analysis]
-- extracting... [rename]
-- extracting... [clean up]
-- extracting... done
[ 33%] No patch step for 'project_libautoconf'
[ 44%] No update step for 'project_libautoconf'
[ 55%] Performing configure step for 'project_libautoconf'
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking build system type... x86_64-unknown-linux-gnu
checking host system type... x86_64-unknown-linux-gnu
configure: autobuild project... GNU Autoconf
configure: autobuild revision... 2.69
configure: autobuild hostname... 5d2hk
configure: autobuild timestamp... 20220305T182721Z
checking whether /bin/sh -n is known to work... no
checking for characters that cannot appear in file names... none
checking whether directories can have trailing spaces... yes
checking for expr... /usr/bin/expr
checking for GNU M4 that supports accurate traces... configure: error: no acceptable m4 could be found in $PATH.
GNU M4 1.4.6 or later is required; 1.4.16 or newer is recommended.
GNU M4 1.4.15 uses a buggy replacement strstr on some systems.
Glibc 2.9 - 2.12 and GNU M4 1.4.11 - 1.4.15 have another strstr bug.
make[2]: *** [CMakeFiles/project_libautoconf.dir/build.make:111: build/src/project_libautoconf-stamp/project_libautoconf-configure] Error 1
make[1]: *** [CMakeFiles/Makefile2:76: CMakeFiles/project_libautoconf.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
CMake Error at CMake/Utilities.cmake:99 (message):
  CMake step for libautoconf failed: 2
Call Stack (most recent call first):
  CMakeLists.txt:73 (build_dependency)

-- Configuring incomplete, errors occurred!
See also "/mnt/c/Users/user/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
See also "/mnt/c/Users/user/Downloads/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeError.log".

niyatim23, 2022-03-06:
Hi @SeowonPark001, have you installed all the dependencies mentioned in the [README](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/develop#configure)? I'm asking because you have not mentioned it as a part of the steps before the execution. If that still does not work for you, can you share details about your platform and toolchain?

disa6302, 2022-03-21:
Closing due to inactivity.

closed 2022-03-21 by disa6302

CW3MH6, 2022-11-15:
Had same issue. For future readers that are searching this error, ```sudo apt install m4``` and ```sudo apt upgrade``` fixed it. Missed it in the docs the first time.