awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #836: Library libproducer.so missing

gi12345, 2022-03-24:
libproducer.so missing

After new checkout and build of amazon-kinesis-video-streams-producer-sdk-cpp the expected library producer.so (which holds linkage callback symbols) is missing:
HW/SW: AWS-EC2 with Ubuntu 20.04 AARM64 pre-installed
 git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
 mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build
 cd amazon-kinesis-video-streams-producer-sdk-cpp/build
 cmake ..
 make

Note: there is library ./build/dependency/libkvscproducer/kvscproducer-src/libcproducer.so but not library producer.so.
When I build and link my application I get linker symbol errors like:
/usr/bin/ld: Cloud/KVSVideoStreamService.o:(.data.rel.ro._ZTVN3com9amazonaws7kinesis5video24SampleCredentialProviderE[_ZTVN3com9amazonaws7kinesis5video24SampleCredentialProviderE]+0x50): undefined reference to `com::amazonaws::kinesis::video::CredentialProvider::getStreamingTokenCallback()'

Does the SDK need some special CMAKE parameters to output a producer.so library?
--------

I browsed the source code and found the callback location:
amazon-kinesis-video-streams-producer-sdk-cpp/src/CallbackProvider.cpp
    callbacks_.getStreamingEndpointFn = getStreamingEndpointCallback();
    callbacks_.getStreamingTokenFn = getStreamingTokenCallback();

I have an "old" producer.so where the link symbols are defined:    
nm -D libproducer.so | grep etStreamingTokenCallback
0000a881 T _ZN3com9amazonaws7kinesis5video18CredentialProvider25getStreamingTokenCallbackEv
0000bf3d T _ZN3com9amazonaws7kinesis5video23DefaultCallbackProvider25getStreamingTokenCallbackEv

disa6302, 2022-03-28:
@gi12345 ,

What version of the SDK is being used? 
What was being used before?

gi12345, 2022-03-28:
KVS version is from last week. I tried a couple of versions since March, all the same.
The version used before is about 2 years old. The person who built it is no longer with us; so I don't know the exact date. But libproducer.so was a critical library as I contains symbols (and thus code) still in use. I looked through the library source code and couldn't find anything like that. I guess it is possible that the library has moved to some other AWS place, but it must still exist somewhere because the symbols are still in use. The callback file above amazon-kinesis-video-streams-producer-sdk-cpp/src/CallbackProvider.cpp has the symbols (calls).

disa6302, 2022-03-28:
@gi12345 ,

Please use official release version. We have a new release out and the builds have been passing in our CI. If the library used was 2 years ago, chances are that you are using a version old version which most likely had a different build system. I would suggest cleaning the local repo and re-building. Without knowing the previous version used, we cannot help much here.

How are you using this project? Are you building it to another directory?

gi12345, 2022-03-28:
The current version I am using is new, checked out last week. The symbol errors come from application code which use this checkout.
From the old version I only have the library. I inspected the symbols and found the unresolved ones there. So there must be some place in the current SDK which resolves the symbols (maybe at some other place).

gi12345, 2022-03-29:
Still trying to gather more info.

gi12345, 2022-03-29:
After linking my app with the KVS-SDK I get the following linkage errors which traverse the AWS Kinesis SDK.  Any help would be appreciated.

[link] ... 
---------------------
""g++ -D_S5 -g -fdata-sections -ffunction-sections  -fPIC  -rdynamic -march=armv8-a -Wl,-rpath,/mnt/plugin/vigilcamera/lib::/home/felix/work/cpro/ThirdPart/S3/kvs/lib -o VIGILCameraServer AppServer/VIGILCameraServer.o NAT/NATConnect.o NAT/NATDVRSessionBase.o NAT/NATDVRSessions.o NAT/NATDVRSessionTCP.o NAT/NATMessage.o NAT/NATTraversalDVR.o NAT/NATUtil.o NAT/VigilConnectDVR.o NAT/VigilEncryption.o NAT/VigilDecryption.o CameraProfile/UDPSettings.o Common/Runable.o Common/Base64.o Common/DvrUtility.o Common/Event.o Common/UPnPHelper.o Common/SimpleSocket.o Common/Tool.o Common/CMutex.o Common/CurlHttpRequest.o Common/CcURLWrapper.o Common/FastParser.o Common/FastRequest.o Common/General.o Common/UDPUtil.o Common/FastRequestEmbeddedEncoder.o Common/DateTime.o Common/CheckSum.o Common/IPCameraUtil.o Common/Membuf.o Common/StringTokenizer.o Common/xmlParser.o Common/CTimeUtil.o Common/ConfigUtil.o Common/UDPSocket.o Common/CodecUtil.o Common/RollOutLog.o DataPortCmds/AccessControlHandler.o DataPortCmds/CheckServerHandler.o DataPortCmds/DataConnect.o DataPortCmds/DataConnectSession.o DataPortCmds/FootageLoader.o DataPortCmds/LoginHandler.o DataPortCmds/LoaderWrapper.o DataPortCmds/SetServerSettingsHandler.o DataPortCmds/SyncTimeHandler.o DataPortCmds/SysInfoLoader.o DataPortCmds/SystemRebootHandler.o DataPortCmds/UsersManagement/UsersManagement.o DataPortCmds/UsersManagement/LoginFailedAccount.o DataPortCmds/UsersManagement/LoginFailedAccounts.o DataPortCmds/GetAudioChannelInfoHandler.o DataPortCmds/GetPortSettingsHandler.o DataPortCmds/GetCameraNamesHandler.o DataPortCmds/GetCameraNamesForSearchHandler.o DataPortCmds/GetServerSettingsHandler.o DataPortCmds/GetSettingsChangeTimeHandler.o DataPortCmds/GetInfoforHMonitorHandler.o DataPortCmds/GetTimeStampHandler.o DataPortCmds/GetRelayStatusHandler.o HTTP/DvrPorts.o HTTP/HTTPLiveImage.o HTTP/HTTPPlaybackManager.o HTTP/HTTPPlaybackSession.o HTTP/HTTPRequest.o HTTP/HTTPResponse.o HTTP/HTTPAuthenticationManager.o HTTP/HTTPDThrottleControl.o HTTP/RemoteSessionHTTP.o HTTP/RemoteSessionPort.o Common/mytop/m_linux.o Common/mytop/utils.o ThirdPart/S5/xml/tinystr.o ThirdPart/S5/xml/tinyxml.o ThirdPart/S5/xml/tinyxmlerror.o ThirdPart/S5/xml/tinyxmlparser.o ThirdPart/S5/xml/TXML.o ThirdPart/S5/json/src/json_reader.o ThirdPart/S5/json/src/json_value.o ThirdPart/S5/json/src/json_writer.o ThirdPart/S5/cpro/LibUDS_Event_Client.o Playback/RemoteSession.o Playback/DvrDaemon.o Playback/DvrFrame.o Playback/DvrFileIndex.o Playback/DvrFrameIndex.o Playback/UDPFrameIndexFile.o RemoteReg/Register.o RemoteReg/VIGILDLL.o RemoteReg/LicenseChecker.o Monitor/UDPConnectUtil.o Monitor/AutoControl.o Monitor/ConnectStatusCheck.o Monitor/MainHTTPCheck.o Monitor/PlaybackHTTPCheck.o Monitor/KVSCheck.o CameraReset/AbstractResetCommand.o CameraReset/DefaultMainStreamStrategy.o CameraReset/DefaultResetCommand.o CameraReset/DefaultSubStreamStrategy.o CameraReset/DefaultThirdStreamStrategy.o CameraReset/OSDStrategy.o CameraReset/ResetCommandContext.o CameraReset/ResetStrategy.o CameraReset/AutoIPStrategy.o CameraReset/RecordSettingsStrategy.o CameraReset/ActionRulesStrategy.o CameraReset/EventRulesStrategy.o CameraReset/TimeStrategy.o Cloud/ThumbnailUploader.o Cloud/MotionUploader.o Cloud/MeasurementClient.o Cloud/KVSVideoStreamService.o Cloud/KVSFrameBufferHandler.o Cloud/ClipUploadSessionManager.o Cloud/ClipUploadSession.o Cloud/MotionNotifications.o Cloud/CloudUtils.o Cloud/BandwidthTestSessionManager.o Cloud/BandwidthTestSession.o -L. -L./ThirdPart/S5/curl -L./ThirdPart/S5/openssl -L./ThirdPart/S5/zlib -L./ThirdPart/S5/reclib -L./ThirdPart/S5/hik -L./ThirdPart/S5/rtsp -L./ThirdPart/S5/aws -L./ThirdPart/S5/kvs/lib -L./ThirdPart/S5/sdcard -lpthread -lcurl -lssl -lcrypto -lz -ldl -lrt -laws-cpp-sdk-core -laws-cpp-sdk-s3 -laws-checksums -laws-c-common -laws-c-event-stream -lcproducer -llog4cplus -lglib-2.0 -lgstreamer-1.0 -lgobject-2.0 -lgstapp-1.0 -lgstbase-1.0 -lrecordingfs

 ------ Linkage Errors --------------------
/usr/bin/ld: Cloud/KVSVideoStreamService.o: in function `com::amazonaws::kinesis::video::StaticCredentialProvider::StaticCredentialProvider(com::amazonaws::kinesis::video::Credentials const&)':
/home/ubuntu/kvssdk_march09/amazon-kinesis-video-streams-producer-sdk-cpp/src/Auth.h:209: undefined reference to `com::amazonaws::kinesis::video::CredentialProvider::CredentialProvider()'
/usr/bin/ld: /home/ubuntu/kvssdk_march09/amazon-kinesis-video-streams-producer-sdk-cpp/src/Auth.h:209: undefined reference to `com::amazonaws::kinesis::video::CredentialProvider::~CredentialProvider()'
/usr/bin/ld: Cloud/KVSVideoStreamService.o: in function `com::amazonaws::kinesis::video::IotCertCredentialProvider::IotCertCredentialProvider(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
[... cut ...]
/home/ubuntu/di_mar16/VIGIL_Edge_Cloud/CPROVIGILCameraServer/Cloud/KVSVideoStreamService.h:277: undefined reference to `com::amazonaws::kinesis::video::systemCurrentTime()'
/usr/bin/ld: Cloud/KVSVideoStreamService.o: in function `com::amazonaws::kinesis::video::SampleDeviceInfoProvider::getDeviceInfo()':
/home/ubuntu/di_mar16/VIGIL_Edge_Cloud/CPROVIGILCameraServer/Cloud/KVSVideoStreamService.h:287: undefined reference to `com::amazonaws::kinesis::video::DefaultDeviceInfoProvider::getDeviceInfo()'

gi12345, 2022-03-31:
I have reproduced the same error today with a brand new checkout/build/link:
cd ~
mkdir kvssdk_mar30
cd kvssdk_mar30
git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build
cd amazon-kinesis-video-streams-producer-sdk-cpp/build
cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_DEPENDENCIES=ON >result1 2>&1
make >result2 2>&1
make install >result3 2>&1
....................................................................................................................
 Libraries generated are:
gi@ubuntu:~/kvssdk_mar30/amazon-kinesis-video-streams-producer-sdk-cpp$ find . -name *.so*
./open-source/local/lib/liblog4cplus.so
./open-source/local/lib/liblog4cplus-1.2.so.5
./open-source/local/lib/libssl.so.1.1
./open-source/local/lib/libcurl.so
./open-source/local/lib/engines-1.1/padlock.so
./open-source/local/lib/engines-1.1/capi.so
./open-source/local/lib/engines-1.1/afalg.so
./open-source/local/lib/libcrypto.so.1.1
./open-source/local/lib/liblog4cplus-1.2.so.5.1.7
./open-source/local/lib/libssl.so
./open-source/local/lib/libcrypto.so
./build/libKinesisVideoProducer.so
./build/libgstkvssink.so
./build/dependency/libkvscproducer/kvscproducer-src/libcproducer.so

gi@ubuntu:~/kvssdk_mar30/amazon-kinesis-video-streams-producer-sdk-cpp$ find . -name *.a
./open-source/local/lib/libcrypto.a
./open-source/local/lib/libssl.a
./build/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/libkvspicUtils.a
./build/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/libkvspic.a
./build/dependency/libkvscproducer/kvscproducer-src/libkvsCommonCurl.a

................................................................................................................
My app links in all these libraries explicitly:
...
    111            -L${KVS_SDK_PATH}/open-source/local/lib \
    112            -L${KVS_SDK_PATH}/build \
    113            -L${KVS_SDK_PATH}/build/dependency/libkvscproducer/kvscproducer-src \
    114            -L${KVS_SDK_PATH}/build/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/ \
    115            -L${KVS_SDK_PATH}/build/dependency/libkvscproducer/kvscproducer-src \
...
................................................................................................................
Error gets produced (same as above):
/usr/bin/ld: /home/gi/kvssdk_mar30/amazon-kinesis-video-streams-producer-sdk-cpp/src/Auth.h:209: undefined reference to `com::amazonaws::kinesis::video::CredentialProvider::~CredentialProvider()'
/usr/bin/ld: Cloud/KVSVideoStreamService.o: in function `com::amazonaws::kinesis::video::IotCertCredentialProvider::IotCertCredentialProvider(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >)':
/home/gi/kvssdk_mar30/amazon-kinesis-video-streams-producer-sdk-cpp/src/credential-providers/IotCertCredentialProvider.h:25: undefined reference to `com::amazonaws::kinesis::video::CredentialProvider::CredentialProvider()'
/usr/bin/ld: /home/gi/kvssdk_mar30/amazon-kinesis-video-streams-producer-sdk-cpp/src/credential-providers/IotCertCredentialProvider.h:25: undefined reference to `vtable for com::amazonaws::kinesis::video::IotCertCredentialProvider'
/usr/bin/ld: /home/gi/kvssdk_mar30/amazon-kinesis-video-streams-producer-sdk-cpp/src/credential-providers/IotCertCredentialProvider.h:25: undefined reference to `com::amazonaws::kinesis::video::CredentialProvider::~CredentialProvider()'
/usr/bin/ld: Cloud/KVSVideoStreamService.o: in function `com::amazonaws::kinesis::video::StaticCredentialProvider::~StaticCredentialProvider()':
/home/gi/kvssdk_mar30/amazon-kinesis-video-streams-producer-sdk-cpp/src/Auth.h:206: undefined reference to `com::amazonaws::kinesis::video::CredentialProvider::~CredentialProvider()'
/usr/bin/ld: Cloud/KVSVideoStreamService.o: in function `com::amazonaws::kinesis::video::SampleCredentialProvider::updateCredentials(com::amazonaws::kinesis::video::Credentials&)':
/home/gi/di_mar30/VIGIL_Edge_Cloud/CPROVIGILCameraServer/Cloud/KVSVideoStreamService.h:278: undefined reference to `com::amazonaws::kinesis::video::systemCurrentTime()'
/usr/bin/ld: Cloud/KVSVideoStreamService.o: in function `com::amazonaws::kinesis::video::SampleDeviceInfoProvider::getDeviceInfo()':
/home/gi/di_mar30/VIGIL_Edge_Cloud/CPROVIGILCameraServer/Cloud/KVSVideoStreamService.h:288: undefined reference to `com::amazonaws::kinesis::video::DefaultDeviceInfoProvider::getDeviceInfo()'
/usr/bin/ld: Cloud/KVSVideoStreamService.o: in function `on_new_sample(_GstElement*, _CustomData*)':
/home/gi/di_mar30/VIGIL_Edge_Cloud/CPROVIGILCameraServer/Cloud/KVSVideoStreamService.cpp:595: undefined reference to `com::amazonaws::kinesis::video::KinesisVideoStream::start(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, unsigned long)'
/usr/bin/ld: /home/gi/di_mar30/VIGIL_Edge_Cloud/CPROVIGILCameraServer/Cloud/KVSVideoStreamService.cpp:623: undefined reference to `com::amazonaws::kinesis::video::systemCurrentTime()'
[... cut ...]
/usr/bin/ld: Cloud/KVSVideoStreamService.o:(.data.rel.ro._ZTIN3com9amazonaws7kinesis5video24StaticCredentialProviderE[_ZTIN3com9amazonaws7kinesis5video24StaticCredentialProviderE]+0x10): undefined reference to `typeinfo for com::amazonaws::kinesis::video::CredentialProvider'
/usr/bin/ld: Cloud/KVSFrameBufferHandler.o: in function `CKVSFrameBufferHandler::AddPrebufferFrames(unsigned long const&, std::shared_ptr<com::amazonaws::kinesis::video::KinesisVideoStream>)':
/home/gi/di_mar30/VIGIL_Edge_Cloud/CPROVIGILCameraServer/Cloud/KVSFrameBufferHandler.cpp:310: undefined reference to `com::amazonaws::kinesis::video::KinesisVideoStream::putFrame(Frame) const'

gi12345, 2022-04-01:
Any idea what the problem could be? Thanks.

gi12345, 2022-04-01:
Found the root cause of the problem:
The kvs-sdk has 2 Auth.h files which are different:
amazon-kinesis-video-streams-producer-sdk-cpp/src/Auth.h
amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/src/source/Common/Auth.h
It's this file: amazon-kinesis-video-streams-producer-sdk-cpp/src/Auth.h which must be included, not the second one. Then the authentication related problems disappear.
Closing this is issue.

closed 2022-04-01 by gi12345