awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #840: [BUG] arm64 cross compile with Build Dependencies fails

BestITUserEUW, 2022-03-29:
**Describe the bug**
Build fails at libopenssl when trying to cross compile for arm64.
`/bin/sh: 1: aarch64-linux-musl-/usr/xcc/aarch64-linux-musl-cross/bin/aarch64-linux-musl-gcc: not found`

**To Reproduce**
Steps to reproduce the behavior:
1. clone https://github.com/dockcross/dockcross.git
2. install linux-arm64-musl cross compile docker
3. build amazon-kinesis-video-streams-producer-sdk-cpp as suggested in the CI but with Build Dependencies
4. See error at libopenssl build step: `/bin/sh: 1: aarch64-linux-musl-/usr/xcc/aarch64-linux-musl-cross/bin/aarch64-linux-musl-gcc: not found`

**Expected behavior**
- libopenssl fails to build.

**SDK version number**
- v3.3.0

**Open source building**
CMake Flags for building as used in CI for [arm64/aarch64]:

> "-DCMAKE_INSTALL_PREFIX:PATH=${CMAKE_PREFIX_PATH}"
> "-DCMAKE_TOOLCHAIN_FILE:PATH=${CMAKE_TOOLCHAIN_FILE}"
> "-DCMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE}"
> "-DBUILD_STATIC=ON"
> "-DBUILD_OPENSSL_PLATFORM=linux-aarch64"
> "-DBUILD_LOG4CPLUS_HOST=arm-linux"
> 

**Platform:**
 - OS: Linux
 - Processor: arm64

**Additional Info**
- Also tried compiling without musl same error
- building other third party libs works
- log4cplus build was succesfull
- C / CXX Compiler GNU 11.2.0

**Screenshots**
[image]

niyatim23, 2022-04-07:
Hi @BestITUserEUW, the errors in the screenshot and your description are different. The error in the screenshot says that the toolchain was not found. I would suggest you to first install the toolchain in the correct location and build openssl independently to figure out where does the issue lie. Also, I would recommend you to use the toolchains used the CI for this repository. It can be found [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/.github/workflows/ci.yml#L321-L340)

disa6302, 2022-04-24:
Closing due to inactivity.

closed 2022-04-24 by disa6302