awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #843: [BUG]

Akill047, 2022-04-02:
**Logging**
Add relevent SDK logging.  IMPORTANT NOTE:  Please make sure to NOT share AWS access credentials under any circumstance!  Please make sure they are not in the logs.

**Describe the bug**
Couldn't able to build cmake with Gstreamer and JNI without any errors.

**To Reproduce**
The following are the exact commands performed one later the other
Steps to reproduce the behavior:
Installing libraries needed to build
-Run the following commands to install the prerequisite libraries to get started:

1. $ sudo apt-get install cmake m4 git build-essential 
2. $ sudo apt-get install gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-plugins-ugly gstreamer1.0-tools

-Also, install the gstreamer1.0-omx package to get the omxh264enc hardware encoder:

3. $ sudo apt-get install gstreamer1.0-omx

-If building JNI, ensure Java JDK is available on the system. To check this, run: java -showversion If nothing shows up, install OpenJDK.

4. $  sudo apt-get install openjdk-8-jdk

Build Download

5. $ git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git

-Configure
-Prepare a build directory in the newly checked out repository:

6. $ mkdir -p amazon-kinesis-video-streams-producer-sdk-cpp/build
7. $ cd amazon-kinesis-video-streams-producer-sdk-cpp/build

GStreamer and JNI is NOT built by default, if you wish to build both you MUST execute the following with the cmake.

8. $ cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE

**Expected behavior**

The following is the output for my command execution of { $ cmake .. -DBUILD_GSTREAMER_PLUGIN=ON -DBUILD_JNI=TRUE }
-- Configuring incomplete, errors occurred!
See also "/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/build/CMakeFiles/CMakeError.log".

**SDK version number**

https://github.com/awslabs
/
[amazon-kinesis-video-streams-producer-sdk-cpp](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp)

**Open source building**
If it is a build issue, include 3rd party library version and steps to how you are building it

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
 - OS: Raspberry Pi OS Debian Buster
 - Processor: Broadcom BCM2837 chipset 1.2GHz Quad-Core ARM Cortex-A53
 - Version : Raspberry Pi 3 – Model B Original with Onboard WiFi and Bluetooth

**Additional context**
Add any other context about the problem here.

The Following Text files shows how i approched to build the SDK. And the screenshot shows the error i faced,
[AWS.Kinesis.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/8402070/AWS.Kinesis.txt)
[image]

niyatim23, 2022-04-07:
Hi @Akill047, this does not seem like an issue with the SDK. It seems like the SDK does not have access to Java installed on your system. I would suggest you to:
1. Set JAVA_HOME to jdk home (Eg: JAVA_HOME=/Library/Java/JavaVirtualMachines/amazon-corretto-16.jdk/Contents/Home)
2. Add the same to your PATH

Akill047, 2022-04-13:
Hi @niyatim23 . Thanks for the suggestion. Yes now i could see the reason for JNI error. I didn't realize that earlier. 

And can you be more specific on ,what does this mean " Add the same to your PATH" . Excuse me for not knowing. I am still learning.

disa6302, 2022-04-24:
Closing due to inactivity.

closed 2022-04-24 by disa6302