awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #844: [CRASH] CPP Producer SDK crash

rdkcteam, 2022-04-05:
**Logging**
Random crash with below stack trace is seen 
	
         thread-0 tid=3186 faulted

	[0] /usr/lib/libcurl.so.4.7.0 + 0x30150

	[1] /usr/lib/libcurl.so.4.7.0 + 0x36c0c

	[2] /usr/lib/libcurl.so.4.7.0 + 0x189d0

	[3] /usr/lib/libcproducer.so!notifyDataAvailable [amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/src/source/Response.c : 462 + 0x8]

	[4] /usr/lib/libcproducer.so!streamDataAvailableAggregate [amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/src/source/CallbacksProvider.c : 1275 + 0x4]

	[5] /usr/lib/libcproducer.so!putFrame [amazon-kinesis-video-streams-producer-sdk-cpp/dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/client/src/Stream.c : 1036 + 0x24]

**Describe the bug**
We are seeing some random crashes in SDK during put frame.

**To Reproduce**
This is random and there are not steps to recreate the issue.

**Expected behavior**
No crash for uninterrupted put frame.

**SDK version number**
v3.1.1

**Platform (please complete the following information):**
 - OS: Linux 4.9.236+
 - Processor:  aarch64 GNU/Linux
 - Version [BusyBox v1.31.1]

rdkcteam, 2022-04-22:
Sorry for late reply.
curl 7.75.0-DEV (arm-unknown-linux-gnu) libcurl/7.75.0-DEV OpenSSL/1.1.1i zlib/1.2.11

rdkcteam, 2022-04-22:
Are you suggesting to use curl version 7.68 with latest SDK v3.3.0 ?

rdkcteam, 2022-04-27:
Is there any plan to update curl library with amazon SDK ? we have updated curl on our platform for security patches.

disa6302, 2022-06-06:
We do not have a plan for this yet. Will update this issue when we have more clarity.

jdelapla, 2023-01-20:
No plans to update curl. Using versions outside of the current recommendations may result in undefined behavior, closing the ticket.

closed 2023-01-20 by jdelapla