awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #845: Compile error at liblog4cplus on Apple M1

yoi-hibino, 2022-04-09:
I have CMAKE error at liblog4cplus. My machine is Apple M1. I pulled liblog4cplus and build independently, but I don't have any issue. I added "GIT_SUBMODULES_RECURSE TRUE" to liblog4cplus-CMakeLists.txt for just in case, but same error I have. Please help me to solve this problem. 

 Here is the error message. 

configure: error: in `/Users/yoi/Development/Devblock/AirWise/kvssink/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/liblog4cplus/build/src/project_log4cplus':
configure: error: C compiler cannot create executables
See `config.log' for more details
make[2]: *** [build/src/project_log4cplus-stamp/project_log4cplus-configure] Error 77
make[1]: *** [CMakeFiles/project_log4cplus.dir/all] Error 2
make: *** [all] Error 2
CMake Error at amazon-kinesis-video-streams-producer-sdk-cpp/CMake/Utilities.cmake:99 (message):
  CMake step for liblog4cplus failed: 2
Call Stack (most recent call first):
  CMakeLists.txt:82 (build_dependency)

and here is CMake.error message

Compiling the C compiler identification source file "CMakeCCompilerId.c" failed.
Compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc 
Build flags: 
Id flags:  

The output was:
1
ld: library not found for -lSystem
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Compiling the CXX compiler identification source file "CMakeCXXCompilerId.cpp" failed.
Compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ 
Build flags: 
Id flags:  

The output was:
1
ld: library not found for -lc++
clang: error: linker command failed with exit code 1 (use -v to see invocation)

Thanks!!

niyatim23, 2022-04-11:
Hi @yoi-hibino, can you try `git clone --recursive https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git` instead of `GIT_SUBMODULES_RECURSE TRUE` in the cmake file. Please check your `CC`, `CXX` values. Please make sure you have installed these dependencies `brew install pkg-config openssl cmake gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly log4cplus gst-libav`

cerofrais, 2022-04-12:
@niyatim23 I tried with your suggestion but I still got the same error as mentioned above. please help. 

@buntyke can you also please check this

yoi-hibino, 2022-04-12:
Hi @niyatim23, thanks for your help! --recursive option of git clone doesn't solve the issue. I noticed that if I delete 
--prefix=${OPEN_SRC_INSTALL_PREFIX}  
option from liblog4plus-CMAKEList.txt, then this issue is done (instead other issue pops up because it now tries to save build output of  liblog4plus to different location). So is this issue related to prefix?
My C and CXX version is Apple clang version 13.1.6 (clang-1316.0.21.2.3)
Thanks!

yoi-hibino, 2022-04-13:
Hi @niyatim23 
by the way, here is some variables to execute CMAKE for log4cplus if it helps you to find issue.

CONFIGURE_COMMAND = [producer sdk root]/open-source/liblog4cplus/build/src/project_log4cplus/configure;CFLAGS=;--prefix=[producer sdk root]/open-source/local

CMAKE_C_FLAGS =  -isysroot/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX12.3.sdk

yoi-hibino, 2022-04-16:
Hi @niyatim23 
One finding. If I manually run cmake --build at the log4cplus directory after finished the fist execute_process in the build_dependency function (after process creates CMakeCache.txt file), then build log4cplus works fine. So it looks like the second execute_process to build liblog4cplus libraries somehow causes error. How does it happen? Do you have any idea?

yoi-hibino, 2022-04-16:
Sorry @niyatim23,  CXX value was wrong. It passes  liblog4cplus build. Thank you so much!

closed 2022-04-16 by yoi-hibino