awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #846: [QUESTION] Gstreamer plugin versus SDK C/C++ sample code

bml1g12, 2022-04-11:
I would like to better understand what functionality the gstreamer plugin offers in comparison to either the C or C++ SDK sample code? 

My understanding is the C sample code (and C++ sample code) implements a [host of SDK callbacks](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-reference-callbacks.html) for robust operation in case of network failure. When using the gstreamer `kvsink` plugin does it provide the same functionality?

I am wondering if it makes more sense when building a new media pipeline which is not already covered by the existing sample code (e.g. a pipeline that can cope with audio-only data) for production usage (robust to momentary network failures) to either:

a) Experiment and construct a pipeline with the command line `gst-launch` tool, then build a gstreamer application with gstreamer code. I understand that if I were to build a working pipeline with gst-launch, I could transfer it into a gstreamer C application with this [interesting command](https://gstreamer.freedesktop.org/documentation/gstreamer/gstparse.html?gi-language=c#gst_parse_launch_full) which converts the one liner into a full gstreamer C code. 

b) Go straight to adapting the C or C++ SDK samples to adjust for my pipeline. My understanding is that the C SDK is not intrinsically related to gstreamer at all, is that correct? Compared the convenience of gst-launch, I can imagine it might be a more time consuming approach to debug. 

c) Any other suggested option?

I'm keen to take advantage of the robust network issue handling your team has worked hard to create (e.g. stream latency pressure callbacks, video buffers when internet disconnects etc.)

Thanks for your time!

disa6302, 2022-05-20:
The CPP SDK has a dependency on Producer C SDK and PIC layers. CPP provides a gstreamer integration and kvssink which bundles a lot of our SDK functionalities (latency pressure, buffer overflow pressure callbacks, etc) through pipeline elements. If you are getting started with the SDK and intend to use GStreamer for your purpose, gst-launch with kvssink would be a good start to experimenting and constructing the pipeline for your use case. We have 3 different samples available that can be played around with based on your media source. We have a file uploader sample that takes file inputs, a single source sample that can take a single rtsp url and another that can take multiple rtsp sources as inputs.

The samples in the producer C SDK use pre-canned frame files as source. To summarize, both the CPP SDK and C SDK provide the features you are looking for.

closed 2022-05-20 by disa6302

bml1g12, 2022-05-20:
Thank you for this summary; I see there are many options available to access the SDK functionalities. 

We managed to get our audio-only pipeline working first with kvsink, then also via the CPP SDK.