awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #847: Sending to kinesis stops suddenly [QUESTION]

bluewales, 2022-04-12:
Sending audio and video to a kinesis video stream, it works fine for several minutes, then `putKinesisVideoFrame` doesn't return.

Both the audio and video threads are stuck in `putKinesisVideoFrame` and never return.  After 30 seconds, a set of WARNING log messages print, some curl thing times out I guess.

** Logging **
2022-04-12 21:11:50 VERBOSE postWriteCallback(): Processed ACK OK.
2022-04-12 21:11:50 VERBOSE postReadCallback(): postBodyStreamingReadFunc (curl callback) invoked
2022-04-12 21:11:50 VERBOSE postReadCallback(): Get Stream data returned: buffer size: 65524 written bytes: 65524 for upload handle: 0 current stream handle: 5157720
2022-04-12 21:11:50 DEBUG   postReadCallback(): Wrote 65524 bytes to Kinesis Video. Upload stream handle: 0
2022-04-12 21:11:50 VERBOSE getKinesisVideoMetrics(): Get the memory metrics size.
2022-04-12 21:11:50 VERBOSE postWriteCallback(): postBodyStreamingWriteFunc (curl callback) invoked
2022-04-12 21:11:50 DEBUG   postWriteCallback(): Curl post body write function for stream with handle: cry-OCB2053052100002 and upload handle: 0 returned: {"EventType":"BUFFERING","FragmentTimecode":4008,"FragmentNumber":"913438523331
82190031793145926067786638077538971"}

2022-04-12 21:11:50 VERBOSE postWriteCallback(): Processed ACK OK.
2022-04-12 21:11:50 VERBOSE postReadCallback(): postBodyStreamingReadFunc (curl callback) invoked
2022-04-12 21:11:50 VERBOSE postReadCallback(): Get Stream data returned: buffer size: 65524 written bytes: 5017 for upload handle: 0 current stream handle: 5157720
2022-04-12 21:11:50 DEBUG   postReadCallback(): Wrote 5017 bytes to Kinesis Video. Upload stream handle: 0
2022-04-12 21:11:50 VERBOSE postWriteCallback(): postBodyStreamingWriteFunc (curl callback) invoked
2022-04-12 21:11:50 DEBUG   postWriteCallback(): Curl post body write function for stream with handle: cry-OCB2053052100002 and upload handle: 0 returned: {"EventType":"PERSISTED","FragmentTimecode":3006,"FragmentNumber":"913438523331
82190026841385768926265045168414013"}

2022-04-12 21:11:50 VERBOSE notifyDataAvailable(): Note data received: duration(100ns): 0 bytes 0 for stream handle 0
2022-04-12 21:11:50 VERBOSE postWriteCallback(): Processed ACK OK.
2022-04-12 21:11:50 VERBOSE postReadCallback(): postBodyStreamingReadFunc (curl callback) invoked
2022-04-12 21:11:50 VERBOSE postReadCallback(): Get Stream data returned: buffer size: 65524 written bytes: 0 for upload handle: 0 current stream handle: 5157720
2022-04-12 21:11:50 VERBOSE postReadCallback(): Get Stream data returned: buffer size: 65524 written bytes: 0 for upload handle: 0 current stream handle: 5157720
2022-04-12 21:11:50 VERBOSE postReadCallback(): Get Stream data returned: buffer size: 65524 written bytes: 0 for upload handle: 0 current stream handle: 5157720
2022-04-12 21:11:50 VERBOSE postReadCallback(): Get Stream data returned: buffer size: 65524 written bytes: 0 for upload handle: 0 current stream handle: 5157720
2022-04-12 21:11:50 VERBOSE postReadCallback(): Get Stream data returned: buffer size: 65524 written bytes: 0 for upload handle: 0 current stream handle: 5157720
2022-04-12 21:11:50 VERBOSE postReadCallback(): Get Stream data returned: buffer size: 65524 written bytes: 0 for upload handle: 0 current stream handle: 5157720
2022-04-12 21:11:50 DEBUG   postReadCallback(): Pausing CURL read for upload handle: 0
STOP UPLOADING CLIP 1649797915
2022-04-12 21:12:25 WARN    curlCompleteSync(): curl perform failed for url https://s-02ab8d15.kinesisvideo.us-west-2.amazonaws.com/putMedia with result Timeout was reached: Operation too slow. Less than 30 bytes/sec transferred the l
ast 30 seconds
2022-04-12 21:12:25 WARN    curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://s-02ab8d15.kinesisvideo.us-west-2.amazonaws.com/putMedia
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=AKIAS235EOPPMLIE5SOX/20220412/us-west-2/kinesisvideo/aws4_request, SignedHeaders=connection;host;transfer-encoding;user-agent;x-amz-date;x-amzn-fragment-acknowledgment-required;x-amzn-fra
gment-timecode-type;x-amzn-producer-start-timestamp;x-amzn-stream-name, Signature=2d9c8fc6289efd62298e8b0cc864f6c99ff2975ab7ca6973ffeac47391dafc8d
    connection: keep-alive
    content-type: application/json
    host: s-02ab8d15.kinesisvideo.us-west-2.amazonaws.com
    transfer-encoding: chunked
    user-agent: AWS-SDK-KVS/3.0.0 GCC/4.9.4 Linux/4.9.84 armv7l
    X-Amz-Date: 20220412T211142Z
    x-amzn-fragment-acknowledgment-required: 1
    x-amzn-fragment-timecode-type: RELATIVE
    x-amzn-producer-start-timestamp: 1649797902.782
    x-amzn-stream-name: cry-OCB2053052100002
2022-04-12 21:12:25 DEBUG   putStreamCurlHandler(): Network thread for Kinesis Video stream: cry-OCB2053052100002 with upload handle: 0 exited. http status: 0
2022-04-12 21:12:25 WARN    putStreamCurlHandler(): Stream with streamHandle 5157720 uploadHandle 0 has exited without triggering end-of-stream. Service call result: 599
2022-04-12 21:12:25 INFO    kinesisVideoStreamTerminated(): Stream 0x4eb358 terminated upload handle 0 with service call result 599.
2022-04-12 21:12:25 DEBUG   defaultStreamStateTransitionHook(): Stream state machine retry count: 0
2022-04-12 21:12:25 DEBUG   defaultStreamStateTransitionHook():
 KinesisVideoStream base result is [599]. Executing KVS retry handler of retry strategy type [1]
2022-04-12 21:12:25 VERBOSE validateAndUpdateExponentialBackoffStatus(): Status changed from BACKOFF_NOT_STARTED to BACKOFF_IN_PROGRESS
2022-04-12 21:12:25 VERBOSE getExponentialBackoffRetryStrategyWaitTime():
 Thread Id [2938303680] Number of retries [8391086161770053633], Retry wait time [119] ms, Retry system time [16497979458697595]
2022-04-12 21:12:25 VERBOSE stepStateMachine(): State Machine - Current state: 0x0000000000000100, Next state: 0x0000000000000200, Current local state retry count [0], Max local state retry count [0], State transition wait time [119]
ms

** Any design considerations/constraints **
This is running on embedded linux.

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

niyatim23, 2022-04-13:
Hi @bluewales, can you please answer the following:
1. What version of the SDK are you using?
2. Are you running any of the SDK sample applications [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/tree/v3.3.0/samples), if not can you please try running those and see if the issue still persists?
3. Are you seeing this issue consistently or was this a single instance?

bluewales, 2022-04-13:
1. The SHA from the commit I'm using it `6a7aab4fcde0791b5b092a15e61a5cacfc57c259` which I think is version 1.2.0
2. This is not using the sample application.  This is sending data that is being generated by a camera
3. This issue occurs consistently.  It always freezes up within 5 minutes of beginning to stream

bluewales, 2022-04-13:
I seem to have created this on the wrong project.  I am actually using the c sdk, not the cpp sdk.  I will create a new issue on the other repository

closed 2022-04-13 by bluewales