awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #855: Build static library for Android

yoi-hibino, 2022-05-05:
Hi, is it possible to build this plugin as the static library for Android? 
I'm getting error while cmake processing liblog4cplus (probably due to CC and CXX setting?)
Please advise what is the proper way to build for Android.  Thanks!

I ran build script like:
export CC=${ANDROID_NDK_DIR}/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang
export CXX=${ANDROID_NDK_DIR}/toolchains/llvm/prebuilt/darwin-x86_64/bin/clang++

CMAKE_TOOLCHAIN_FILE=${ANDROID_NDK_DIR}/build/cmake/android.toolchain.cmake
CMAKE_BUILD_TYPE=Release
ABI=armeabi-v7a

cmake \
    -DCMAKE_TOOLCHAIN_FILE=${CMAKE_TOOLCHAIN_FILE} \
    -DCMAKE_BUILD_TYPE=${CMAKE_BUILD_TYPE} \
    -DANDROID_NATIVE_API_LEVEL=${ANDROID_API_LEVEL} \
    -DANDROID_NDK=${ANDROID_NDK_DIR} \
    -DANDROID_ABI=${ABI} \
    -DBUILD_GSTREAMER_PLUGIN=ON \
    -DBUILD_DEPENDENCIES=ON \
    ..

disa6302, 2022-05-20:
We have a producer SDK equivalent  in android already: https://github.com/aws-amplify/aws-sdk-android/tree/main/aws-android-sdk-kinesisvideo
And a demo app here: https://github.com/awslabs/aws-sdk-android-samples/tree/main/AmazonKinesisVideoDemoApp

Any reason you want to cross compile this SDK for Android?

niyatim23, 2022-05-31:
Closing due to inactivity, feel free to reopen in case of further issues

yoi-hibino, 2022-06-08:
@disa6302 , thanks for your response. Sorry I didn't check your reply til now. I'm working on QT project which have already has GStreamer implemented. I need to add KVS sink to the existing pipeline. I modified CMakeList to build this as static library instead of module, then this plugin works just fine inside the QT project on MacOS. Now I need to make the project runs on Android. That's why I need to cross compile this SDK for Android. 

I quickly run cmake with some valuables like following.

  export CC=${ANDROID_TOOLCHAIN_NAME}${APIn}-clang
  export CXX=${ANDROID_TOOLCHAIN_NAME}${APIn}-clang++
  export AS=${ANDROID_TOOLCHAIN_NAME}${APIn}-as
  export LD=ld
  export AR=llvm-ar
  export RANLIB=llvm-ranlib
  export STRIP=llvm-strip
  export NM=llvm-nm

  cmake \
    -DCMAKE_TOOLCHAIN_FILE=$ANDROID_NDK_ROOT/build/cmake/android.toolchain.cmake \
    -DCMAKE_ANDROID_ARCH_ABI=$ABI \
    -DABI=$ABI \
    -DCMAKE_BUILD_TYPE=Release \
    -DCMAKE_VERBOSE_MAKEFILE=ON \
    -DANDROID_TOOLCHAIN=clang \
    -DANDROID_ABI=$ABI \
    -DANDROID_PLATFORM=$APIn \
    -DBUILD_DEPENDENCIES=TRUE \
    -DBUILD_GSTREAMER_PLUGIN=TRUE \
    ..
 
It looks like the dependencies are built fine, but when the process comes to libkvspic, it fails with the following error. 

[100%] Built target libkvspic-download
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the system variable OPENSSL_ROOT_DIR (missing: OPENSSL_CRYPTO_LIBRARY OPENSSL_INCLUDE_DIR) 
-- Checking for module 'libcurl'
--   Found libcurl, version 7.68.0-DEV
-- Bitness 32 bits
CMake Error at CMakeLists.txt:113 (find_package):
  Could not find a package configuration file provided by "CURL" with any of
  the following names:

    CURLConfig.cmake
    curl-config.cmake

  Add the installation prefix of "CURL" to CMAKE_PREFIX_PATH or set
  "CURL_DIR" to a directory containing one of the above files.  If "CURL"
  provides a separate development package or SDK, be sure it has been
  installed.

yoi-hibino, 2022-06-08:
Some updates. 
By adding CMAKE_FIND_ROOT_PATH  this error is solved. It shows another error that says compatibility issue of CURL, but it also been solved by removing OPEN_SRC_INSTALL_PREFIX from find_package calls. 

Now I have an issue at  "Linking C shared library libcproducer.so"
[ 88%] Building C object dependency/libkvscproducer/kvscproducer-src/CMakeFiles/cproducer.dir/src/source/StreamLatencyStateMachine.c.o

[ 89%] Linking C shared library libcproducer.so
ld: error: ../../../../open-source/local/lib/libssl.so is incompatible with armelf_linux_eabi
ld: error: ../../../../open-source/local/lib/libcrypto.so is incompatible with armelf_linux_eabi
ld: error: /root/develop/test2/amazon-kinesis-video-streams-producer-sdk-cpp/open-source/local/lib/libcurl.so is incompatible with armelf_linux_eabi
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [dependency/libkvscproducer/kvscproducer-src/CMakeFiles/cproducer.dir/build.make:342: dependency/libkvscproducer/kvscproducer-src/libcproducer.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:185: dependency/libkvscproducer/kvscproducer-src/CMakeFiles/cproducer.dir/all] Error 2
make: *** [Makefile:136: all] Error 2

It looks like  libcproducer is been built for Linux (my host env) instead of Android. (that's why it also throw -lrt link error as well).  Any thought?