awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #857: Build Issue - missing  include file - com/amazonaws/kinesis/video/cproducer/Include.h

chrisfabri, 2022-05-10:
Hi - I'm trying to build this project, initially to get the gstreamer and jni components.

I'm following the instructions in the readme to build on the mac.  I've successfully installed 

amazon-kinesis-video-streams-producer-sdk-cpp-3.3.0/build
cmake -S ../src DBUILD_DEPENDENCIES=OFF  -DBUILD_GSTREAMER_PLUGIN=ON
-- The C compiler identification is AppleClang 12.0.5.12050022
-- The CXX compiler identification is AppleClang 12.0.5.12050022
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Library/Developer/CommandLineTools/usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Library/Developer/CommandLineTools/usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Configuring done
-- Generating done
CMake Warning:
  Manually-specified variables were not used by the project:

    BUILD_DEPENDENCIES
    BUILD_GSTREAMER_PLUGIN

but the make then fails immediately with a missing dependency which appears to be from the jni side - the file and path 

'com/amazonaws/kinesis/video/cproducer/Include.h'

does not exist in the project:

Here is the build output
-----------
make
[  4%] Building CXX object CMakeFiles/KinesisVideoProducer.dir/Auth.cpp.o
In file included from /xxx/amazon-kinesis-video-streams-producer-sdk-cpp-3.3.0/src/Auth.cpp:1:
/xxx/amazon-kinesis-video-streams-producer-sdk-cpp-3.3.0/src/Auth.h:10:10: fatal error: 'com/amazonaws/kinesis/video/cproducer/Include.h' file not found
#include "com/amazonaws/kinesis/video/cproducer/Include.h"
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/KinesisVideoProducer.dir/Auth.cpp.o] Error 1
make[1]: *** [CMakeFiles/KinesisVideoProducer.dir/all] Error 2
make: *** [all] Error 2

The make cannot find that file - nor can I - it doesn't appear to exist.

Any idea how to get this to build successfully?

Tx.

** If you would not like to open an issue to discuss your solution in open-platform, please email your question to kinesis-video-support@amazon.com **

chrisfabri, 2022-05-13:
I got past the cmake step but not the make step.

I had to copy the CMakeLists.txt, CMake, and src directories into the build directory.

I then ran 

cmake -DBUILD_DEPENDENCIES=ON -S .

which actually does something now, downloads various dependencies and completes without issue.

The make step, however, fails on type defs not found…
 
C02G73AXMD6R:build f673783$ make
Consolidate compiler generated dependencies of target kvspicUtils
[ 22%] Built target kvspicUtils
Consolidate compiler generated dependencies of target kvsCommonCurl
[ 33%] Built target kvsCommonCurl
Consolidate compiler generated dependencies of target kvspic
[ 75%] Built target kvspic
Consolidate compiler generated dependencies of target cproducer
[ 89%] Built target cproducer
Consolidate compiler generated dependencies of target KinesisVideoProducer
[ 89%] Building CXX object CMakeFiles/KinesisVideoProducer.dir/src/Auth.cpp.o
In file included from /Users/f673783/IntellijProjects/amazon-kinesis-video-streams-producer-sdk-cpp/build/src/Auth.cpp:1:
/Users/f673783/IntellijProjects/amazon-kinesis-video-streams-producer-sdk-cpp/build/src/Auth.h:72:18: error: unknown type name 'PCHAR'
    inline const PCHAR getSessionTokenIfExist() const
                          ^
/Users/f673783/IntellijProjects/amazon-kinesis-video-streams-producer-sdk-cpp/build/src/Auth.h:35:89: error: use of undeclared identifier 'MAX_UINT64'
                const std::chrono::duration<uint64_t> expiration = std::chrono::seconds(MAX_UINT64)) :
                                                                    ^
/Users/f673783/IntellijProjects/amazon-kinesis-video-streams-producer-sdk-cpp/build/src/Auth.h:25:108: error: use of undeclared identifier 'MAX_UINT64'
…

chrisfabri, 2022-05-16:
The OS compiler is the following:

C02G73AXMD6R:build f673783$ make --version
GNU Make 3.81
Copyright (C) 2006  Free Software Foundation, Inc.
This is free software; see the source for copying conditions.
There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
PARTICULAR PURPOSE.

This program built for i386-apple-darwin11.3.0

i did delete the $ROOT/build dir and start afresh; from within the build dir,

   - cmake -DBUILD_DEPENDENCIES=ON -DBUILD_GSTREAMER_PLUGIN=ON -S . 

It does pull down kvsc and pic - and builds both to static libraries locally.

it's just when it hits the local c++ code, that it then fails to find that file, with the type defs in it - even though that file is there, in the dependency tree - kvsc --> pic: 

C02G73AXMD6R:build xxxx$ ll dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/common/include/com/amazonaws/kinesis/video/common/CommonDefs.h
88 -rw-r--r--  1 xxxx  staff  41711 May 16 14:56 dependency/libkvscproducer/kvscproducer-src/dependency/libkvspic/kvspic-src/src/common/include/com/amazonaws/kinesis/video/common/CommonDefs.h

I copy the following into the build dir - src, sample, Cmake, CMakeLists.txt, kvs_log_configuration; then I run cmake from there.

chrisfabri, 2022-05-16:
[make.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/8703773/make.txt)

disa6302, 2022-06-24:
@chrisfabri , 

It does not seem to be pulling in PIC for some reason, but we have not encountered this issue in the MAC builds (our CI seems to have run fine). 

Although you have tried this, I would like to make sure the steps were followed. Can you go back to your root directory and run:

```
rm -rf build dependency open-source
mkdir build
cd build
cmake .. -DBUILD_GSTREAM_PlUGIN=ON
```

jdelapla, 2022-07-10:
Closing from inactivity

closed 2022-07-10 by jdelapla